APS-USAXS/usaxs-bluesky-ended-2023 #358: missing metadata

prjemian, 2020-07-06:
As noted in #352 

**all of the diode values:**

"upd size", needed.

```
:entry:metadata:upd_gain0
:entry:metadata:upd_gain1
:entry:metadata:upd_gain2
:entry:metadata:upd_gain3
:entry:metadata:upd_gain4
:entry:metadata:upd_bkg0
:entry:metadata:upd_bkg1
:entry:metadata:upd_bkg2
:entry:metadata:upd_bkg3
:entry:metadata:upd_bkg4
:entry:metadata:upd_bkg_err0
:entry:metadata:upd_bkgErr1
:entry:metadata:upd_bkgErr2
:entry:metadata:upd_bkgErr3
:entry:metadata:upd_bkgErr4
```

prjemian, 2020-07-06:
@jilavsky : I'm puzzled what work to do here.  Can you clarify?

jilavsky, 2020-07-06:
All of these PVs from epics are not written in the step scanning Nexus file. At least I cannot find them. 
Some of these are simply PVs in epics (gain) and some of these should be written by measure_ dark_currents (all bkg values). 
We need to add these into the Nexus file, they are important. They are written in FLyscan Nexus, so addresses are available there.

prjemian, 2020-07-22:
No `size` here:

```
In [13]: object_explorer(upd_autorange_controls)                                                                                            
============================= ============================ ===============
name                          PV reference                 value          
============================= ============================ ===============
counts_per_volt               9idcLAX:pd01:seq02:vfc       100000.0       
gain                          9idcLAX:pd01:seq02:gain      1000000000000.0
gainD                         9idcLAX:pd01:seq02:gainD     650000.0       
gainU                         9idcLAX:pd01:seq02:gainU     4000.0         
lucounts                      9idcLAX:pd01:seq02:lucounts  0              
lucurrent                     9idcLAX:pd01:seq02:lucurrent 0.0            
lurange                       9idcLAX:pd01:seq02:lurange   0              
lurate                        9idcLAX:pd01:seq02:lurate    0.0            
mode                          9idcLAX:pd01:seq02:mode      2              
ranges.gain0.background       9idcLAX:pd01:seq02:bkg0      5.0            
ranges.gain0.background_error 9idcLAX:pd01:seq02:bkgErr0   0.0            
ranges.gain0.gain             9idcLAX:pd01:seq02:gain0     10000.0        
ranges.gain1.background       9idcLAX:pd01:seq02:bkg1      5.0            
ranges.gain1.background_error 9idcLAX:pd01:seq02:bkgErr1   0.0            
ranges.gain1.gain             9idcLAX:pd01:seq02:gain1     1000000.0      
ranges.gain2.background       9idcLAX:pd01:seq02:bkg2      9.0            
ranges.gain2.background_error 9idcLAX:pd01:seq02:bkgErr2   2.2360679774998
ranges.gain2.gain             9idcLAX:pd01:seq02:gain2     100000000.0    
ranges.gain3.background       9idcLAX:pd01:seq02:bkg3      755.0          
ranges.gain3.background_error 9idcLAX:pd01:seq02:bkgErr3   52.796780204857
ranges.gain3.gain             9idcLAX:pd01:seq02:gain3     10000000000.0  
ranges.gain4.background       9idcLAX:pd01:seq02:bkg4      80373.0        
ranges.gain4.background_error 9idcLAX:pd01:seq02:bkgErr4   4743.7005596897
ranges.gain4.gain             9idcLAX:pd01:seq02:gain4     1000000000000.0
reqrange                      9idcLAX:pd01:seq02:reqrange  4              
selected                      9idcLAX:pd01:seq02:selected  0              
status                        9idcLAX:pd01:seq02:updating  0              
updating                      9idcLAX:pd01:seq02:updating  0              
============================= ============================ ===============
```

prjemian, 2020-07-22:
Add these to `baseline`:

    upd_autorange_controls
    trd_autorange_controls
    I0_autorange_controls
    I00_autorange_controls

prjemian, 2020-07-22:
@jilavsky : Do you want to record diode size?  Where can I find that information?  PV?

jilavsky, 2020-07-24:
9idcLAX:USAXS:UPDsize

prjemian, 2020-07-24:
`terms.USAXS.diode.upd_size`

prjemian, 2020-07-24:
This will go into the baseline stream (as part of `terms`)

commit 9f05c99b (2020-07-24): fixes #358
closed 2020-07-24 by prjemian