APS-USAXS/usaxs-bluesky-ended-2023 #360: Start user configuration

jilavsky, 2020-07-22:
Need user startup command in BS, which will do multiple things. 
1. Make a new "spec" file
2. Create a same named folder/directory for all user data.
3. Probe BSS and ESAF systems for user, and allow staff to reconfigure. 
4. Change BS file saving so subfolders for SAXS/USAXS/WAXS are inside this user main folder.

prjemian, 2020-09-08:
> 3. Probe BSS and ESAF systems for user, and allow staff to reconfigure

This is done by new `apsbss` program from `apstools.beamtime.apsbss`.  That tool needs proposal and ESAF number and APS cycle (AND beam line name: `9-ID-B,C`).

> ... allow staff to reconfigure

@jilavsky: How do you intend the bluesky code to do this?

prjemian, 2020-09-08:
@jilavsky : Liking this more now.  It looks through the unexpired ESAFs and Proposals and picks the first one that ends after the current time and matches the given user (last) name.  That process runs in seconds.  Example of that new code here:

```
In [60]: matchUserInApsbss("Ilavsky")
W Tue-13:08:12 - proposal(s) [67742, 57504] match user Ilavsky at this time, picking first one
I Tue-13:08:12 - ESAF 226319
I Tue-13:08:12 - Proposal 67742
I Tue-13:08:13 - APSBSS updated.
I Tue-13:08:13 - ESAF & Proposal Overview:
=============== ============================ ========================
key             PV                           value                   
=============== ============================ ========================
ESAF number     9idc:bss:esaf:id             226319                  
ESAF title      9idc:bss:esaf:title          Commission 9ID and USAXS
ESAF names      9idc:bss:esaf:users          Ilavsky,Maxey,Kuzmenko  
ESAF start      9idc:bss:esaf:startDate      2020-05-26 08:00:00     
ESAF end        9idc:bss:esaf:endDate        2020-09-28 08:00:00     
Proposal number 9idc:bss:proposal:id         67742                   
Proposal title  9idc:bss:proposal:title      USAXS of rubber compound
Proposal names  9idc:bss:proposal:users      Ilavsky,Qu              
Proposal start  9idc:bss:proposal:startDate  2020-09-17 07:00:00     
Proposal end    9idc:bss:proposal:endDate    2020-09-17 15:00:00     
Is Mail-in?     9idc:bss:proposal:mailInFlag OFF                     
=============== ============================ ========================

W Tue-13:08:13 - You should check that PVs in APSBSS contain correct information.

```

jilavsky, 2020-09-08:
OK, let me try to understand here. When we run "newUser" BSS will populate the 9idc:bss PVs with the values from BSS. These will then be stored as part of data collected by BS.  Am I correct that if this newUser populates incorrectly (there is change in operations not reflected in BSS), staff can edit the screen and simply write in correct values? 
That makes sense and is reasonable.

prjemian, 2020-09-08:
Exactly!

jilavsky, 2020-09-08:
One thing to keep in mind is, that there are many times we need to run newUser and there is no specific BSS record - commissioning, testing, setup, mail in users. In these cases the system should not fail catastrophically, but do something 'reasonable".

jilavsky, 2020-09-08:
Some users (NIST with week long experiments) prefer to run with multiple "newUser" folders to separate specific experiments. It is useful and needed when they run for 5-6 days with very different setups. In this case we may not be able to use name of user, but may have to use name of sample.

prjemian, 2020-09-08:
That's why `newUser()` is a command-line function and NOT a bluesky plan.  It should be run by staff of experienced user.

When `user` is not found, the IDs are python `None` and no update from APS Oracle is done.  Staff (experienced user) should check that PVs are edited properly.

Would you like the code to completely clear the PVs if no proposal is found?  ... no ESAF is found?

jilavsky, 2020-09-08:
No, I suggest leaving prior data in there. May be report message on command line like "Record for user name: Username  has not been found, BSS record is not updated. Prior records are retained. Edit if necessary." 
Keep in mind that it may be Fan, Andrew, Lyle or similar experienced user staring at this, they need to get some feedback to remind them what to do. They can easily call for help when reminded.

prjemian, 2020-09-08:
That's the way it is now.

prjemian, 2020-09-08:
Here's your test cases:

* `newSpecFile("sample")`
* `newUser("Ilavsky")`
* `newUser("steel")`
* `newUser("1040 steel")`

```
----> 1 newSpecFile("sample")

~/.ipython/profile_bluesky/startup/instrument/framework/callbacks.py in newSpecFile(title, scan_id)
     47     NOTE: Now part of ``instrument.utils.setup_new_user.newFile()``
     48     """
---> 49     raise RuntimeError("DEPRECATED: use ``newFile()``")
     50     # global specwriter
     51     # dt = datetime.datetime.now()

RuntimeError: DEPRECATED: use ``newFile()``

In [2]: newUser("Ilavsky")
I Tue-14:54:43 - Creating user directory: /share1/USAXS_data/2020-09/09_08_Ilavsky
I Tue-14:54:43 - Change working directory to /share1/USAXS_data/2020-09/09_08_Ilavsky
I Tue-14:54:43 - SPEC file name : 09_08_Ilavsky.dat
I Tue-14:54:43 - File will be created at end of next bluesky scan.
W Tue-14:54:43 - proposal(s) [67742, 57504] match user Ilavsky at this time, picking first one
I Tue-14:54:43 - ESAF 226319
I Tue-14:54:43 - Proposal 67742
I Tue-14:54:44 - APSBSS PVs updated from APS Oracle databases.
I Tue-14:54:45 - ESAF & Proposal Overview:
=============== ======================== ============================
key             value                    PV                          
=============== ======================== ============================
ESAF number     226319                   9idc:bss:esaf:id            
ESAF title      Commission 9ID and USAXS 9idc:bss:esaf:title         
ESAF names      Ilavsky,Maxey,Kuzmenko   9idc:bss:esaf:users         
ESAF start      2020-05-26 08:00:00      9idc:bss:esaf:startDate     
ESAF end        2020-09-28 08:00:00      9idc:bss:esaf:endDate       
Proposal number 67742                    9idc:bss:proposal:id        
Proposal title  USAXS of rubber compound 9idc:bss:proposal:title     
Proposal names  Ilavsky,Qu               9idc:bss:proposal:users     
Proposal start  2020-09-17 07:00:00      9idc:bss:proposal:startDate 
Proposal end    2020-09-17 15:00:00      9idc:bss:proposal:endDate   
Mail-in flag    OFF                      9idc:bss:proposal:mailInFlag
=============== ======================== ============================

W Tue-14:54:45 - You should check that PVs in APSBSS contain correct information.
Out[2]: '/share1/USAXS_data/2020-09/09_08_Ilavsky'

In [3]: newUser("Ilavsky")
I Tue-14:54:54 - Change working directory to /share1/USAXS_data/2020-09/09_08_Ilavsky
I Tue-14:54:54 - SPEC file name : 09_08_Ilavsky.dat
I Tue-14:54:54 - File will be created at end of next bluesky scan.
W Tue-14:54:55 - proposal(s) [67742, 57504] match user Ilavsky at this time, picking first one
I Tue-14:54:55 - ESAF 226319
I Tue-14:54:55 - Proposal 67742
I Tue-14:54:55 - APSBSS PVs updated from APS Oracle databases.
I Tue-14:54:56 - ESAF & Proposal Overview:
=============== ======================== ============================
key             value                    PV                          
=============== ======================== ============================
ESAF number     226319                   9idc:bss:esaf:id            
ESAF title      Commission 9ID and USAXS 9idc:bss:esaf:title         
ESAF names      Ilavsky,Maxey,Kuzmenko   9idc:bss:esaf:users         
ESAF start      2020-05-26 08:00:00      9idc:bss:esaf:startDate     
ESAF end        2020-09-28 08:00:00      9idc:bss:esaf:endDate       
Proposal number 67742                    9idc:bss:proposal:id        
Proposal title  USAXS of rubber compound 9idc:bss:proposal:title     
Proposal names  Ilavsky,Qu               9idc:bss:proposal:users     
Proposal start  2020-09-17 07:00:00      9idc:bss:proposal:startDate 
Proposal end    2020-09-17 15:00:00      9idc:bss:proposal:endDate   
Mail-in flag    OFF                      9idc:bss:proposal:mailInFlag
=============== ======================== ============================

W Tue-14:54:56 - You should check that PVs in APSBSS contain correct information.
Out[3]: '/share1/USAXS_data/2020-09/09_08_Ilavsky'

In [4]: newUser("steel")
I Tue-14:55:13 - Creating user directory: /share1/USAXS_data/2020-09/09_08_steel
I Tue-14:55:13 - Change working directory to /share1/USAXS_data/2020-09/09_08_steel
I Tue-14:55:13 - SPEC file name : 09_08_steel.dat
I Tue-14:55:13 - File will be created at end of next bluesky scan.
W Tue-14:55:13 - No unexpired ESAFs found that match user steel
W Tue-14:55:14 - No unexpired proposals found that match user steel
W Tue-14:55:14 - APSBSS not updated.
W Tue-14:55:14 - You should check that PVs in APSBSS contain correct information.
Out[4]: '/share1/USAXS_data/2020-09/09_08_steel'

In [5]: newUser("1040 steel")
I Tue-14:55:28 - Creating user directory: /share1/USAXS_data/2020-09/09_08_1040_steel
I Tue-14:55:28 - Change working directory to /share1/USAXS_data/2020-09/09_08_1040_steel
I Tue-14:55:28 - SPEC file name : 09_08_1040_steel.dat
I Tue-14:55:28 - File will be created at end of next bluesky scan.
W Tue-14:55:28 - No unexpired ESAFs found that match user 1040 steel
W Tue-14:55:29 - No unexpired proposals found that match user 1040 steel
W Tue-14:55:29 - APSBSS not updated.
W Tue-14:55:29 - You should check that PVs in APSBSS contain correct information.
Out[5]: '/share1/USAXS_data/2020-09/09_08_1040_steel'
```

prjemian, 2020-09-09:
@jilavsky : *SINCE* we have a PV for `9idcLAX:userName` and we already use that in `user_data.user_name`, the `newUser(user)` function should call: `user_data.user_name.put(user)`

Do you agree?  This can help the technique-based file writers (area detectors and fly data) since the path is: `/share1/USAXS_data/{year}-{month}/{month}_{day}_{user}/{month}_{day}_{user}_{technique}/`

prjemian, 2020-09-09:
Currently, `user_name` is set by execution of technique-based scans.  The value is the login user, as defined by https://github.com/APS-USAXS/ipython-usaxs/blob/e92e0b2a9cdb6299fd94c8f70adbe6b3bc9b839f/profile_bluesky/startup/instrument/framework/metadata.py#L35

Any reason we can't change that now?

prjemian, 2020-09-09:
Commit 0c478f7 refers to this issue, the comment above.

PR #379 (2020-09-09): Rearrange data location - master folder
fixes #360, fixes #361, fixes #103

* [x] (103) collect a user's data under one master folder per beam time
* [x] (360) make a new SPEC file
* [x] (360) create master folder structure
* [x] (360) get ESAF & proposal data for user
* [x] (360) configure Bluesky to save data within master folder by technique
* [x] (360) configure area detector to save data within master folder by technique
* [x] (360) update `user_data` PVs
* [x] (361) create `newUser()` function to manage all aspects of the setup for a user
* [x] (361) make issue for livedata page
* [x] (3 [... cut]
closed 2020-09-09 by prjemian