APS-USAXS/usaxs-bluesky-ended-2023 #361: Rearrange data location

jilavsky, 2020-07-25:
Modify data locations.
We want to change arrangement of data after we will be done with current users - after 8/15/2020). We want to move folders and files around. 
User of "UserName" will, starting 2020-09 have :
"spec" file "UserName.dat" located in folder 2020-09
folder "09_21_UserName" (where 09_21 is date of start of his experiment)  located in folder 2020-09
and inside the folder "09_21_UserName" will be folders for USAXS, SAXS, and WAXS. 
e.g.  
.../2020-09/09_21_UserName/09_21_UserName_usaxs 
.../2020-09/09_21_UserName/09_21_UserName_saxs
.../2020-09/09_21_UserName/09_21_UserName_waxs
Will make it easier to share data for mail-in and remote users.

prjemian, 2020-07-25:
Related to #103

prjemian, 2020-07-27:
`newFile()` should be generalized (not specific to SPEC or NeXus) according to the ideas here.

jilavsky, 2020-08-11:
Itemized list of changes:
1. newSpecFile() (or newFile()) function needs to get string ("UserName"), append month_day_ before it and create new folder in current working directory. Ideally, the new records file (aka spec file) could be stored there - But that is likely going to break livedata page and spec itself will not be able to do so. So may be leave the "spec file" in the current working directory, it is mostly used for live data page. This will create new:
.../2020-09/09_21_UserName.dat    //spec file
.../2020-09/09_21_User_Name        //new folder for all data

2. Area detector paths need to change with new folder inserted in path. SAXS and WAXS path to data will look like this:
.../2020-09/09_21_User_Name/09_21_User_Name_saxs/here_goes_file_name
.../2020-09/09_21_User_Name/09_21_User_Name_waxs/here_goes_file_name

3. Flyscans and Step scans will use the same method:
.../2020-09/09_21_User_Name/09_21_User_Name_usaxs/here_goes_file_name

As results, each "user" will have in current working directory only one folder to share or copy, folder called 09_21_UserName which will contain rest of the data structure. 
If agreed, I will modify spec to match this structure at the end of this week. 
USAXS is running until Wednesday midnight. Any testing Thursday or later.

prjemian, 2020-08-11:
Rename: replace `newSpecFile` with `newDataFile`?

prjemian, 2020-08-11:
Or, as stated in #349, `newFile`?

prjemian, 2020-08-11:
Ok

prjemian, 2020-09-08:
This affects the livedata page and the SPEC plots, as well.

prjemian, 2020-09-08:
@jilavsky
The name `newUser()` seems more appropriate.  This is a utility function.

Here is the plan for the new data directories (as will be described in the new source code):

> Create (if necessary) new user directory in
> current working directory with month, day, and
> given user name as shown in the following table.
> Each technique (SAXS, USAXS, WAXS) will be 
> reponsible for creating its subdirectory
> as needed.
> 
> purpose | folder
> --- | ---
> user data folder base |  `<CWD>/MM_DD_USER`
> SPEC data file        |  `<CWD>/MM_DD_USER/MM_DD_USER.dat`
> AD folder - SAXS      |  `<CWD>/MM_DD_USER/MM_DD_USER_saxs/`
> folder - USAXS        |  `<CWD>/MM_DD_USER/MM_DD_USER_usaxs/`
> AD folder - WAXS      |  `<CWD>/MM_DD_USER/MM_DD_USER_waxs/`

prjemian, 2020-09-08:
We'll need to point the log files into the new user directory.

prjemian, 2020-09-08:
Regarding the [log file directory](https://github.com/APS-USAXS/ipython-usaxs/issues/361#issuecomment-688608239), the log directory is set at the start of the session.  It's difficult to change it without disrupting the log process (so I believe).  While it might be nice for a user's log files to stay with the data, that's more of an operational issue than a data collection one.  The log files do not *need* to be in the same directory as the user's data.

jilavsky, 2020-09-08:
OK, truth is that log files are rarely used since metadata are in nexus files. Leave it there. 
I am pretty sure I did not move spec files inspec code, so they will be in the top director (CWD above) also. Did not even ocure to me.

prjemian, 2020-09-08:
Now is the time to change the directory where SPEC files are stored.

PR #379 (2020-09-09): Rearrange data location - master folder
fixes #360, fixes #361, fixes #103

* [x] (103) collect a user's data under one master folder per beam time
* [x] (360) make a new SPEC file
* [x] (360) create master folder structure
* [x] (360) get ESAF & proposal data for user
* [x] (360) configure Bluesky to save data within master folder by technique
* [x] (360) configure area detector to save data within master folder by technique
* [x] (360) update `user_data` PVs
* [x] (361) create `newUser()` function to manage all aspects of the setup for a user
* [x] (361) make issue for livedata page
* [x] (3 [... cut]
closed 2020-09-09 by prjemian