APS-USAXS/usaxs-bluesky-ended-2023 #369: Live data do not work for flyscans

jilavsky, 2020-08-05:
FLyscans generated using BS do not show in Livedata page. I assumed there was name issue with _flyscan_ in the name, but it is not working still.

prjemian, 2020-08-05:
Possibly related to https://github.com/APS-USAXS/livedata/issues/40

prjemian, 2020-08-05:
It's the Bluesky `Flyscan` that does not plot.

```
In [1]: listruns()                                                                                                                                                            
========= ========================== ======= ======= ========================================
short_uid date/time                  exit    scan_id command                                 
========= ========================== ======= ======= ========================================
36e64fc   2020-08-05 12:41:29.683250 success 84      Flyscan(pos_X=25.0, pos_Y=43.0, thic ...
ff9bfc4   2020-08-05 12:41:21.731059 success 83      measure_USAXS_Transmission(detectors ...
c819599   2020-08-05 12:38:50.468503 success 82      Flyscan(pos_X=125.0, pos_Y=43.0, thi ...
dfb40a4   2020-08-05 12:38:42.473130 success 81      measure_USAXS_Transmission(detectors ...
5b07253   2020-08-05 12:36:23.166480 success 80      Flyscan(pos_X=115.0, pos_Y=43.0, thi ...
7f6a6a8   2020-08-05 12:36:15.312854 success 79      measure_USAXS_Transmission(detectors ...
30a7dfb   2020-08-05 12:33:55.687340 success 78      Flyscan(pos_X=105.0, pos_Y=43.0, thi ...
608937c   2020-08-05 12:33:47.796471 success 77      measure_USAXS_Transmission(detectors ...
136a507   2020-08-05 12:33:34.364334 success 76      tune_a2rp(detectors=['PD_USAXS'], nu ...
47a0a9a   2020-08-05 12:33:16.559883 success 75      tune_ar(detectors=['PD_USAXS'], num= ...
1ed25b9   2020-08-05 12:33:00.532098 success 74      tune_mr(detectors=['I0_USAXS'], num= ...
c4b09d7   2020-08-05 12:30:46.141228 success 73      Flyscan(pos_X=95.0, pos_Y=43.0, thic ...
d3b18b8   2020-08-05 12:30:38.112004 success 72      measure_USAXS_Transmission(detectors ...
629f7f8   2020-08-05 12:28:18.602391 success 71      Flyscan(pos_X=85.0, pos_Y=43.0, thic ...
ba69ac6   2020-08-05 12:28:10.744894 success 70      measure_USAXS_Transmission(detectors ...
80fb1a9   2020-08-05 12:25:51.355312 success 69      Flyscan(pos_X=75.0, pos_Y=43.0, thic ...
8b028c7   2020-08-05 12:25:43.218373 success 68      measure_USAXS_Transmission(detectors ...
cfa7651   2020-08-05 12:23:23.844117 success 67      Flyscan(pos_X=65.0, pos_Y=43.0, thic ...
dd2f60d   2020-08-05 12:23:15.993760 success 66      measure_USAXS_Transmission(detectors ...
0e15033   2020-08-05 12:23:02.223840 success 65      tune_a2rp(detectors=['PD_USAXS'], nu ...
========= ========================== ======= ======= ========================================
```

This Bluesky scan:

index | title | type | scan | file | started | ended
-- | -- | -- | -- | -- | -- | --
4753 | BK11_Blank | FlyScan | 84 | /share1/USAXS_data/2020-08/08_05_Bevand.dat | 2020-08-05 12:41:29 | 2020-08-05 12:43:41

has this exception: `IndexError: list index out of range`

prjemian, 2020-08-05:
Here's what the SPEC file shows:

```
#S 84  Flyscan(pos_X=25.0, pos_Y=43.0, thickness=4.0, scan_title=BK11_Blank)
#D Wed Aug 05 12:41:29 2020
#C Wed Aug 05 12:41:29 2020.  plan_type = generator
#C Wed Aug 05 12:41:29 2020.  uid = 36e64fc6-f272-4195-8374-ed07a859cadd
#C Wed Aug 05 12:41:30 2020.  start USAXS Fly scan
#C Wed Aug 05 12:41:30 2020.  HDF5 configuration file: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
#MD uid = 36e64fc6-f272-4195-8374-ed07a859cadd
#MD EPICS_CA_MAX_ARRAY_BYTES = 1280000
#MD EPICS_HOST_ARCH = linux-x86_64
#MD action = USAXSscan
#MD archive = Command file: usaxs.mac
====== ======================= =====================
line # action                  parameters           
====== ======================= =====================
1      CURRENT_EXPERIMENT_NAME Samples              
19     saxsExp                 25, 43, 4, BK11_Blank
20     saxsExp                 35, 43, 4, BK1       
21     saxsExp                 45, 43, 4, BK2       
22     saxsExp                 55, 43, 4, BK3       
23     saxsExp                 65, 43, 4, BK4       
24     saxsExp                 75, 43, 4, BK5       
25     saxsExp                 85, 43, 4, BK6       
26     saxsExp                 95, 43, 4, BK7       
27     saxsExp                 105, 43, 4, BK8      
28     saxsExp                 115, 43, 4, BKB9     
29     saxsExp                 125, 43, 4, BKB10    
30     saxsExp                 25, 43, 4, BK11_Blank
33     waxsExp                 25, 43, 4, BK11_Blank
34     waxsExp                 35, 43, 4, BK1       
35     waxsExp                 45, 43, 4, BK2       
36     waxsExp                 55, 43, 4, BK3       
37     waxsExp                 65, 43, 4, BK4       
38     waxsExp                 75, 43, 4, BK5       
39     waxsExp                 85, 43, 4, BK6       
40     waxsExp                 95, 43, 4, BK7       
41     waxsExp                 105, 43, 4, BK8      
42     waxsExp                 115, 43, 4, BKB9     
43     waxsExp                 125, 43, 4, BKB10    
44     waxsExp                 25, 43, 4, BK11_Blank
46     preUSAXStune                                 
49     USAXSscan               25, 43, 4, BK11_Blank
50     USAXSscan               35, 43, 4, BK1       
51     USAXSscan               45, 43, 4, BK2       
52     USAXSscan               55, 43, 4, BK3       
53     USAXSscan               65, 43, 4, BK4       
54     USAXSscan               75, 43, 4, BK5       
55     USAXSscan               85, 43, 4, BK6       
56     USAXSscan               95, 43, 4, BK7       
57     USAXSscan               105, 43, 4, BK8      
58     USAXSscan               115, 43, 4, BKB9     
59     USAXSscan               125, 43, 4, BKB10    
60     USAXSscan               25, 43, 4, BK11_Blank
====== ======================= =====================

#MD beamline_id = APS 9-ID-C USAXS
#MD bss_aps_cycle = 2020-2
#MD bss_beamline_name = 9-ID-B,C
#MD datetime = 2020-08-05 11:55:15.794281
#MD esaf_id = 226319
#MD esaf_title = Commission 9ID and USAXS
#MD filename = usaxs.mac
#MD fly_scan_time = 120.0
#MD full_filename = /share1/USAXS_data/2020-08/usaxs.mac
#MD hdf5_file = BK11_Blank_0171.h5
#MD hdf5_path = /share1/USAXS_data/2020-08/08_05_Bevand_usaxs
#MD iso8601 = 2020-08-05 12:41:03.468106
#MD line_number = 60
#MD login_id = usaxs@usaxscontrol.xray.aps.anl.gov
#MD mail_in_flag = OFF
#MD parameters = ['25', '43', '4', 'BK11_Blank']
#MD pid = 35446
#MD principal_user = Hammons
#MD proposal_id = 67436
#MD proposal_title = Understanding the Kinetics of Helium Bubble Growth and Mobility in Different Irradiated Metals at Temperature
#MD proprietary_flag = OFF
#MD purpose = tuner
#MD sample_thickness_mm = 4.0
#MD sx = 25.0
#MD sy = 43.0
#MD thickness = 4.0
#MD title = BK11_Blank
#MD versions = {'bluesky': '1.6.4', 'ophyd': '1.5.1', 'databroker': '1.0.6', 'apstools': '1.3.3', 'epics': '3.4.2', 'numpy': '1.18.5', 'matplotlib': '3.2.2', 'spec2nexus': '2021.1.7', 'pyRestTable': '2020.0.2', 'pymongo': '3.9.0'}
#P0 
#N 0
#C no data column labels identified
#C Wed Aug 05 12:43:40 2020.  fly scan completed in 130.11181211471558 s
#C Wed Aug 05 12:43:40 2020.  finished writing fly scan HDF5 file: /share1/USAXS_data/2020-08/08_05_Bevand_usaxs/BK11_Blank_0171.h5
#C Wed Aug 05 12:43:40 2020.  num_events_baseline = 2
#C Wed Aug 05 12:43:40 2020.  num_events_mca = 1
#C Wed Aug 05 12:43:40 2020.  exit_status = success
```

prjemian, 2020-08-05:
The problem is that line `#MD archive = Command file: usaxs.mac` and the lines that follow it are all part of the same metadata but `spec2nexus` reads the additional lines as experiment data.

Best not to put this into the `RE.md` dictionary then.  Find another place to keep the raw command list.

prjemian, 2020-08-05:
Tested this by removing that extra content and this scan passed the `spec_scan.interpret()` step in `scanplots.py`.

prjemian, 2020-08-05:
Maybe a `archive.splitlines()` would do it?  (Takes the multi-line string and makes it a list, one string per line.)

prjemian, 2020-08-05:
@jilavsky : Do you really want the command list to go into the run's metadata (and thus end up in the SPEC data file)?  For now, I'll comment it out to test that as a fix for the problem.

commit 63ce071a (2020-08-05): fixes #369
closed 2020-08-05 by prjemian