APS-USAXS/usaxs-bluesky-ended-2023 #372: Set in Progress PV in autocollection

jilavsky, 2020-08-10:
In order to prevent Igor from moving motors while instrument moves between positions, we need to set pv: 9idcLAX:dataColInProgress to 1 when starting preUSAXStune and useModeRadiography (well, the BS equivalents) and set to 0 when done. Igor checks on this PV when attempting to move motors in epics. 
This PV is already set by execution of command file. We just need to add this to execution of the other two commands.

prjemian, 2020-08-10:
Looks easy.

prjemian, 2020-08-28:
`user_data.collection_in_progress`

prjemian, 2020-08-28:
For `instrument.plans.mode_changes.mode_Radiography()` (spec: useModeRadiography), setting `user_data.collection_in_progress` is easy.  Where to *unset* it?

jilavsky, 2020-08-28:
Unset when it arrives to radiography mode and finishes moving. At the very end of the code. Basically, we want to prevent Igor from moving any motors when spec or BS is moving them. Only one should be in charge.

prjemian, 2020-08-28:
Got it: in `mode_Radiography()`, set it to 1 before moving anything, set it to 0 before exiting `mode_Radiography()`.

PR #375 (2020-08-28): set value of user_data.collection_in_progress as appropriate
will fix #372
closed 2020-08-28 by prjemian