APS-USAXS/usaxs-bluesky-ended-2023 #373: Add jpg images from 9idFLY2

jilavsky, 2020-08-18:
We now have Blackfly camera providing optical images of the sample while data are collected. 
We want to add option to stare jpg image at the time of each data collection - USAXS, SAXS, and WAXS.  There is PV which controls this:
9idcLAX:saveFLY2Image, bo, 0=No, 1=Yes.  
Uses AD, PV prefix: 9idFLY2:

Need to set path to path of each technique (reminder, it will be e.g., CWD/09_01_UserName/09_01_UserName_saxs per another issue here).  Use same name for image with extension jpg. If PV is set to 1, trigger save jpg image in the same location as Nexus file at the time of triggering USAXS/SAXS/ or WAXS data collection. 
Camera will be configured - time/parameters/focus - manually by staff during setup until we know more and can prepare some script. 
9idcLAX:saveFLY2Image PV will be available to users on GUI and provided for user control.

prjemian, 2020-08-26:
Nice!

prjemian, 2020-08-28:
@kmpeters , @jilavsky 
With EPICS AreaDetector prefix `9IDFLY2:`, what ADcamera plugin is used to interface with this camera?  Or do I just need the image plugin?

prjemian, 2020-08-28:
> enable BS to collect radiography images if needed

Means: configure to use the HDF5 FileWriter plugin

prjemian, 2020-08-28:
prefix | image subject | features
--- | ---- | ---
`9idFLY2:` | optical images of the sample | We want to add option to record jpg image at the time of each data collection - USAXS, SAXS, and WAXS. PV `9idcLAX:saveFLY2Image` (bo controlled by users on GUI)  0=No, 1=Yes
`9idFLY:` | X-ray radiography images

prjemian, 2020-08-28:
```
(/share1/bluesky/conda_envs/bluesky_2020_9) bash-4.2$ caget -S 9idFLY2:JPEG1:FilePath_RBV
9idFLY2:JPEG1:FilePath_RBV /mnt/share1/USAXS_data/2020-08/
```

prjemian, 2020-08-28:
@jilavsky What about the previous (image-only) Blackfly camera? https://github.com/APS-USAXS/ipython-usaxs/blob/2d49d2ea0f2dd951f5bac030046519e5a0752012/profile_bluesky/startup/instrument/devices/area_detector_common.py#L36

    9idFLY1:

Leave that alone for now?

prjemian, 2020-08-28:
Only `9idFLY2:` is on now but all three cameras are defined in the bluesky config at this time.

prefix | bluesky (ophyd) name | capabilities
--- | --- | ---
`9idFLY1:` | `blackfly_det` | no file writer configured
`9idFLY2:` | `blackfly_optical` | JPEG writer configured
`9idFLY:` | `blackfly_radiography` | HDF5 writer configured

Here's proof of the bluesky setup for `blackfly_optical`:

<details>

```
In [2]: object_explorer(blackfly_optical)                                                                                                   
=========================== ==================================== ===================================================
name                        PV reference                         value                                              
=========================== ==================================== ===================================================
cam.acquire                 9idFLY2:cam1:Acquire_RBV             0                                                  
cam.acquire_period          9idFLY2:cam1:AcquirePeriod_RBV       0.2                                                
cam.acquire_time            9idFLY2:cam1:AcquireTime_RBV         0.03275173950195313                                
cam.array_callbacks         9idFLY2:cam1:ArrayCallbacks_RBV      1                                                  
cam.array_counter           9idFLY2:cam1:ArrayCounter_RBV        3916                                               
cam.array_rate              9idFLY2:cam1:ArrayRate_RBV           0.0                                                
cam.array_size.array_size_x 9idFLY2:cam1:ArraySizeX_RBV          1288                                               
cam.array_size.array_size_y 9idFLY2:cam1:ArraySizeY_RBV          964                                                
cam.array_size.array_size_z 9idFLY2:cam1:ArraySizeZ_RBV          0                                                  
cam.array_size_bytes        9idFLY2:cam1:ArraySize_RBV           1241632                                            
cam.asyn_io                 9idFLY2:cam1:AsynIO                  0                                                  
cam.bandwidth               9idFLY2:cam1:Bandwidth               5.920562744140625                                  
cam.bin_x                   9idFLY2:cam1:BinX_RBV                1                                                  
cam.bin_y                   9idFLY2:cam1:BinY_RBV                1                                                  
cam.binning_mode            9idFLY2:cam1:BinningMode_RBV         0                                                  
cam.color_mode              9idFLY2:cam1:ColorMode_RBV           0                                                  
cam.convert_pixel_format    9idFLY2:cam1:ConvertPixelFormat_RBV  0                                                  
cam.corrupt_frames          9idFLY2:cam1:CorruptFrames_RBV       0                                                  
cam.data_type               9idFLY2:cam1:DataType_RBV            1                                                  
cam.detector_state          9idFLY2:cam1:DetectorState_RBV       0                                                  
cam.driver_dropped          9idFLY2:cam1:DriverDropped_RBV       0                                                  
cam.dropped_frames          9idFLY2:cam1:DroppedFrames_RBV       0                                                  
cam.firmware_version        9idFLY2:cam1:FirmwareVersion                                                            
cam.format7_mode            9idFLY2:cam1:Format7Mode_RBV         0                                                  
cam.frame_rate              9idFLY2:cam1:FrameRate_RBV           0                                                  
cam.frame_type              9idFLY2:cam1:FrameType_RBV           0                                                  
cam.gain                    9idFLY2:cam1:Gain_RBV                0.0                                                
cam.image_mode              9idFLY2:cam1:ImageMode_RBV           2                                                  
cam.manufacturer            9idFLY2:cam1:Manufacturer_RBV        Point Grey Research                                
cam.max_packet_size         9idFLY2:cam1:MaxPacketSize           1440                                               
cam.max_size.max_size_x     9idFLY2:cam1:MaxSizeX_RBV            1288                                               
cam.max_size.max_size_y     9idFLY2:cam1:MaxSizeY_RBV            964                                                
cam.min_x                   9idFLY2:cam1:MinX_RBV                0                                                  
cam.min_y                   9idFLY2:cam1:MinY_RBV                0                                                  
cam.model                   9idFLY2:cam1:Model_RBV               Blackfly BFLY-PGE-13H2C                            
cam.nd_attributes_file      9idFLY2:cam1:NDAttributesFile                                                           
cam.num_exposures           9idFLY2:cam1:NumExposures_RBV        1                                                  
cam.num_exposures_counter   9idFLY2:cam1:NumExposuresCounter_RBV 0                                                  
cam.num_images              9idFLY2:cam1:NumImages_RBV           1                                                  
[... cut ...]
jpeg.width                  9idFLY2:JPEG1:ArraySize0_RBV         0                                                  
jpeg.write_file             9idFLY2:JPEG1:WriteFile_RBV          0                                                  
jpeg.write_message          9idFLY2:JPEG1:WriteMessage                                                              
jpeg.write_status           9idFLY2:JPEG1:WriteStatus            0                                                  
=========================== ==================================== ===================================================
```

</details>

PR #376 (2020-08-28): add 2 new PointGrey Blackfly cameras
fixes #373
closed 2020-08-28 by prjemian