APS-USAXS/usaxs-bluesky-ended-2023 #378: implement _when_ to save blackfly_optical images

prjemian, 2020-08-28:
In #373, it was noted there is a PV (controlled by user in GUI) that indicates when a JPEG should be saved (into the same folder with NeXus and other data files).

> `9idcLAX:saveFLY2Image`, bo, 0=No, 1=Yes.

> Need to set path to path of each technique (reminder, per #361 it will be e.g., CWD/09_01_UserName/09_01_UserName_saxs). Use same name as for that technique (SAXS, WAXS, USAXS) for image but use extension jpg and jpg writer (not hdf5 writer). If PV is set to 1, trigger save jpg image in the same location as Nexus file at the time of triggering USAXS/SAXS/ or WAXS data collection.
Camera will be configured - time/parameters/focus - manually by staff during setup until we know more and can prepare some script.

prjemian, 2020-09-10:
technique-based path is easy:

```
profile_bluesky/startup/instrument/plans/scans.py:    flyscan_path = techniqueSubdirectory("usaxs")
profile_bluesky/startup/instrument/plans/scans.py:    SAXSscan_path = techniqueSubdirectory("saxs")
profile_bluesky/startup/instrument/plans/scans.py:    WAXSscan_path = techniqueSubdirectory("waxs")
```

prjemian, 2020-09-10:
@jilavsky As I see written above, you want like this in each plan (USAXS, SAXS, WAXS):

```
# in device setup
should_save_jpeg = EpicsSignal("9idcLAX:saveFLY2Image", name="should_save_jpeg")

#  in each technique's scan plan:
detector_list = [...]
if should_save_jpeg.get() in (1, True, "Yes", "save the JPEG image, please"):
    # save JPEG image
    detector_list.append(blackfly_optical)
# then, trigger all the detectors in detector_list
```

prjemian, 2020-09-10:
er, not quite.  Don't want a JPEG at _every_ step of a uascan.  Maybe just one at start?

prjemian, 2020-09-10:
More like this, *before* the scan does its thing:

```
    if blackfly_optical.should_save_jpeg.get() in (1, "Yes"):
        # TODO: set directory & file name for JPEG
        yield from blackfly_optical.take_image()
        jpeg_name = blackfly_optical.jpeg1.full_file_name.get()
        # TODO: path name needs to be corrected from detector to usaxscontrol
        _md["sample_image_jpeg"] = jpeg_name
    yield from uascan(
        ...
```

path correction needed because this is how it looks:

```
In [7]: blackfly_optical.jpeg1.full_file_name.get()
Out[7]: '/mnt/share1/USAXS_data/2020-08/test_0009.jpg'
In [8]: !ls /mnt/share1
ls: cannot access /mnt/share1: No such file or directory
```

prjemian, 2020-09-10:
These are the principal steps executed from command line, one-by-one:

```
In [1]: newUser("Pete")
I Thu-16:37:38 - Creating user directory: /share1/USAXS_data/2020-09/09_10_Pete
I Thu-16:37:38 - Change working directory to /share1/USAXS_data/2020-09/09_10_Pete
I Thu-16:37:38 - SPEC file name : 09_10_Pete.dat
I Thu-16:37:38 - File will be created at end of next bluesky scan.
W Thu-16:37:38 - No unexpired ESAFs found that match user Pete
W Thu-16:37:38 - No unexpired proposals found that match user Pete
W Thu-16:37:38 - APSBSS not updated.
W Thu-16:37:38 - You should check that PVs in APSBSS contain correct information.
Out[1]: '/share1/USAXS_data/2020-09/09_10_Pete'

In [10]: uascan_path = techniqueSubdirectory("Jemian")

In [12]: uascan_path
Out[12]: '/share1/USAXS_data/2020-09/09_10_Pete/09_10_Pete_Jemian'

In [14]: os.path.exists("/share1/USAXS_data/2020-09/09_10_Pete/09_10_Pete_Jemian/")
Out[14]: True

In [16]: RE(bps.mv(
    ...:             blackfly_optical.jpeg1.file_path,
    ...:             "/mnt" + os.path.abspath(uascan_path) + "/"  # MUST end with "/"
    ...:             )
    ...:             )
Out[16]: ()

In [17]: RE(blackfly_optical.take_image())
D Thu-16:45:41 - make_filename: /mnt/share1/USAXS_data/2020-09/09_10_Pete/09_10_Pete_Jemian/test_0010.jpg
D Thu-16:45:41 - write_path: /mnt/share1/USAXS_data/2020-09/09_10_Pete/09_10_Pete_Jemian/
Out[17]: ()                                                                                                                                 

In [18]: jpeg_name = blackfly_optical.jpeg1.full_file_name.get()

In [19]:         if jpeg_name.startswith("/mnt/share1"):
    ...:             jpeg_name = jpeg_name[4:]
    ...: 

In [20]: os.path.exists(jpeg_name)
Out[20]: True

In [21]: jpeg_name
Out[21]: '/share1/USAXS_data/2020-09/09_10_Pete/09_10_Pete_Jemian/test_0010.jpg'
```

jilavsky, 2020-09-11:
Yes, 
this looks exactly like what I expect and what spec is doing (hopefully). 
- Note, that AD software should create path requested, if necessary. All out AD iocs are set to create up to 2 levels deep path if necessary (I just verified for FLY2). Therefore checking on path existence is nice touch, but may not be necessary. 
- **Problem (I think)** Code needs to sync **complete name** with technique name - for string part I think that is line 362 in the modified code, BUT what I do not see is syncing to usaxs order number. We want to make sure there are both: test_0010.jpg and test_0010.h5 = with the same order numbers. In case user zeroes order numbers (routine), these order numbers would get out of sync. In spec I push into "9idFLY2:JPEG1:FileNumber" proper order number in. I do not see that done in BS code. (basically, if we collect USAXS, SAXS, WAXS sequence, Jpeg writer will write 3x same name_orderNumber, but in three different paths). 

Else, this is what we want for usaxs, and also saxs and waxs. But that is copy of the same, really.

prjemian, 2020-09-11:
Good contribution

prjemian, 2020-09-11:
Create a plan for the steps described in e05ad68f.  This should be in the `plans` directory (not with the device) since devices should not be implementing plans directly.  (This could create problem loops when importing some other part of the instrument from a device.  These loops can stop bluesky from starting.)

prjemian, 2020-09-11:
@jilavsky : I'll leave in-place the code to make the directory if it does not exist.  This makes the code testable without relying on area detector to make the directory.  It also detects early on if the directory cannot be created (for whatever reason).  Not sure how area detector reports that type of problem.

prjemian, 2020-09-11:
Right, the code should set the `detector.plugin.FileNumber` to the usaxs order number (PV: `9idcLAX:USAXS:FS_OrderNumber`, ophyd object: `terms.FlyScan.order_number`).

In the example above, the full path is:

    /share1/USAXS_data/2020-09/09_10_Pete/09_10_Pete_Jemian/test_0010.jpg

Symbolically, that is:

    USAXS_DATA/YYYY-MM/MM_DD_USER/MM_DD_USER_TECHNIQUE/FILENAME_ORDERNUMBER.EXT

What should we set for the file name?

prjemian, 2020-09-11:
Use this for the file name: https://github.com/APS-USAXS/ipython-usaxs/blob/19635a53ced11e24f16e64c6db7e8f42ba42b0d1/profile_bluesky/startup/instrument/plans/scans.py#L437

prjemian, 2020-09-11:
In reply to https://github.com/APS-USAXS/ipython-usaxs/pull/384#commitcomment-42236082, we should make the process of how we set the file name very clear.

**FlyScan**
https://github.com/APS-USAXS/ipython-usaxs/blob/19635a53ced11e24f16e64c6db7e8f42ba42b0d1/profile_bluesky/startup/instrument/plans/scans.py#L446-L451

**SAXS**
https://github.com/APS-USAXS/ipython-usaxs/blob/19635a53ced11e24f16e64c6db7e8f42ba42b0d1/profile_bluesky/startup/instrument/plans/scans.py#L669
https://github.com/APS-USAXS/ipython-usaxs/blob/19635a53ced11e24f16e64c6db7e8f42ba42b0d1/profile_bluesky/startup/instrument/plans/scans.py#L680-L684

**WAXS**
https://github.com/APS-USAXS/ipython-usaxs/blob/19635a53ced11e24f16e64c6db7e8f42ba42b0d1/profile_bluesky/startup/instrument/plans/scans.py#L829
https://github.com/APS-USAXS/ipython-usaxs/blob/19635a53ced11e24f16e64c6db7e8f42ba42b0d1/profile_bluesky/startup/instrument/plans/scans.py#L840-L844

If you want different, now is the best time to change it.

prjemian, 2020-09-11:
`scan_title` is passed to each scan: `def TECHNIQUE(pos_X, pos_Y, thickness, scan_title, md=None):`

prjemian, 2020-09-11:
Follow-up, looking at `terms.FlyScan`, there is no file name attribute: https://github.com/APS-USAXS/ipython-usaxs/blob/19635a53ced11e24f16e64c6db7e8f42ba42b0d1/profile_bluesky/startup/instrument/devices/general_terms.py#L21-L31

We use the `scan_title` consistently for all plans to set the file name.

PR #384 (2020-09-14): record JPEG of sample when scanning
fixes #378
closed 2020-09-14 by prjemian