APS-USAXS/usaxs-bluesky-ended-2023 #387: guard slit motors stuck in moving state after bluesky tune

prjemian, 2020-09-14:
After tuning the guard slit motors (top, bottom, inboard, outboard), each of the physical motor records is stuck in the moving state as shown.
[image]

prjemian, 2020-09-14:
Toggling the Torque to `Disable` sets the motors in motion!  Which ends soon.  Setting the Torque back to `Enable` (to return to previous setting).

Why does this happen?

prjemian, 2020-09-14:
[image]

prjemian, 2020-09-14:
Repeated tuning of guard slit centers with Torque disabled.  All four blades tuned as expected but tune ended with Exception shown:
[image]

possibly fixed in https://github.com/bluesky/ophyd/issues/757

kmpeters, 2020-09-14:
> @kmpeters: Why do these motors get stuck Moving and why is disabling the torque effective? These are stepping motors using StepPack drivers and MaxV controllers.

The commands that are sent to the controller when torque is enabled/disabled should only have an effect on the axis if it is a servo motor or a stepper motor with an encoder:

  https://github.com/epics-motor/motorOms/blob/b7c2bd4c610fc58e3e204e12bfa06844e8272a82/omsApp/src/devOmsCom.cc#L127-L128

Here is a description of the HN command from the MAXv command reference manual:
> **HN HOLD ON**
> For servo axes, the HN command closes the loop, enabling the PID. For servo axes, this mode is
disabled when the HF command is entered, when the servo error becomes too large or a limit is
encountered.
> For stepper with encoder feedback axes, the HN command enables position correction after a
move and activates the HD, HG and HV commands for stepper axes with encoders. For stepper
axes with encoders, this mode will be canceled (as though via an HF command) if an JG, JF, HM,
HR, SO, SP command is entered, if a limit is encountered or the maximum allowable position
error is executed.

It is likely that disabling torque works because it causes the motor record to process.  If that is the case, a ``caput <motor>.PROC 1`` would also work to unstick the motor.

I have some questions:
* Is the motor actually moving when the motor record gets stuck in the moving state?
* Is it possible to get just one axis stuck in the moving state?

The inconsistency of the readback positions in both screenshots is worrisome.

kmpeters, 2020-09-14:
LAX's iocConsole wasn't working.  I've restarted it.  Please recreate the problem again.  I'd like to see if there are any motors on the VxWorks shell.

prjemian, 2020-09-14:
The torque setting is not important.  It is *changing* the torque setting that triggers the `.STUP` (status update) field to update.  The problem with setting the `.STUP` field from bluesky/ophyd is in the way that it works.  Sending a 1 to `.STUP` tells the record support to update the status.  When that is done, it sets `.STUP` to 0.  Since ophyd is waiting for the record support to provide a CA monitor that says `.STUP=1`, it times out and leaves its `ophyd.DeviceStatus` object hanging.  This results in a `RuntimeError: Another set() operation is in progress.`

prjemian, 2020-09-14:
@kmpeters The motors were not in motion when they found to be stuck.  Processing the motor sends it moving to its destination.

prjemian, 2020-09-14:
@kmpeters What is the difference between writing ` to `.STUP` or to `.PROC`?  Requests a status update or process the motor record, of course.  Does one of these do something else?

It's problematic to use the `.STUP` field (ophyd waits for it to get to 1 but the record sets it to 0 - that's a timeout issue for ophyd).  `PROC` is much easier since it stays at 1.

prjemian, 2020-09-14:
We got it stuck again.  Setting the `.PROC` to 1 for each motor reset them.  This would be OK with ophyd since it will get 1 when it expects to get 1.

kmpeters, 2020-09-14:
Why is ophyd monitoring the STUP field?  Most of the time other fields (MSTA, MIP, DMOV) should be monitored instead.

Writing to the STUP field will cause the record to process and do a status update.  This is really only useful for older drivers, since model-3 drivers have idle polling.

Writing to the PROC field will cause the motor record to process.  What happens when it processes should be determined by which other fields have changed.  I don't know if a status update is the default behavior if nothing else changes.

prjemian, 2020-09-14:
@kmpeters We were sending `.STUP` a 1 after the tune guard slits plan based on what I thought was your advice.  Anyway, not doing that now.  Instead, sending a 1 to `.PROC` after doing a sync on each axis.  Working now.

prjemian, 2020-09-14:
Marking as duplicate of #253.

commit edafd6ae (2020-09-14): fixes #253 fixes #387
closed 2020-09-14 by prjemian