APS-USAXS/usaxs-bluesky-ended-2023 #392: Need useModeBlackFly

jilavsky, 2020-09-14:
This is in spec, sets to imaging mode, but then moves to dx/dy for BlackFly radiography camera. Has 4 new values - dx, dy, fliterAl, filterTi for Black Fly in radiography, This is useful, not urgent. Will need 4 new PVs, for now is all in spec. Needs to move in epics for configuration, so October.

prjemian, 2020-11-11:
Belongs in `instrument/plans/mode_changes.py` and entry added to `instrument/plans/command_list.py`.

prjemian, 2020-11-11:
What PVs are needed?

prjemian, 2020-11-11:
This is a start but needs more details before it is ready.

jilavsky, 2020-11-12:
At this moment we need at least two new PVs:  dx and dy positions for Blackfly mode. It is similar to USAXS mode, but with different positions of dx/dy, basically Radiography mode. For completeness we should also add ALfilter and TiFilter and exposure time. I will add this tomorrow when I am at APS. LAX sometimes hangs and it is useful to be local to reboot it manually. I will provide PV names. 
My code sets first USAXS mode and then moves dx/dy to the BlackFly camera.

jilavsky, 2020-11-12:
New PVs: 
9idcLAX:USAXS:BlackFly_dx
9idcLAX:USAXS:BlackFly_dy
9idcLAX:USAXS:BlackFly_Al_Filter
9idcLAX:USAXS:BlackFly_Ti_Filter

new command modeBlackFly should function same as modeRadiography but set to these dx/dy/filters.

jilavsky, 2020-11-12:
Here is existing spec code :

  useModeUSAXS
  DCMfeedbackOFF
  print "Preparing for BlackFly mode ... please wait ..."
  moveDetector   __FlyDxInPlace   __FlyDyInPlace
  openTiFilterShutter
  insertPF4filters __FlyAlFilter  __FlyTiFilter
  comment "Ready for BlackFly mode"
  epics_put ("9idcLAX:USAXS:timeStamp",   date())
  epics_put ("9idcLAX:USAXS:state",       "BlackFly camera Mode")
  epics_put ("9idcLAX:USAXS:macroFileTime",      date())
  epics_put ("9idcLAX:USAXS:scanning",    0)
  epics_put ("9idFLY1:cam1:Acquire",    1)

The only additional feature is starting for user the FLY1 AD image collection. This is displayed in ImageJ picture whichis manually controlled.

prjemian, 2020-11-12:
```
In [3]: listdevice(terms.USAXS.blackfly)
=============================== ====== ==========================
name                            value  timestamp                 
=============================== ====== ==========================
terms_USAXS_blackfly_dx         2.1    2020-11-12 12:24:45.413105
terms_USAXS_blackfly_dy         101.75 2020-11-12 12:24:52.296438
terms_USAXS_blackfly_filters_Al 15     2020-11-12 12:24:56.179771
terms_USAXS_blackfly_filters_Ti 1      2020-11-12 12:24:59.429771
=============================== ====== ==========================

Out[3]: <pyRestTable.rest_table.Table at 0x7fb41ff1df10>

In [4]: object_explorer(terms.USAXS.blackfly)
========================= ================================ ======
name                      PV reference                     value 
========================= ================================ ======
USAXS.blackfly.dx         9idcLAX:USAXS:BlackFly_dx        2.1   
USAXS.blackfly.dy         9idcLAX:USAXS:BlackFly_dy        101.75
USAXS.blackfly.filters.Al 9idcLAX:USAXS:BlackFly_Al_Filter 15    
USAXS.blackfly.filters.Ti 9idcLAX:USAXS:BlackFly_Ti_Filter 1     
========================= ================================ ======
```

prjemian, 2020-11-12:
@jilavsky emailed:

> Put line 97 (insert filters) before Open Ti shutter and then I agree. It would not matter for this camera, but it is always good practice to insert filters first and then open shutter...

prjemian, 2020-11-12:
Agree.  Is that what SPEC does?  Your text (above) suggests not.  Might check it to confirm.

closed 2020-11-12 by prjemian