APS-USAXS/usaxs-bluesky-ended-2023 #396: SAXS FLY2 images order number does not advance

jilavsky, 2020-09-15:
Collecting sequence of SAXS images, FLY2 jpg file saver is still on order number 49 while Hdf files are on 70. We need to add before each FLY2 image collecting sync between SAXS camera FileNumber and JPG writer FileNumber.

jilavsky, 2020-09-15:
Here is the bug:
record_sample_image_on_demand in sample_imaging.py sets line 42 to  :
      'det.jpeg1.file_number, terms.FlyScan.order_number.get()-1'
That advances ONLY when we do Flyscans which is wrong. 
This function needs to get the order_number from the call - it needs to be moved to parameters. And then each calling functions in scans.py needs to call this with its own orderNumber of their technique. 
Keep in mind,that when we collect 100 samples, we collect 100 SAXS images (so SAXS order number increases by 100), then 100 WAXS images and only at the end 100 Flyscans. 
We need to change  record_sample_image_on_demand
and then lines 763, 356, 578, and 925, in each case pass in at that moment current technique order number. 
I believe those are:
waxs_det.hdf1.file_number.get()
saxs_det.hdf1.file_number.get()
terms.FlyScan.order_number.get()
and really cannot figure out what step scan one is. But I am sure you know. 
Hopefully this helps.

prjemian, 2020-09-15:
We must verify each technique advances its order number appropriately.

prjemian, 2020-09-15:
SAXS & WAXS are advanced by area detector.

PR #400 (2020-09-15): order numbers are specific to each scan technique
The *order number* is written into the data file name for each technique and for the associated sample image.

Ensure the numbers are coordinated at the start of a set of samples and that they increment properly for each scan technique.

fixes #396 and fixes #398
closed 2020-09-15 by jilavsky