APS-USAXS/usaxs-bluesky-ended-2023 #401: axes tunes fail sometimes

jilavsky, 2020-09-15:
Messages come with "tune ... failed" from BS. 
see tune_mr 'edae2b21", scan number 73
We get error reports: LivePlot did not get any data that corresponds to the x axis. m_stage_r. 
same message for axis y, I0_USAXS. 
This seems quite common today, have not seen any over night, but received at least 6 or more today morning.

jilavsky, 2020-09-15:
See tunebug.txt in CWD

jilavsky, 2020-09-15:
11:57 e-mail: USAXS tune failed for axis m_stage_r
11:58 e-mail: USAXS tune failed for axis a_stage_r
This is not very common occurrence, but scan data look OK.

jilavsky, 2020-09-15:
This is now getting critical, it is happening every time we tune the data.

prjemian, 2020-09-15:
might be a duplicate of #390

prjemian, 2020-09-15:
We did not understand these tunes should start failing since we are using the same session that ran almost 1k runs last night with no problem.  Instead, we exited the session, did a `git pull` on the instrument package, then restarted bluesky.

I suspect we created a problem sometime in our interactive session this morning and that caused the problem.

prjemian, 2020-09-15:
not a problem after we restarted

prjemian, 2020-09-15:
transient problem gone away after restart

closed 2020-09-15 by prjemian

prjemian, 2020-09-15:
FWIW, here was the console output we observed with the problem.  Perfectly decent tune scan.  LivePlot could not find either of the plottable axes.  I do not see a column with the X axis (motor positions):

```
Transient Scan ID: 73     Time: 2020-09-15 11:46:09
Persistent Unique Scan ID: 'edae2b21-b91c-4c37-983b-6cce472df33c'
New stream: 'baseline'
New stream: 'aps_current_monitor'
New stream: 'primary'                                                                                                                                                         
+-----------+------------+------------+
|   seq_num |       time |   I0_USAXS |
+-----------+------------+------------+
|         1 | 11:46:10.8 |        541 |
|         2 | 11:46:11.2 |        752 |                                                                                                                                       
|         3 | 11:46:11.6 |       1216 |                                                                                                                                       
|         4 | 11:46:12.1 |       2145 |                                                                                                                                       
|         5 | 11:46:12.5 |       3851 |                                                                                                                                       
|         6 | 11:46:12.9 |       7005 |                                                                                                                                       
|         7 | 11:46:13.3 |      10687 |                                                                                                                                       
|         8 | 11:46:13.7 |      16346 |                                                                                                                                       
|         9 | 11:46:14.1 |      21881 |                                                                                                                                       
|        10 | 11:46:14.6 |      27599 |                                                                                                                                       
|        11 | 11:46:15.0 |      32012 |                                                                                                                                       
|        12 | 11:46:15.4 |      34801 |                                                                                                                                       
|        13 | 11:46:15.8 |      37044 |                                                                                                                                       
|        14 | 11:46:16.2 |      38260 |                                                                                                                                       
|        15 | 11:46:16.6 |      38882 |                                                                                                                                       
|        16 | 11:46:17.0 |      39157 |                                                                                                                                       
|        17 | 11:46:17.5 |      39397 |                                                                                                                                       
|        18 | 11:46:17.9 |      39391 |                                                                                                                                       
|        19 | 11:46:18.3 |      38617 |                                                                                                                                       
|        20 | 11:46:18.7 |      36936 |                                                                                                                                       
|        21 | 11:46:19.1 |      33898 |                                                                                                                                       
|        22 | 11:46:19.5 |      30408 |                                                                                                                                       
|        23 | 11:46:20.0 |      24834 |                                                                                                                                       
|        24 | 11:46:20.4 |      17924 |                                                                                                                                       
|        25 | 11:46:20.8 |      12196 |                                                                                                                                       
|        26 | 11:46:21.2 |       7266 |                                                                                                                                       
|        27 | 11:46:21.6 |       3918 |                                                                                                                                       
|        28 | 11:46:22.0 |       1997 |                                                                                                                                       
[... cut ...]
center           8.747700554073973
initial_position 8.7477005        
final_position   8.7477005        
================ =================

                                                                                                                                                                              
I Tue-12:33:24 - PeakStats : no max reported
I Tue-12:33:24 - axis: a_stage_r
I Tue-12:33:24 - starting position: 8.7477005
I Tue-12:33:24 - peak detected: False
I Tue-12:33:24 - final position: 8.7477005
```