APS-USAXS/usaxs-bluesky-ended-2023 #411: Instrument flag that it is in operations now now working with autocollection

jilavsky, 2020-09-18:
I think we have broken the flag that instrument is in operations. This is flag which should be raised any time instrument executes command file. Request that it is raised when mode_Radiography and preUSAXStune are run broke its original function. These functions are called during run_command_file and so this flag is removed even though user is collecting long sequence of data. We need to get the logic better. 
Or, if needed, remove the flag raise/removal from preUSAXStune and mode_Radiography and live with the fact the users will have time they do not know instrument is working. But at this time this flag is not available while running 12 hours long experiment in BS. 
And Igor experiment relies on this flag to prevent users from moving motors while collecting data.

prjemian, 2020-09-18:
Probably, this is the offending line: https://github.com/APS-USAXS/ipython-usaxs/blob/05e0c7c25e30c0f260420d23023dbc7f8219aaa5/profile_bluesky/startup/instrument/plans/scans.py#L147

Is it OK if this PV toggles?  ON at start of `preUSAXStune` (and others), then OFF at end.  Then, the SAXS, USAXSscan, and WAXS scans need to set ON and OFF for their specific scan sections.

prjemian, 2020-09-18:
Checking the other scans, this ophyd object is ONLY controlled during `preUSAXStune`.  I will add ON/OFF settings around the four scan actions.

prjemian, 2020-09-18:
Note:  There are two other symbols: `terms.SAXS.collecting` (`9idcLAX:collectingSAXS`) and `terms.WAXS.collecting` (`9idcLAX:collectingWAXS`) .

prjemian, 2020-09-18:
@jilavsky Next time you restart the bluesky session, this *should* be fixed.

jilavsky, 2020-09-18:
OK, I'll try and we will see.

jilavsky, 2020-09-20:
The flag does not work as needed. **I will fix this as test myself after next week as exercise.** 
The flag needs to be raised under these conditions:
when run_command_file is invoked from command line - at the beginning and lowered at the end
when running autocollection (remote operations) when:
1. user submitted command file, raise at the beginning, lower at the end of file
2. User requetse preUSAXStune, raise at the beginning, lower at  the end
3. User requested mode_Radiography, raise at the start of move, lower at the end. 
The flag control belongs in the plans (run_command_file, and autocollection), not in the commands themselves. I think this is completely wrong sentence, but makes sense to me... 

This flag is also used to play music at the end of data collection, and at this moment that software is jukebox, not useful information tool. And annoying everyone... 

But I want to use this as training for my own. Will need help to get all setup and configured, and mainly checked for syntax... October job!

prjemian, 2020-09-21:
Marked as *Good first issue* for you to learn the github toolset.

prjemian, 2020-10-01:
I believe the problem is one of design.  In the [conditions above](https://github.com/APS-USAXS/ipython-usaxs/issues/411#issuecomment-695827759), **1** is easy to do but is quickly subverted whenever **2** is run from the command set.  There must be some additional flag that says _command file is running so `preUSAXStune` should not obey condition 2_.  Agree?

prjemian, 2020-10-01:
Also condition **3** is not run from the command file, right?

jilavsky, 2020-10-01:
Yes, this is by design at this moment. This will be easily fixed - the flag needs to be raised by auto_collect plan before it starts the mode_radiography and preUSAXStune, and lowered after these are done, these plans themselves should not touch the flag. That way when they are run manually they do not cause troubles.  I know how to fix this.  I am nearly setup on my computer to edit the files and make the changes. I just need to create a branch for it, so I do not screw up anything else.

prjemian, 2020-10-01:
You want a short Teams meetup so I can look over you shoulder?

prjemian, 2020-10-01:
you --> your

jilavsky, 2020-10-01:
sure. Good idea. I have time until 3pm

commit 9dbd2fa2 (2020-09-18): fixes #411
PR #418 (2020-10-01): Raises Flag "in operations" properly before and after command
Fixes #411
commit f8a2f8b9 (2020-10-01): FIxes #411
closed 2020-10-01 by jilavsky