APS-USAXS/usaxs-bluesky-ended-2023 #412: Suspender did not work with remote_ops

jilavsky, 2020-09-18:
APS had beam dump, instrument suspended correctly. After beam came back, instrument did not continue with execution, but went back to waiting for user next command, it bailed out from executing the usaxs.mac. Something about set() call. See below what was available. 

'''
Suspender SuspendFloor(EpicsSignalRO(read_pv='S:SRcurrentAI', name='aps_current', parent='aps', value=10.38883101109133, timestamp=1600457315.691016, auto_monitor=False, string=False), sleep=100, pre_plan=None, post_plan=None,tripped_message=) reports a return to nominal conditions. Will sleep for 100 seconds and then release suspension at 2020-09-18 14:30:15.
Suspender SuspendBoolLow(EpicsSignal(read_pv='9idcLAX:blCalc:userCalc1', name='usaxs_CheckBeamStandard', value=1.0, timestamp=1600457914.179518, tolerance=1e-05, auto_monitor=False, string=False, write_pv='9idcLAX:blCalc:userCalc1', limits=False, put_complete=False), sleep=0, pre_plan=None, post_plan=None,tripped_message=) reports a return to nominal conditions. Will sleep for 0 seconds and then release suspension at 2020-09-18 14:38:34.
D Fri-14:40:45 - Error: writing fly scan HDF5 file: /share1/USAXS_data/2020-09/09_18_Rishi/09_18_Rishi_usaxs/KM_J_0012.h5 - Another set() call is still in progress for user_data_state
I Fri-14:40:45 - finished writing fly scan HDF5 file: /share1/USAXS_data/2020-09/09_18_Rishi/09_18_Rishi_usaxs/KM_J_0012.h5
I Fri-14:40:45 - HDF5 output complete: /share1/USAXS_data/2020-09/09_18_Rishi/09_18_Rishi_usaxs/KM_J_0012.h5
New stream: 'mca'                                                                                                                                                                                
D Fri-14:40:45 - after return: 5402.783227682114s
W Fri-14:40:45 - Exception during execution of command usaxs.mac:
Another set() call is still in progress for user_data_state
I Fri-14:40:45 - waiting for next user command
'''

prjemian, 2020-09-19:
There were two exceptions here that led to the fail to resume after the suspender released its hold.  The suspender did its job, as reported, but encountered the `Another set() call is still in progress` exception on completion of the next fly scan at the new call to save data.

The `another set in progress` exception is a [known problem](https://github.com/bluesky/ophyd/issues/757) which may have an [upstream fix](https://github.com/bluesky/ophyd/pull/901).

For now, I believe we could do either of these:

* [ ] apply locally the [suggested fix](https://github.com/bluesky/ophyd/pull/901#issuecomment-685951874)
* [ ] wait for the ophyd PR to be tested and merged into the next ophyd release

@jilavsky: What's your preference?

jilavsky, 2020-11-18:
This is likely consequence of : Another set() call is still in progress for user_data_state which is fixed now. Close for now.,

closed 2020-11-18 by jilavsky