APS-USAXS/usaxs-bluesky-ended-2023 #413: spec Flyscan save crashed out and stopped data collection

jilavsky, 2020-09-19:
User & staff report:
Around 1am the flyscan got stuck and I received a call from Kabir. The Python error message is copied below. I had to use CTRL^C to stop it. The experiment was restarted with a modified macro (we already did three cycles). This looks like a communication error (stale EPICS).
 Ivan
__________________________________________
 
 
ERROR:root:Not connected PV=None  ophyd=metadata_0078
Traceback (most recent call last):
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.py", line 368, in <module>
    main()  # production system - SPEC uses this
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.py", line 346, in main
    sfs = SaveFlyScan(dataFile, configFile)
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.py", line 61, in __init__
    self._prepare_to_acquire()
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.py", line 193, in _prepare_to_acquire
    raise EpicsNotConnected()
__main__.EpicsNotConnected
^Citing for Flyscan to save data 4169 secec

jilavsky, 2020-09-19:
This must have been momentary loss of connection to some PV, same worked before 100s of times a later 100s of times. Possibly due to IT network activities which are scheduled after midnight - backups, scanning, etc... 
We need to make this more robust and if a PV is not available, write 0 or empty string instead of its value and keep going. Loss of one data set is acceptable - but loss of time resolved series of data, as in this case, should not be stopped by not connected PVs. We have enough issues when major piece of hardware/software fail. And even in those cases one could have argument system should continue in case useful data can still be generated... 
Network and ioc issues are way too common to be allowed to cause catastrophic failure of time series experiment.

jilavsky, 2020-09-19:
I wonder if we could on Monday - no beam - try to harden the system. Try to run the saveFlyData code and remove some PVs it is using... We can simply shut down soft ioc - like the bss/esaf ioc. And harden the python against "PV not available".   May be PVs should have default values and if not available, default values should be used. 
At the end, the correct solution is to expect that PVs are not available and handle that as best as possible.

prjemian, 2020-09-19:
sounds good

prjemian, 2020-09-19:
until then, hands-off unless this happens once more?

jilavsky, 2020-09-19:
Yes, let's not try to debug with real users. 
Ivan will run heater experiment for users today. Tomorrow are also real remote users, I will use BS again during daytime with users on the call and helping them with data. And then is Fan+Linkam+spec annealing run.  No time for debug unless critically important.

jilavsky, 2020-09-19:
More reported:
The same error happened two more times before 3am.
Below is the log for the second occurrence at 2am.

Sat Sep 19 02:02:14 2020.  Please wait approximately 135s to complete FlyScan..
ERROR:root:Not connected PV=None  ophyd=metadata_0144
Traceback (most recent call last):
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.py", line 368, in <module>
    main()  # production system - SPEC uses this
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.py", line 346, in main
    sfs = SaveFlyScan(dataFile, configFile)
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.py", line 61, in __init__
    self._prepare_to_acquire()
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.py", line 193, in _prepare_to_acquire
    raise EpicsNotConnected()
__main__.EpicsNotConnected

jilavsky, 2020-09-19:
@prjemian - anything to do here? Roll back to non-BS code until Monday? Any obvious fix possible? What is available?

prjemian, 2020-09-21:
The code failed because certain PVs were not available (_not connected_) at the time the fly scan data file was to be written.  The code to identify the specific PVs was not properly reporting the PV name(s).

Diagnostics today revealed these PVs were for support of one of the Linkam temperature controllers.  Since the instrument makes the choice whether or not to run this IOC, we can expect that these PVs might be missing for _any_ scan.  The `saveFlyData` code must be robust to handle such a situation.

We came to the agreement that when a PV is not connected, we still create an HDF5 dataset but set its value to `not connected` (a string).  This is different from the case when a PV has a `None` value in which we write the string `no data`.

We tested the revised code by stopping the Linkam IOC and writing the file.  Log warnings are generated but the file is written as planned.  With the Linkam IOC started, no log warnings are generated since all PVs have connected, and the file is written as planned.  We then verified further by stopping another IOC (`9idc:bss:`) and confirmed that the relevant PVs were marked with `not connected` values.

prjemian, 2020-09-21:
As part of this work, we created a management script for the 9idcbss IOC (#414).

commit ae034f85 (2020-09-21): fixes #413
closed 2020-09-21 by prjemian