APS-USAXS/usaxs-bluesky-ended-2023 #416: BS does not post command file to web page

jilavsky, 2020-09-23:
Re(run_command_file()) does not seem to post copy of command file to our web page: 
https://usaxslive.xray.aps.anl.gov/specmacro.txt

This is issue when running:
1. BS directly using Re(run_command_file())
2. Using BS to run remote operations mode.

prjemian, 2020-09-23:
See this run: Bluesky `c5d77c3`

prjemian, 2020-09-23:
```
In [12]: listruns(num=5, plan_name="documentation_run")
========= ========================== ======= ======= ===================
short_uid date/time                  exit    scan_id command            
========= ========================== ======= ======= ===================
c5d77c3   2020-09-23 14:05:57.413814 success 316     documentation_run()
88e0a04   2020-09-23 12:34:41.992328 success 231     documentation_run()
db79edf   2020-09-23 11:08:53.880900 success 146     documentation_run()
c12241a   2020-09-23 09:57:22.922008 success 80      documentation_run()
9a8147e   2020-09-23 08:44:15.051844 success 11      documentation_run()
========= ========================== ======= ======= ===================

Out[12]: <pyRestTable.rest_table.Table at 0x7fa8e0b9d910>
```

prjemian, 2020-09-23:
Need to find out if bluesky is writing the command file name into `9idcLAX:USAXS:macroFile`

From the current data, it is not obvious that bluesky is writing it.  From https://usaxslive.xray.aps.anl.gov/raw-report.html:

name | id | description | value | units | timestamp
-- | -- | -- | -- | -- | --
9idcLAX:USAXS:userDir | spec_dir | SPEC data directory | /share1/USAXS_data/2020-09/09_23_Gassensmith |   | 2020-09-23 15:09:57.443248
9idcLAX:USAXS:macroFile | spec_macro_file | SPEC macro file | Anovitz_Stretch.mac |   | 2020-09-22 19:03:30.607811
9idcLAX:USAXS:specScan | spec_scan | SPEC scan number | 8 |   | 2020-09-23 15:19:58.349669
9idcLAX:USAXS:state | state | SPEC what is happening? | SAXS Mode |   | 2020-09-23 15:19:58.365564

prjemian, 2020-09-23:
This is related to #361.  This current code sets the PV: https://github.com/APS-USAXS/ipython-usaxs/blob/ae034f85ba7ef9a3d043d327cbe7ddd21524d06a/profile_bluesky/startup/instrument/plans/command_list.py#L76

prjemian, 2020-09-23:
That code is called here: https://github.com/APS-USAXS/ipython-usaxs/blob/ae034f85ba7ef9a3d043d327cbe7ddd21524d06a/profile_bluesky/startup/instrument/plans/command_list.py#L123-L124

prjemian, 2020-09-23:
And that is the problem.  Should be called as a bluesky plan, with `yield from `.

prjemian, 2020-09-23:
@jilavsky : Once you next restart your bluesky session, you can verify this problem is fixed.  Re-open if still a problem.

jilavsky, 2020-09-23:
OK, in 20 minutes

prjemian, 2020-09-24:
Moving this issue to the `livedata` repo

commit c401357e (2020-09-23): fixes #416