APS-USAXS/usaxs-bluesky-ended-2023 #420: Need solution for expected operations

jilavsky, 2020-10-15:
In December run (November 17-18 setup and testing, then operations until end of cycle) we will need to run following unique plan:
Get all ready for measurement (usaxs or saxs or waxs).
**Reset motor and push Jog button in epics on specific epics motor. See PI_onR and PI_onF for procedure.** 
Collect data (USAXS uses step scans, includes transmission measurement for USAXS and SAXS). 
**Stop the jog - see PI_Off.**
continue to next item on the list in command file. 

FYI: This is motor 9idcPI:c867:c0:m1  
This is to rotate sample with appropriate speed while collecting data to integrate over same area on sample for each point. 
This motor is intentionally NOT controlled by spec and presumably also is not in BS. Pushing jog button starts rotation at speed which is setup in epics. Switching off and on (with resets) is needed since else epics runs out of counters and stops mid way. 
in spec we put commands PI_onR or PI_onL before executing a line with USAXSscan, saxsExp, or waxsExp and PI_Off after each scan. 

I need to know how to do this in BS properly. Do we have some pre-scan and post-scan macros we can define as needed at time of user operations and change as needed? Keep in mind, this is something we will run for few hours during daytime, then run something else, come back to this day or two later, etc. We need to modify the code running this on fly.

I can imagine two options:
1. teach the command run_command_file commands PI_onR, PI_onL, PI_Off 
2. implement pre/post user hook function where we can reload this function as needed. Is this something which could be done as user plan and be redefined on fly by run - modulename?   

def PI_onR '{
          epics_put("9idcPI:c867:c0:m1.HOMR",1)
          sleep(10)
          epics_put("9idcPI:c867:c0:m1.JOGR",1)
}'

def PI_onF '{
          epics_put("9idcPI:c867:c0:m1.HOMF",1)
          sleep(10)
          epics_put("9idcPI:c867:c0:m1.JOGF",1)
}'

def PI_Off '{
          epics_put("9idcPI:c867:c0:m1.JOGR",0)
          sleep(1)
          epics_put("9idcPI:c867:c0:m1.JOGF",0)
          sleep(1)
}'

prjemian, 2020-11-04:
Usual caution statements (from [motor record documentation](https://epics.anl.gov/bcda/synApps/motor/motorRecord.html)):

> field | RW? | name | type
> -- | -- | -- | --
> HOMF | R/W* | Home Forward | SHORT
> HOMR | R/W* | Home Reverse | SHORT
> JOGF | R/W* | Jog motor Forward | SHORT
> JOGR | R/W* | Jog motor Reverse | SHORT
> 
> When either `HOMF` or `HOMR` is set to 1, the motor will decelerate to a stop if already moving, move in the indicated direction (in dial coordinates) at the acceleration specified by ACCL and a speed specified by HVEL, until the hardware detects the "home" switch has become active. Then the hardware will do something hardware dependent in response to its "home" command, if any. (The OMS hardware causes the motor to decelerate to a stop.) When the motor stops, the VAL field will be set equal to the RBV field, and the DVAL field will be set equal to the DRBV field.  These fields can be set to 1, but setting either field to 0 results in an error.  The record sets HOM[F/R] to zero when the homing procedure is either completed or aborted.
> 
> When either `JOGF` or `JOGR` is set to 1, the motor will decelerate to a stop if already moving, and move in the indicated direction (in user coordinates) at an acceleration specified by ACCL and speed VELO, until the field goes to 0. Then the motor will set VAL to RBV and DVAL to DRBV, decelerate to a stop, and execute a (backlash-corrected, if BDST is nonzero) move to the position at which the field went to 0. 
> 
> The `JOGx` fields are dangerous when used over channel access, because the motor does not stop moving until a second message is received. If a very busy network should cause that second message to be lost, the motor will travel to its limit switch or hard stop.

prjemian, 2020-11-04:
Not so difficult to implement the `9idcPI:c867:c0:m1` controls as you describe.

1. Create a Device class that has components for HOMF, HOMR, JOGF, JOGR fields
1. Add _plan_-style methods for `onF()`, `onR()`, & `off()`
1. Create instance of device class with the `9idcPI:c867:c0:m1` PV, name might be `PI`.
1. call `yield from PI.onF()` (or other) as needed from new custom plan.

prjemian, 2020-11-04:
**Q**: Are there _pre_ & _post_ hooks that can be modified on the fly?
**A**: We can add the hooks that would call features modified with a `%run -m module`, default value would be a no-op.  (python calls this `pass`)

prjemian, 2020-11-04:
It is easy to 

> teach the [new commands] to `run_command_file`

prjemian, 2020-11-04:
Looks like this:

```
In [2]: pi_c867.read()
Out[2]: 
OrderedDict([('pi_c867_home_forward',
              {'value': 0, 'timestamp': 1603226370.131181}),
             ('pi_c867_home_reverse',
              {'value': 0, 'timestamp': 1603226370.131181}),
             ('pi_c867_jog_forward',
              {'value': 0, 'timestamp': 1603226370.131181}),
             ('pi_c867_jog_reverse',
              {'value': 0, 'timestamp': 1603226370.131181}),
             ('pi_c867_speed',
              {'value': 3600.0, 'timestamp': 1603226370.131181}),
             ('pi_c867_velocity',
              {'value': 360.0, 'timestamp': 1603226370.131181})])

In [3]: listdevice(pi_c867)
==================== ====== ==========================
name                 value  timestamp                 
==================== ====== ==========================
pi_c867_home_forward 0      2020-10-20 15:39:30.131181
pi_c867_home_reverse 0      2020-10-20 15:39:30.131181
pi_c867_jog_forward  0      2020-10-20 15:39:30.131181
pi_c867_jog_reverse  0      2020-10-20 15:39:30.131181
pi_c867_speed        3600.0 2020-10-20 15:39:30.131181
pi_c867_velocity     360.0  2020-10-20 15:39:30.131181
==================== ====== ==========================

Out[3]: <pyRestTable.rest_table.Table at 0x7f5bce3ae340>

```

prjemian, 2020-11-04:
Now, need three plans and then add them to the lingo for the `run_command_file`.

prjemian, 2020-11-04:
> Reset motor and ...

What does this mean, exactly?  Redefine the motor position as zero?

prjemian, 2020-11-04:
All done but the reset feature

jilavsky, 2020-11-06:
Reset motor is done by following sequence:
epics_put("9idcPI:c867:c0:m1.HOMR",1)
sleep(10)
It is wrong terminology on my side. What I meant is home the motor in epics to make sure counter is set to 0. Note, that we need to wait for 10 seconds after Home command to let motor finish. It seeks home marker. Can take up to 8 seconds, I believe, so we set timeout 10. Alternative would be to wait until finish if such signal is available.

prjemian, 2020-11-06:
Like "Done Moving" (`.DONE`) goes to 1?

prjemian, 2020-11-06:
Need to check that this works:

1.  Before homing, DONE=1
1. while homing, DONE=0
1. when home found, DONE=1 again

prjemian, 2020-11-10:
test sequence (using `camonitor sky:m2.{{HOM,JOG}{F,R},DMOV}`)

Note in all cases the initial time stamp is from a previous operation and does not reflect accurately the time the motion PV was set.

### home forward

In other terminal: `caput sky:m2.HOMF 1` and motor *homes*, then stops automatically (it has no home signal so it stops right away).

```
sky:m2.HOMF                    2020-11-10 15:42:44.659373 1  
sky:m2.DMOV                    2020-11-10 15:43:46.214289 0  
sky:m2.DMOV                    2020-11-10 15:43:49.817971 1  
sky:m2.HOMF                    2020-11-10 15:43:49.817971 0  
```

### home reverse

Same as above, using `sky:m2.HOMR`

```
sky:m2.HOMR                    2020-11-10 15:43:49.817971 1  
sky:m2.DMOV                    2020-11-10 15:43:57.385292 0  
sky:m2.DMOV                    2020-11-10 15:43:57.725142 1  
sky:m2.HOMR                    2020-11-10 15:43:57.725142 0  
```

### jog forward

A *jog* runs until stopped.  Here we start a forward jog in a different console, using: `caput sky:m2;JOGF 1`, then stop it with  `caput sky:m2;JOGF 0`.  The motor `DMOV` registers the stop about 0.2 s after the command was sent.

```
sky:m2.JOGF                    2020-11-10 15:43:57.725142 1  
sky:m2.DMOV                    2020-11-10 15:44:11.569006 0  
sky:m2.JOGF                    2020-11-10 15:44:15.841740 0  
sky:m2.DMOV                    2020-11-10 15:44:16.041945 1  
```

### jog reverse

same as above, but with  `sky:m2;JOGR`

```
sky:m2.JOGR                    2020-11-10 15:44:16.041945 1  
sky:m2.DMOV                    2020-11-10 15:44:23.898986 0  
sky:m2.JOGR                    2020-11-10 15:44:29.454658 0  
sky:m2.DMOV                    2020-11-10 15:44:29.654876 1  
```

prjemian, 2020-11-10:
We can create a `.set()` method with a callback on the `DMOV` field to end *home* operation before the timeout has expired.  That would be more optimal.  It's a change from the very simple-minded implementation now.  Would need testing with the motor able to reach the *home* marker.  Is that possible?  Otherwise, this is good to use as-is.

prjemian, 2020-11-11:
@jilavsky Can we test with the motor now?  It needs to be able to find the home marker for the test.

jilavsky, 2020-11-12:
I plan to go to APS tomorrow morning. let me setup the motor for you. This specific PI motor is always able to reach its home marker as it is used to rotate sample at 1Hz during the measurements. It is spinning wheel.  It can always rotate safely. It is now powered down. I'll let you know when ready for testing.

prjemian, 2020-11-12:
Timeout when `PI_onF()`, need to check move algorithm.

<details>

```
In [13]: RE(PI_onF())
---------------------------------------------------------------------------
FailedStatus                              Traceback (most recent call last)
<ipython-input-13-df22cd8b46c2> in <module>
----> 1 RE(PI_onF())

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
     79             except StopIteration as e:
     80                 # discard the exhausted generator

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in monitor_during_wrapper(plan, signals)
    801     plan1 = plan_mutator(plan, insert_after_open)
    802     plan2 = plan_mutator(plan1, insert_before_close)
--> 803     return (yield from plan2)
    804 
    805 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
[... 71 lines cut ...]
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

FailedStatus: Status(obj=EpicsSignal(read_pv='9idcPI:c867:c0:m1.JOGF', name='pi_c867_jog_forward', parent='pi_c867', value=0, timestamp=1605221216.672273, auto_monitor=True, string=False, write_pv='9idcPI:c867:c0:m1.JOGF', limits=False, put_complete=False), done=True, success=False)
```

</details>

Here's a `camonitor` trace:

```
9idcPI:c867:c0:m1.HOMF         2020-11-12 16:45:33.482908 1  
9idcPI:c867:c0:m1.DMOV         2020-11-12 16:46:46.660746 0  
9idcPI:c867:c0:m1.HOMF         2020-11-12 16:46:46.661121 0  
9idcPI:c867:c0:m1.DMOV         2020-11-12 16:46:46.661327 1  
9idcPI:c867:c0:m1.JOGF         2020-11-12 16:46:46.661598 1  
9idcPI:c867:c0:m1.JOGF         2020-11-12 16:46:56.672274 0  
```

prjemian, 2020-11-12:
Is the motor ON now?

prjemian, 2020-11-13:
Now watches motor DMOV field and returns when HOME signal is found or raises exception after given timeout.

prjemian, 2020-11-18:
Solved.

closed 2020-11-18 by prjemian