APS-USAXS/usaxs-bluesky-ended-2023 #421: Can Nexus file content be controlled for step scans?

jilavsky, 2020-10-15:
Steps scans in BS generate relatively larger files which take surprisingly long time to read in Igor. I have eventually identified this as related to content of streams folder. There are few streams we are using from there - see list below. But 99% of that folder is not useful and we would save lots of time if we could reduce some of the clutter there. 

I understand this is metadata and potentially useful. If this is automatically generated, then we don't really have control. If this is generated by list, may be some of this can be removed. 

Currently used streams:
	Wave/Z UPDsize=:entry:instrument:bluesky:streams:baseline:terms_USAXS_diode_upd_size:value					
	Wave updG1=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain0_gain:value
	Wave updG2=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain1_gain:value
	Wave updG3=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain2_gain:value
	Wave updG4=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain3_gain:value
	Wave updG5=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain4_gain:value	
	Wave updBkg1=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain0_background:value
	Wave updBkg2=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain1_background:value
	Wave updBkg3=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain2_background:value
	Wave updBkg4=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain3_background:value
	Wave updBkg5=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain4_background:value
	Wave updBkgErr1=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain0_background_error:value
	Wave updBkgErr2=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain1_background_error:value
	Wave updBkgErr3=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain2_background_error:value
	Wave updBkgErr4=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain3_background_error:value
	Wave updBkgErr5=:entry:instrument:bluesky:streams:baseline:upd_autorange_controls_ranges_gain4_background_error:value
	Wave/Z USAXSPinT_I0Counts=:entry:instrument:bluesky:streams:baseline:terms_USAXS_transmission_I0_counts:value
	Wave/Z USAXSPinT_I0Gain=:entry:instrument:bluesky:streams:baseline:terms_USAXS_transmission_I0_gain:value
	Wave/Z USAXSPinT_AyPosition=:entry:instrument:bluesky:streams:baseline:terms_USAXS_transmission_ay:value
	Wave/Z USAXSPinT_pinCounts=:entry:instrument:bluesky:streams:baseline:terms_USAXS_transmission_diode_counts:value
	Wave/Z USAXSPinT_pinGain=:entry:instrument:bluesky:streams:baseline:terms_USAXS_transmission_diode_gain:value
	Wave/Z USAXSPinT_Time= :entry:instrument:bluesky:streams:baseline:terms_USAXS_transmission_count_time:value

prjemian, 2020-11-18:
These are all *monitor* streams (streams that record a value when a CA monitor is received on an `EpicsSignal`.).  Monitor streams can be controlled at various levels.  We can refactor our plans outside of normal beam time.

prjemian, 2023-04-11:
@ilavsky:  For APS-U commissioning, you could recommend which monitored streams are useful and which are not.  What evidence do you have to support this statement?

> Steps scans in BS generate relatively larger files which take surprisingly long time to read in Igor. I have eventually identified this as related to content of streams folder.