APS-USAXS/usaxs-bluesky-ended-2023 #422: RE Exception due to `terms` baseline

prjemian, 2020-11-11:
Using `b = sd.baseline.pop(); RE(bp.count([noisy_det]))` to identify problem baseline object.  Repeating this until no error.  Problem baseline is:

```
In [12]: type(b)
Out[12]: instrument.devices.general_terms.GeneralParameters
```

This is our defined object `terms`, the Device with all the USAXS terms.

_Originally posted by @prjemian in https://github.com/APS-USAXS/ipython-usaxs/issues/404#issuecomment-725713565_

prjemian, 2020-11-11:
```
/home/beams11/USAXS/Documents/projects/ophyd/ophyd/signal.py in describe(self)
   1221         desc = dict(
   1222             source='PV:{}'.format(self._read_pvname),
-> 1223             dtype=data_type(val),
   1224             shape=data_shape(val),
   1225             units=self._metadata['units'],

/home/beams11/USAXS/Documents/projects/ophyd/ophyd/utils/epics_pvs.py in data_type(val)
    320             return json_type
    321 
--> 322     raise ValueError(
    323         f'Cannot determine the appropriate bluesky-friendly data type for '
    324         f'value {val} of Python type {type(val)}. '

ValueError: Cannot determine the appropriate bluesky-friendly data type for value <epics.dbr.c_ubyte_Array_1 object at 0x7fe3c16dc6c0> of Python type <class 'epics.dbr.c_ubyte_Array_1'>. Supported types include: int, float, str, and iterables such as list, tuple, np.ndarray, and so on.
```

prjemian, 2020-11-11:
This is the problem:

```
In [16]: terms.Imaging.title
Out[16]: EpicsSignal(read_pv='9idcLAX:USAXS_Img:ExperimentTitle', name='terms_Imaging_title', parent='terms_Imaging', value=<epics.dbr.c_ubyte_Array_1 object at 0x7ff8d09f9dc0>, timestamp=631152000.0, auto_monitor=True, string=True, write_pv='9idcLAX:USAXS_Img:ExperimentTitle', limits=False, put_complete=False)
```

prjemian, 2020-11-11:
The problem is that `terms.Imaging.title` gets string data from a waveform of characters.  Of type `str` and that is listed in the `data_type()` code (that raised the exception) amongst `bad_iterables = (str, bytes, dict)`.

jilavsky, 2020-11-12:
I am confused. This PV is defined in USAXSterms.db as FTVL "CHAR",  which is same as 9idcLAX:sampleTitle. Length (NELM) of 256 CHAR.

prjemian, 2020-11-12:
Don't be confused. The error is raised by python code in ophyd.  I'm
suspecting it's the branch with some new code to be tested.  And there's a
problem there.  Clear?

prjemian, 2020-11-12:
The root cause is that the PV in question, `9idcLAX:USAXS_Img:ExperimentTitle`, has not been set with a value yet.  Once a text string has been put to this PV, the problem is gone.

1. The ophyd code needs to be updated to handle an unintialized waveform string
1. The `9idcLAX:USAXS_Img:ExperimentTitle` should be initialized with an empty string.  (Set the `VAL` field in the `.db` file to "")

@jilavsky , @kmpeters : Can either of you modify the `.db` file for this PV?

prjemian, 2020-11-12:
I set the PV's value to "" which resolves this specific error for now.  The proper fix is item 2 from above, initialize the PV with a value.  How and when that work gets done is another issue.  Closing this one.

closed 2020-11-12 by prjemian