APS-USAXS/usaxs-bluesky-ended-2023 #433: PI_Off stops everything

jilavsky, 2020-11-18:
When I put PI_Off command in the file (for example so it is not spinning for preUSAXStune), it stops all collection of that usaxs.mac file.

prjemian, 2020-11-21:
Might this be related to #435?  Let's test before changing anything.  This plan has worked before.

prjemian, 2020-11-23:
When it *stops all collection*, is there a python exception?

jilavsky, 2020-11-24:
No it simply skipped rest of the command file. Need to test again. Let's plan some test times. I'll talk to you tomorrow sometimes, situation is bit complicated now.

prjemian, 2020-11-25:
Seems obvious that the run stops with a `FailedStatus` exception _because `PI_Off()` uses the motor `.STOP` field while the motor is running_.

    FailedStatus: Status(obj=EpicsSignal(read_pv='9idcPI:c867:c0:m1.STOP', name='pi_c867_motor_stop', parent='pi_c867', value=0, timestamp=1606321111.020195, auto_monitor=True, string=False, write_pv='9idcPI:c867:c0:m1.STOP', limits=False, put_complete=False), done=True, success=False)

<details>

```python
In [3]: RE(run_command_file("test_bluesky.mac"))
I Wed-10:17:19 - Command file: test_bluesky.mac
====== ====== ==========
line # action parameters
====== ====== ==========
3      PI_Off           
4      PI_onF           
5      PI_Off           
====== ====== ==========

I Wed-10:17:19 - memory report: pmem(rss=268709888, vms=3224010752, shared=46084096, text=2506752, lib=0, data=2702229504, dirty=0)
I Wed-10:17:19 - Measuring background for: PD_USAXS
                                                                                                                                                                                                                 
I Wed-10:17:21 - PD_USAXS range=4 gain=1e12 bkg=1.0 +/- 0.0
I Wed-10:17:21 - I0_USAXS range=4 gain=1e9 bkg=263.2 +/- 0.7483314773547882
I Wed-10:17:21 - I00_USAXS range=4 gain=1e9 bkg=415.8 +/- 0.4
                                                                                                                                                                                                                 
I Wed-10:17:22 - PD_USAXS range=3 gain=1e10 bkg=1.0 +/- 0.0
I Wed-10:17:22 - I0_USAXS range=3 gain=1e8 bkg=254.4 +/- 0.4898979485566356
I Wed-10:17:22 - I00_USAXS range=3 gain=1e8 bkg=374.8 +/- 0.4
                                                                                                                                                                                                                 
I Wed-10:17:24 - PD_USAXS range=2 gain=1e8 bkg=1.0 +/- 0.0
I Wed-10:17:24 - I0_USAXS range=2 gain=1e7 bkg=253.6 +/- 0.4898979485566356
I Wed-10:17:24 - I00_USAXS range=2 gain=1e7 bkg=370.6 +/- 0.48989794855663565
                                                                                                                                                                                                                 
I Wed-10:17:25 - PD_USAXS range=1 gain=1e6 bkg=1.0 +/- 0.0
I Wed-10:17:25 - I0_USAXS range=1 gain=1e6 bkg=254.6 +/- 0.4898979485566356
I Wed-10:17:25 - I00_USAXS range=1 gain=1e6 bkg=370.4 +/- 0.48989794855663565
                                                                                                                                                                                                                 
I Wed-10:17:27 - PD_USAXS range=0 gain=1e4 bkg=1.0 +/- 0.0
I Wed-10:17:27 - I0_USAXS range=0 gain=1e5 bkg=255.0 +/- 0.0
I Wed-10:17:27 - I00_USAXS range=0 gain=1e5 bkg=370.2 +/- 0.4
I Wed-10:17:27 - TR diode range=0 gain=1e5 bkg=1.0 +/- 0.0
                                                                                                                                                                                                                 
                                                                                                                                                                                                                 
                                                                                                                                                                                                                 
I Wed-10:17:31 - Found UsaxsSaxsMode = SAXS in beam
I Wed-10:17:31 - Moving to proper USAXS mode
                                                                                                                                                                                                                 
I Wed-10:17:37 - Removed SAXS from beam position
I Wed-10:17:37 - Moving to USAXS modeeam
                                                                                                                                                                                                                 
                                                                                                                                                                                                                 
                                                                                                                                                                                                                 
                                                                                                                                                                                                                 
I Wed-10:17:50 - tuning axis: m_stage_r                                                                                                                                                                          
                                                                                                                                                                                                                 
D Wed-10:17:50 - gain=100000000.0  rate: 20.0  max: 950000  converged=[False, True]
D Wed-10:17:50 - gain=10000000.0  rate: 1300.0  max: 950000  converged=[False, True, False, True]
[... cut ...]
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

FailedStatus: Status(obj=EpicsSignal(read_pv='9idcPI:c867:c0:m1.STOP', name='pi_c867_motor_stop', parent='pi_c867', value=0, timestamp=1606321111.020195, auto_monitor=True, string=False, write_pv='9idcPI:c867:c0:m1.STOP', limits=False, put_complete=False), done=True, success=False)

In [4]: 

```

</details>

NOTE: Also, I'll revise the test procedure so that we don't send emails during the testing.

prjemian, 2020-11-25:
Confirmed (with a simpler plan).  That is the problem:

<details>

```python
In [7]: %run -i ./test_bluesky.py
I Wed-10:35:38 - ./test_bluesky.py

In [8]: test_plan??
Signature: test_plan()
Docstring: <no docstring>
Source:   
def test_plan():
    logger.info("before first PI_Off()")
    yield from PI_Off()
    logger.info("after first PI_Off()")

    logger.info("before PI_onF()")
    yield from PI_onF()
    logger.info("before PI_onF()")

    logger.info("before sleep(5)")
    yield from bps.sleep(5)
    logger.info("before sleep(5)")

    logger.info("before last PI_Off()")
    yield from PI_Off()
    logger.info("after last PI_Off()")
File:      /share1/USAXS_data/2020-11/test_bluesky.py
Type:      function

In [9]: RE(test_plan())
I Wed-10:36:24 - before first PI_Off()
I Wed-10:36:24 - after first PI_Off()
I Wed-10:36:24 - before PI_onF()
I Wed-10:36:25 - before PI_onF()
I Wed-10:36:25 - before sleep(5)
---------------------------------------------------------------------------
FailedStatus                              Traceback (most recent call last)
/share1/USAXS_data/2020-11/test_bluesky.py in <module>
----> 1 RE(test_plan())

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

[... 96 lines cut ...]
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

FailedStatus: Status(obj=EpicsSignal(read_pv='9idcPI:c867:c0:m1.STOP', name='pi_c867_motor_stop', parent='pi_c867', value=0, timestamp=1606322184.820158, auto_monitor=True, string=False, write_pv='9idcPI:c867:c0:m1.STOP', limits=False, put_complete=False), done=True, success=False)

In [10]: 

```

</details>

The first `PI_Off()` does not fail since the motor is already stopped.  *Only* after the `PI_onF()`, then the call to `PI_Off()` fails (raises a `FailedStatus` exception) because the motor status object on the `.STOP` field ends with `success=False`.

prjemian, 2020-11-25:
Try a local re-definition of the `PI_Off()` plan:

```python
def PI_Off(md=None):
    yield from bps.mv(
        pi_c867.jog_forward, 0,
        pi_c867.jog_reverse, 0,
        timeout=1,
    )
```

This is successful.

```python
In [12]: %run -i ./test_bluesky.py
I Wed-10:46:18 - ./test_bluesky.py

In [13]: PI_Off??
Signature: PI_Off(md=None)
Docstring: <no docstring>
Source:   
def PI_Off(md=None):
    yield from bps.mv(
        pi_c867.jog_forward, 0,
        pi_c867.jog_reverse, 0,
        timeout=1,
    )
File:      /share1/USAXS_data/2020-11/test_bluesky.py
Type:      function

In [14]: RE(test_plan())
I Wed-10:46:30 - before first PI_Off()
I Wed-10:46:30 - after first PI_Off()
I Wed-10:46:30 - before PI_onF()
I Wed-10:46:35 - before PI_onF()
I Wed-10:46:35 - before sleep(5)
I Wed-10:46:40 - before sleep(5)
I Wed-10:46:40 - before last PI_Off()
I Wed-10:46:40 - after last PI_Off()
Out[14]: ()

In [15]: 

```

This *now* matches the implementation in SPEC, which has no such note of failure when STOP is encountered while a motor is running.  Curious coincidence.

commit 81d0bae1 (2020-11-25): fixes #433 as tested
closed 2020-11-25 by prjemian