APS-USAXS/usaxs-bluesky-ended-2023 #434: BlackFly optical image does not seem to be collected.

jilavsky, 2020-11-18:
I tried to add the Blackfly optical images and do not seem to have those where I am expecting them. We need to check this.

prjemian, 2020-11-21:
Interested if solution of #435 will resolve this issue, too.  Evaluate at next beam time before taking further action.

prjemian, 2020-11-24:
May have been caused (and then solved) by the soft IOC which was rebooted.

prjemian, 2020-11-25:
Tested that we *can* take an image (as original issue title asserts): _yes_, see  [bluesky_0412.zip](https://github.com/APS-USAXS/ipython-usaxs/files/5598715/bluesky_0412.zip)

```python
In [10]: RE(blackfly_optical.take_image())
D Wed-12:49:57 - make_filename: /mnt/share1/USAXS_data/testing/bluesky_0412.tif
D Wed-12:49:57 - write_path: /mnt/share1/USAXS_data/testing/
Out[10]: ()                                                                                                                                 
```

prjemian, 2020-11-25:
But, reading more closely:

> not ... *where* I am expecting them

Changing the issue title.

prjemian, 2020-11-25:
To control *where*, need to configure: `blackfly_optical.tiff1.file_path`

prjemian, 2020-11-25:
Currently:

```python
In [11]: blackfly_optical.tiff1.file_path.get()
Out[11]: '/mnt/share1/USAXS_data/testing/'

```

prjemian, 2020-11-25:
The `record_sample_image_on_demand()` plan: https://github.com/APS-USAXS/ipython-usaxs/blob/8407a82273ddaca88a66a84dd3858b7d1ab843ab/profile_bluesky/startup/instrument/plans/sample_imaging.py#L19 controls the file path https://github.com/APS-USAXS/ipython-usaxs/blob/8407a82273ddaca88a66a84dd3858b7d1ab843ab/profile_bluesky/startup/instrument/plans/sample_imaging.py#L37
and takes an image: https://github.com/APS-USAXS/ipython-usaxs/blob/8407a82273ddaca88a66a84dd3858b7d1ab843ab/profile_bluesky/startup/instrument/plans/sample_imaging.py#L46-L52

prjemian, 2020-11-25:
Need to make this adapt to either JPEG or TIFF choice.  Now, it is specific to JPEG.  Waiting on #442 to proceed.

PR #443 (2020-11-25): Control directory for blackfly2 images
fixes #434
closed 2020-11-25 by jilavsky