APS-USAXS/usaxs-bluesky-ended-2023 #435: Step scan does not seem to work?

jilavsky, 2020-11-18:
I run step scans. I got exception 

generator uascan ['703cf260'] (scan num: 168)
W Wed-11:41:37 - Exception during execution of command usaxs.mac:
No conversion path for dtype: dtype('<U1')
I Wed-11:41:37 - waiting for next user command

This is when system was suppose to save data in hdf5 file.

prjemian, 2020-11-18:
root cause: https://github.com/h5py/h5py/issues/289

analysis: One of the streams contains a signal that includes an EPICS waveform string that is expressed as a unicode array (the typical representation).  It must be converted into something that numpy can render into convertible storage for h5py.

Which one?  The diagnostics are poor for this right now.  Need to override locally to gain diagnostics to identify which signal(s).  That will lead to proper corrective action.

prjemian, 2020-11-18:
Need diagnostics which signal and which stream from:

* `apstools.filwriters.write_streams`
* `apstools.filwriters.write_stream_internal`

prjemian, 2020-11-18:
That code is writing the `baseline` stream.

jilavsky, 2020-11-18:
Run scan with following output:

```
generator uascan ['cfe26250'] (scan num: 229)
E Wed-13:43:50 - integer diagnostics_PSS_a_beam_active TypeError(No conversion path for dtype: dtype('<U7')) ['BEAM ON', 'BEAM ON']
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
~/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py in write_stream_internal(self, parent, d, subgroup, stream_name, k, v)
    129                 try:
--> 130                     ds = subgroup.create_dataset(key, data=d[item])
    131                     self.add_dataset_attributes(ds, v, k)

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/h5py/_hl/group.py in create_dataset(self, name, shape, dtype, data, **kwds)
    135         with phil:
--> 136             dsid = dataset.make_new_dset(self, shape, dtype, data, **kwds)
    137             dset = dataset.Dataset(dsid)

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/h5py/_hl/dataset.py in make_new_dset(parent, shape, dtype, data, chunks, compression, shuffle, fletcher32, maxshape, compression_opts, fillvalue, scaleoffset, track_times, external, track_order, dcpl)
    117             dtype = numpy.dtype(dtype)
--> 118         tid = h5t.py_create(dtype, logical=1)
    119 

h5py/h5t.pyx in h5py.h5t.py_create()

h5py/h5t.pyx in h5py.h5t.py_create()

h5py/h5t.pyx in h5py.h5t.py_create()

TypeError: No conversion path for dtype: dtype('<U7')

During handling of the above exception, another exception occurred:

UnicodeError                              Traceback (most recent call last)
<ipython-input-2-04b994fdd9b0> in <module>
----> 1 RE(run_command_file("usaxs.mac"))

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 
[... 309 lines cut ...]
   1254 
   1255         md_group = self.write_metadata(bluesky_group)
-> 1256         self.write_streams(bluesky_group)
   1257 
   1258         bluesky_group["uid"] = md_group["run_start_uid"]

/home/beams11/USAXS/Documents/projects/apstools/apstools/filewriters.py in write_streams(self, parent)
   1519                     self.write_stream_external(parent, d, subgroup, stream_name, k, v)
   1520                 else:
-> 1521                     self.write_stream_internal(parent, d, subgroup, stream_name, k, v)
   1522 
   1523                 t = np.array(v["time"])

~/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py in write_stream_internal(self, parent, d, subgroup, stream_name, k, v)
    132                     ds.attrs["target"] = ds.name
    133                 except TypeError as exc:
--> 134                     raise UnicodeError(
    135                         (
    136                             "Could not write '%s' to h5py from baseline:"

UnicodeError: ("Could not write '%s' to h5py from baseline: k=%s, v=%s, key=%s", 'BEAM ON', 'diagnostics_PSS_a_beam_active', {'source': 'PV:PA:09ID:A_BEAM_ACTIVE.VAL', 'dtype': 'integer', 'shape': [], 'units': None, 'lower_ctrl_limit': None, 'upper_ctrl_limit': None, 'precision': 0, 'object_name': 'diagnostics', 'data': ['BEAM ON', 'BEAM ON'], 'time': [1605727866.500598, 1605727866.500598], 'external': False}, 'value_start')

In [3]: 

```

jilavsky, 2020-11-18:
Better diagnostics:

```
generator uascan ['511514c9'] (scan num: 241)                                                                                                                                                                    
E Wed-14:04:46 - integer diagnostics_PSS_a_beam_active TypeError(No conversion path for dtype: dtype('<U7')) ['BEAM ON', 'BEAM ON']
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
~/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py in write_stream_internal(self, parent, d, subgroup, stream_name, k, v)
    129                 try:
--> 130                     ds = subgroup.create_dataset(key, data=d[item])
    131                     self.add_dataset_attributes(ds, v, k)

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/h5py/_hl/group.py in create_dataset(self, name, shape, dtype, data, **kwds)
    135         with phil:
--> 136             dsid = dataset.make_new_dset(self, shape, dtype, data, **kwds)
    137             dset = dataset.Dataset(dsid)

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/h5py/_hl/dataset.py in make_new_dset(parent, shape, dtype, data, chunks, compression, shuffle, fletcher32, maxshape, compression_opts, fillvalue, scaleoffset, track_times, external, track_order, dcpl)
    117             dtype = numpy.dtype(dtype)
--> 118         tid = h5t.py_create(dtype, logical=1)
    119 

h5py/h5t.pyx in h5py.h5t.py_create()

h5py/h5t.pyx in h5py.h5t.py_create()

h5py/h5t.pyx in h5py.h5t.py_create()

TypeError: No conversion path for dtype: dtype('<U7')

During handling of the above exception, another exception occurred:

UnicodeError                              Traceback (most recent call last)
<ipython-input-5-04b994fdd9b0> in <module>
----> 1 RE(run_command_file("usaxs.mac"))

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 
[... 316 lines cut ...]
   1254 
   1255         md_group = self.write_metadata(bluesky_group)
-> 1256         self.write_streams(bluesky_group)
   1257 
   1258         bluesky_group["uid"] = md_group["run_start_uid"]

/home/beams11/USAXS/Documents/projects/apstools/apstools/filewriters.py in write_streams(self, parent)
   1519                     self.write_stream_external(parent, d, subgroup, stream_name, k, v)
   1520                 else:
-> 1521                     self.write_stream_internal(parent, d, subgroup, stream_name, k, v)
   1522 
   1523                 t = np.array(v["time"])

~/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py in write_stream_internal(self, parent, d, subgroup, stream_name, k, v)
    132                     ds.attrs["target"] = ds.name
    133                 except TypeError as exc:
--> 134                     raise UnicodeError(
    135                         (
    136                             f"Could not write '{d[item]' to h5py from baseline:"

UnicodeError: ("Could not write '%s' to h5py from baseline: k=%s, v=%s, key=%s", 'BEAM ON', 'diagnostics_PSS_a_beam_active', {'source': 'PV:PA:09ID:A_BEAM_ACTIVE.VAL', 'dtype': 'integer', 'shape': [], 'units': None, 'lower_ctrl_limit': None, 'upper_ctrl_limit': None, 'precision': 0, 'object_name': 'diagnostics', 'data': ['BEAM ON', 'BEAM ON'], 'time': [1605727866.500598, 1605727866.500598], 'external': False}, 'value_start')

In [6]: 

```

prjemian, 2020-11-18:
Diagnostic:

```
~/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py in write_stream_internal(self, parent, d, subgroup, stream_name, k, v)
    132                     ds.attrs["target"] = ds.name
    133                 except TypeError as exc:
--> 134                     raise UnicodeError(
    135                         f"Could not write '{d[item]}' to h5py from baseline:"
    136                         f" k={k}, v={v}, key={key}"

UnicodeError: Could not write 'X' to h5py from baseline: k=sample_data_magnetic_field_dir, v={'source': 'PV:9idcSample:MagneticFieldDir', 'dtype': 'integer', 'shape': [], 'units': None, 'lower_ctrl_limit': None, 'upper_ctrl_limit': None, 'precision': 0, 'object_name': 'sample_data', 'data': ['X', 'X'], 'time': [631152000.0, 631152000.0], 'external': False}, key=value_start
No conversion path for dtype: dtype('<U1')

```

This is not consistent with:

```
In [41]: h.table("baseline")["sample_data_magnetic_field_dir"].to_numpy()
Out[41]: array(['X', 'X'], dtype=object)

In [42]: h.table("baseline")["sample_data_magnetic_field_dir"].to_numpy()[0]
Out[42]: 'X'

In [43]: type(h.table("baseline")["sample_data_magnetic_field_dir"].to_numpy()[0])
Out[43]: str

```

So, why is `'dtype': 'integer',`?  Check the descriptor document.

prjemian, 2020-11-18:
In `devices.sample_data`, it is defined: https://github.com/APS-USAXS/ipython-usaxs/blob/3f50f85d7fe2b907389f4caf38e88a8377053509/profile_bluesky/startup/instrument/devices/sample_data.py#L32

Yet:

```
In [45]: sample_data.magnetic_field_dir.describe()
Out[45]: 
{'sample_data_magnetic_field_dir': {'source': 'PV:9idcSample:MagneticFieldDir',
  'dtype': 'integer',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('X', 'Y', 'Z')}}

```

prjemian, 2020-11-18:
And there's your answer (learn something new every day).

```python
In [50]: d = EpicsSignal("9idcSample:MagneticFieldDir", name="d")

In [51]: d.describe()
Out[51]: 
{'d': {'source': 'PV:9idcSample:MagneticFieldDir',
  'dtype': 'integer',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('X', 'Y', 'Z')}}

In [52]: d.get()
Out[52]: 0

In [53]: d.put("X")

In [54]: d.get()
Out[54]: 0

In [55]: d.put("Y")

In [56]: d.get()
Out[56]: 1

In [57]: d.put("X")

In [58]: 

```

### answer

This is a `mbbo` record and the `EpicsSignal` for it cannot handle the `string=True` attribute properly.  It **still** sets the dtype to `integer`, even if it was rendering a string.  Looks like an ophyd bug to me.

prjemian, 2020-11-18:
Can still set this signal by string value (shown above) but can't expect `.get()` to return a string result unless using the `as_string=True` kwarg:

```python
In [7]: d = EpicsSignal("9idcSample:StressFieldDir", name="d")

In [8]: d.get()
Out[8]: 0

In [9]: d.get(as_string=True)
Out[9]: 'X'

```

prjemian, 2020-11-18:
Odd.  Testing with same device construction in two separate windows (one is our bluesky ipython console, the other is a new terminal window with just ipython, not full instrument setup).  Both are in same conda environment so ophyd version is identical.

```python
In [1]: from ophyd import EpicsSignal

In [2]: d = EpicsSignal("S:DesiredMode", name="d", string=True)

In [3]: import ophyd
```

In our bluesky ipython console, this diagnostic:

```python
In [10]: d.as_string
Out[10]: True

In [11]: d.describe()
Out[11]: 
{'d': {'source': 'PV:S:DesiredMode',
  'dtype': 'integer',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('State Unknown',
   'USER OPERATIONS',
   'Bm Ln Studies',
   'INJ Studies',
   'ASD Studies',
   'NO BEAM',
   'MAINTENANCE')}}

In [13]: ophyd.__version__
Out[13]: '1.5.4.post44+gc91317f'

```

In separate ipython-only console, this diagnostic:

```
In [13]: d.describe()["d"]["dtype"]
Out[13]: 'string'

In [14]: d.describe()
Out[14]: 
{'d': {'source': 'PV:S:DesiredMode',
  'dtype': 'string',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('State Unknown',
   'USER OPERATIONS',
   'Bm Ln Studies',
   'INJ Studies',
   'ASD Studies',
   'NO BEAM',
   'MAINTENANCE')}}

In [16]: ophyd.__version__
Out[16]: '1.5.4.post44+gc91317f'

```

prjemian, 2020-11-18:
Found the reason for the difference, in the ipython-only console, we did not call `EpicsSignalBase.set_defaults()`.  Here is a new session where we call that the same way we started the bluesky ipython (instrument) session:

```python
In [1]: from ophyd.signal import EpicsSignalBase

In [5]: EpicsSignalBase.set_defaults(auto_monitor=True, timeout=60, write_timeout=60, connection_timeout=1)

In [6]: from ophyd import EpicsSignal

In [7]: d = EpicsSignal("S:DesiredMode", name="d", string=True)

In [8]: d.as_string
Out[8]: True

In [9]: d.describe()
Out[9]: 
{'d': {'source': 'PV:S:DesiredMode',
  'dtype': 'integer',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('State Unknown',
   'USER OPERATIONS',
   'Bm Ln Studies',
   'INJ Studies',
   'ASD Studies',
   'NO BEAM',
   'MAINTENANCE')}}

In [10]: 

```

And the wrong dtype is obtained: ` 'dtype': 'integer',`

prjemian, 2020-11-18:
New ophyd bug when using `EpicsSignalBase.set_defaults()`.  It's reproducible with *any* PV that has an enumeration, such as every record's `.SCAN` field or a `mbbo` record instance.  Here, we use `S:DesiredMode.SCAN` to show the bug.
 
### buggy

```python
In [1]: from ophyd.signal import EpicsSignalBase

In [2]: EpicsSignalBase.set_defaults(auto_monitor=True, timeout=60, write_timeout=60, connection_timeout=1)

In [3]: from ophyd import EpicsSignal

In [4]: d = EpicsSignal("S:DesiredMode.SCAN", name="d", string=True)

In [5]: d.describe()
Out[5]: 
{'d': {'source': 'PV:S:DesiredMode.SCAN',
  'dtype': 'integer',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('Passive',
   'Event',
   'I/O Intr',
   '10 second',
   '5 second',
   '2 second',
   '1 second',
   '.5 second',
   '.2 second',
   '.1 second')}}

In [6]: d = EpicsSignal("S:DesiredMode.SCAN", name="d")

In [7]: d.describe()
Out[7]: 
{'d': {'source': 'PV:S:DesiredMode.SCAN',
  'dtype': 'integer',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('Passive',
   'Event',
   'I/O Intr',
   '10 second',
   '5 second',
   '2 second',
   '1 second',
   '.5 second',
   '.2 second',
   '.1 second')}}
```

### not buggy

```python
In [1]: from ophyd import EpicsSignal

In [2]: d = EpicsSignal("S:DesiredMode.SCAN", name="d")

In [3]: d.describe()
Out[3]: 
{'d': {'source': 'PV:S:DesiredMode.SCAN',
  'dtype': 'integer',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('Passive',
   'Event',
   'I/O Intr',
   '10 second',
   '5 second',
   '2 second',
   '1 second',
   '.5 second',
   '.2 second',
   '.1 second')}}

In [4]: d = EpicsSignal("S:DesiredMode.SCAN", name="d", string=True)

In [5]: d.describe()
Out[5]: 
{'d': {'source': 'PV:S:DesiredMode.SCAN',
  'dtype': 'string',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('Passive',
   'Event',
   'I/O Intr',
   '10 second',
   '5 second',
   '2 second',
   '1 second',
   '.5 second',
   '.2 second',
   '.1 second')}}

```

started each session with this setup:

```
(/share1/bluesky/conda_envs/bluesky_2020_9) bash-4.2$ ipython
Python 3.8.2 (default, Mar 26 2020, 15:53:00) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.1 -- An enhanced Interactive Python. Type '?' for help.
```

prjemian, 2020-11-19:
Changed *In Progress* column to *Waiting input* for PR https://github.com/bluesky/ophyd/pull/946.

prjemian, 2020-11-20:
@jilavsky Found the problem and have a PR with the fix.

When using `auto_monitor=True`, the local cached value is stored in `EpicsSignal._readback`.  The callback routine (`EpicsSignal._read_changed()`) gets the new value from the CA monitor update.  What's missing is when the `string=True` attribute has been added to the `EpicsSignal()`, then `EpicsSignal._read_changed()` should handle the conversion from raw to string.  Fortunately, PyEpics provides that string value in the `char_value` kwarg.

prjemian, 2020-11-21:
Installed [ophyd branch with upstream fix for this problem](https://github.com/bluesky/ophyd/pull/946/files).  Confirmed that it resolves the mismatch between the `dtype` and the value (compare with [above](https://github.com/APS-USAXS/ipython-usaxs/issues/435#issuecomment-729960305)):

```python
In [1]: sample_data.magnetic_field_dir.describe()
Out[1]: 
{'sample_data_magnetic_field_dir': {'source': 'PV:9idcSample:MagneticFieldDir',
  'dtype': 'string',
  'shape': [],
  'units': None,
  'lower_ctrl_limit': None,
  'upper_ctrl_limit': None,
  'enum_strs': ('X', 'Y', 'Z')}}
```

prjemian, 2020-11-23:
@jilavsky tested this last night with the `preUSAXStune()` plan.  It worked.

closed 2020-11-23 by prjemian