APS-USAXS/usaxs-bluesky-ended-2023 #439: Change from jpeg to tiff writer

jilavsky, 2020-11-23:
For BlackFly2 optical camera, collecting images for users on request. 
jpeg has 8 bit limitation, so we need to change to tiff file writer. Should be trivial.

prjemian, 2020-11-23:
I anticipate a couple small bumps in switching the ophyd callback from jpeg
to Tiff.  Similar to those I had with the jpeg callback in the first place.
Nothing big.

prjemian, 2020-11-25:
Also, see #434 for related matter.

prjemian, 2020-11-25:
```python
RE(blackfly_optical.take_image())
```

We have an 8-bit image: [bluesky_0405.zip](https://github.com/APS-USAXS/ipython-usaxs/files/5598497/bluesky_0405.zip)

prjemian, 2020-11-25:
FYI: `evince` can be used from linux to view these tiff files:

```
evince /share1/USAXS_data/testing/bluesky_0410.tif
```

prjemian, 2020-11-25:
```python
%mov blackfly_optical.cam.pixel_format Mono16
```

This will produce 16-bit B/W images.

BUT, the detector is a color camera.  How to get color out of this beast?

prjemian, 2020-11-25:
These are the possibilities:

```python
In [1]: blackfly_optical.cam.pixel_format.enum_strs
Out[1]: 
('Mono8',
 'YUV411',
 'YUV422',
 'YUV444',
 'RGB8',
 'Mono16',
 'Raw8',
 'Raw16',
 'Mono12',
 'Raw12',
 'BGR16')
```

prjemian, 2020-11-25:
```python
In [4]: blackfly_optical.cam.pixel_format.put("BGR16")

In [5]: from instrument.devices.blackfly import AD_prime_plugin

In [6]: AD_prime_plugin(blackfly_optical, blackfly_optical.tiff1)

In [8]: blackfly_optical.tiff1.full_file_name.get()
Out[8]: '/mnt/share1/USAXS_data/testing/bluesky_0411.tif'

In [9]: !evince /share1/USAXS_data/testing/bluesky_0411.tif

```

Image is color with 8-bit per color (24-bit): [bluesky_0411.zip](https://github.com/APS-USAXS/ipython-usaxs/files/5598690/bluesky_0411.zip)

prjemian, 2020-11-25:
This is working.

prjemian, 2020-11-25:
This will be closed by PR #442

PR #442 (2020-11-25): blackfly2 should take TIFF images
fixes #439
closed 2020-11-25 by jilavsky