APS-USAXS/usaxs-bluesky-ended-2023 #445: SAXS detector was not primed when started collecting data

jilavsky, 2020-11-27:
I restarted iocs for SAXS and WAXS to prevent crash. 

Now when I tried to collect data, I get priming error:

 inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1423                         # exceptions (coming in via throw) can be
   1424                         # raised
-> 1425                         new_response = await coro(msg)
   1426 
   1427                     # special case `CancelledError` and let the outer

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _stage(self, msg)
   2166         if not hasattr(obj, 'stage'):
   2167             return []
-> 2168         result = obj.stage()
   2169         self._staged.add(obj)  # add first in case of failure below
   2170         await self._reset_checkpoint_state_coro()

/home/beams11/USAXS/Documents/projects/ophyd/ophyd/areadetector/trigger_mixins.py in stage(self)
    121     def stage(self):
    122         self._acquisition_signal.subscribe(self._acquire_changed)
--> 123         super().stage()
    124 
    125     def unstage(self):

/home/beams11/USAXS/Documents/projects/ophyd/ophyd/areadetector/base.py in stage(self, *args, **kwargs)
    287 
    288     def stage(self, *args, **kwargs):
--> 289         ret = super().stage(*args, **kwargs)
    290         try:
    291             self.validate_asyn_ports()

/home/beams11/USAXS/Documents/projects/ophyd/ophyd/device.py in stage(self)
    557                 device = getattr(self, attr)
    558                 if hasattr(device, 'stage'):
--> 559                     device.stage()
    560                     devices_staged.append(device)
    561         except Exception:

/home/beams11/USAXS/Documents/projects/ophyd/ophyd/areadetector/plugins.py in stage(self)
    923     def stage(self):
    924         if np.array(self.array_size.get()).sum() == 0:
--> 925             raise UnprimedPlugin(f"The plugin {self.dotted_name} on the "
    926                                  f"area detector with name {self.root.name} "
    927                                  f"has not been primed.")

UnprimedPlugin: The plugin hdf1 on the area detector with name saxs_det has not been primed.

jilavsky, 2020-11-27:
Run ended, ctrl-C did not abort, enter key did return command line. Not sure what to do...

jilavsky, 2020-11-27:
I have collected from epics image in each SAXS and WAXS detectcor. Assume this will fix it?

prjemian, 2021-01-11:
Reviewing how to test that plugin is primed, these items must match:

* array size
* data type
* color mode

```python
cam.array_size.get() == plugin.array_size.get()
cam.data_type.get(as_string=True) == plugin.data_type.get(as_string=True)
cam.color_mode.get(as_string=True) == plugin.color_mode.get(as_string=True)
```

prjemian, 2021-01-25:
dealt with this in #467

closed 2021-01-25 by prjemian