APS-USAXS/usaxs-bluesky-ended-2023 #452: Add user info into sample name

jilavsky, 2020-11-28:
We need function which modifies sampleTtitle at appropriate time and adds to it needed experiment information. 
E.g., temperature, sx, time. 
I suggest that : USAXSscan, SAXS, WAXS run this modifier function when called with SampleTitile and change that per user request. DO this at the very top of the routine and have ability for user to modify the routine on fly.

prjemian, 2021-01-25:
where is `sampleTitle` in the code?

```
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ git grep sampleTitle
profile_bluesky/startup/instrument/devices/user_data.py:    sample_title = Component(EpicsSampleNameDevice, "9idcLAX:sampleTitle", string=
profile_bluesky/startup/usaxs_support/saveFlyData.xml:                <PV label="title" pvname="9idcLAX:USAXS:sampleTitle" string="true" /
profile_bluesky/startup/usaxs_support/saveFlyData.xml:                    <PV label="name" pvname="9idcLAX:USAXS:sampleTitle" string="true
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ git grep sample_title
profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py:        nxentry["sample/name"] = self.get_sample_title()
profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py:    def get_sample_title(self):
profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py:        return self.get_stream_link("user_data_sample_title")
profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py:                f"{cleanupText(user_data.sample_title.get())}"
profile_bluesky/startup/instrument/devices/user_data.py:        >>> user_data.sample_title.register_handler(handler)
profile_bluesky/startup/instrument/devices/user_data.py:        >>> RE(bps.mv(user_data.sample_title, "Glassy Carbon"))
profile_bluesky/startup/instrument/devices/user_data.py:        >>> user_data.sample_title.get()
profile_bluesky/startup/instrument/devices/user_data.py:    sample_title = Component(EpicsSampleNameDevice, "9idcLAX:sampleTitle", string=
profile_bluesky/startup/instrument/plans/scans.py:        user_data.sample_title, scan_title,
profile_bluesky/startup/instrument/plans/scans.py:        user_data.sample_title, scan_title,
profile_bluesky/startup/instrument/plans/scans.py:        user_data.sample_title, scan_title,
profile_bluesky/startup/instrument/plans/scans.py:        user_data.sample_title, scan_title,
profile_bluesky/startup/instrument/plans/tune_guard_slits.py:        user_data.sample_title, title,
```

prjemian, 2021-01-25:
So, basically, you want it to work like this?

```python
In [1]: getSampleTitle("a long walk in the country")
Out[1]: 'a long walk in the country'

In [2]: def ford(title):
   ...:     return f"black: scan{RE.md['scan_id']}"
   ...: 

In [3]: ford("What color do you want your car?")
Out[3]: 'black: scan36'

In [4]: setSampleTitleFunction(ford)

In [5]: getSampleTitle("a long walk in the country")
Out[5]: 'black: scan36'

In [6]: resetSampleTitleFunction()
```

prjemian, 2021-01-25:
Or, more typically:

```python
In [8]: def myTitleFunction(title):
   ...:     return f"{title} I={aps.current.get():.2f}"
   ...: 

In [9]: setSampleTitleFunction(myTitleFunction)

In [10]: getSampleTitle("blank")
Out[10]: 'blank I=102.09'

In [11]: def myTitleFunction(title):
    ...:     return f"{title} I={aps.current.get():.2f} scan={RE.md['scan_id']}"
    ...: 

In [12]: setSampleTitleFunction(myTitleFunction)

In [13]: getSampleTitle("blank")
Out[13]: 'blank I=101.95 scan=36'

In [14]: from instrument.utils.cleanup_text import cleanupText

In [15]: cleanupText(getSampleTitle("blank"))
Out[15]: 'blank_I_102_01_scan_36'

In [16]: resetSampleTitleFunction()

In [17]: getSampleTitle("blank")
Out[17]: 'blank'

```

PR #475 (2021-01-25): Users want to update the sample title with dynamic information
FIX #452
closed 2021-01-25 by jilavsky