APS-USAXS/usaxs-bluesky-ended-2023 #453: SAXS series failed.

jilavsky, 2020-12-01:
Started SAXS series using run_command)file on usaxs.mac, failed. 

```
 Tue-10:22:44 - Area Detector HDF5 file: /share1/USAXS_data/2020-12/12_01_Shoaib/12_01_Shoaib_saxs/SwyS31_0049.hdf
---------------------------------------------------------------------------l/USAXS_data/2020-12/12_01_Shoaib/12_01_Shoaib_saxs/SwyS31_0049.hdf
FailedStatus                              Traceback (most recent call last)
<ipython-input-9-04b994fdd9b0> in <module>
----> 1 RE(run_command_file("usaxs.mac"))

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
     79             except StopIteration as e:
     80                 # discard the exhausted generator

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in monitor_during_wrapper(plan, signals)
    801     plan1 = plan_mutator(plan, insert_after_open)
    802     plan2 = plan_mutator(plan1, insert_before_close)
--> 803     return (yield from plan2)
    804 
    805 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
     79             except StopIteration as e:
[... 93 lines cut ...]
/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    193         try:
    194             # yield out the 'current message' and collect the return
--> 195             inner_ret = yield msg
    196         except GeneratorExit:
    197             # special case GeneratorExit.  We must clean up all of our plans

FailedStatus: Status(obj=EpicsSignal(read_pv='9idcLAX:state', name='user_data_state', parent='user_data', value='Moving USAXS to SAXS mode', timestamp=1606839764.525845, auto_monitor=True, string=True, write_pv='9idcLAX:state', limits=False, put_complete=False), done=True, success=False)

In [10]: 
```

jilavsky, 2020-12-01:
`FailedStatus` object from:

```
--> 108         yield from bps.abs_set(self.state, msg, wait=confirm)
```

Default for `confirm` is `True`.  We don't care about waiting here so flip the default.  No caller (at present) changes this kwarg value.

jilavsky, 2020-12-01:
Nope.  Failed with `another set call in progress` `RuntimeError` exception.

jilavsky, 2020-12-01:
Original failure happened during SAXS collection after a couple dozen successful previous collections in the same run.  Just need to report the situation (with `try..except`) and move on.

jilavsky, 2020-12-01:
Note also the default write_timeout: https://github.com/APS-USAXS/ipython-usaxs/blob/c8ecbdd0544dd2d9e6b1ea74e37a1dee6a552036/profile_bluesky/startup/instrument/framework/initialize.py

**_Could_** decrease that to 0.1 seconds (since we do not care about it anyway).  If so, then might get lots more messages from the logger.  If and when those messages become a nuisance, then replace this call to `logger.warning()` with a `pass`: https://github.com/APS-USAXS/ipython-usaxs/blob/c8ecbdd0544dd2d9e6b1ea74e37a1dee6a552036/profile_bluesky/startup/instrument/devices/user_data.py#L110-L114

prjemian, 2020-12-01:
We've run through this code now many times and no problem reports.  Closing.

closed 2020-12-01 by prjemian