APS-USAXS/usaxs-bluesky-ended-2023 #456: Running mode_Radiography with doors open hangs instrument

jilavsky, 2020-12-02:
This is not usual, we run mode change while searching hutch. mode_radiography tries to open shutter, and does nto give up after some time out...

```
In [13]: RE(mode_Radiography())                                                                                                                                                                                                             
I Tue-19:04:29 - Found UsaxsSaxsMode = SAXS in beam
I Tue-19:04:29 - Moving to proper USAXS mode
                                                                                                                                                                                                                 
I Tue-19:04:36 - Removed SAXS from beam position
                                                                                                                                                                                                             
                                                                                                                                                                                                                 
mono_shutter [In progress. No progress bar available.]                                                                                                                                                           
Exception in thread Thread-147938:                                                                                                                                                                               
Traceback (most recent call last):                                                                                                                                                                               
  File "/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/threading.py", line 932, in _bootstrap_inner                                                                                                     
    self.run() - Preparing for USAXS mode ... please wait ...
  File "/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/Documents/projects/apstools/apstools/devices.py", line 462, in move_it
    self.open()
  File "/home/beams11/USAXS/Documents/projects/apstools/apstools/devices.py", line 815, in open
    self.wait_for_state(
  File "/home/beams11/USAXS/Documents/projects/apstools/apstools/devices.py", line 807, in wait_for_state
    raise TimeoutError(msg)
TimeoutError: Timeout (10 s) waiting for shutter state to reach a value in [1, 'O', 'N']

```
It should give up and finish setting up without opening the shutter. Some reasonable timeout is fine (10sec?).

prjemian, 2020-12-02:
See #303, that might have helped us here.

prjemian, 2020-12-02:
Also had to unjam the shutter by `mono_shutter.busy.put(False)`

prjemian, 2021-01-25:
While working on this issue, conflated it with issue #303.  I'll join these two in the PR (TBA).

As to be solved here:

Algorithm:

1. move instrument to Radiography configuration (as before)
2. request mono to open if permitted
3. report if not permitted to open mono
4. check if mono open (when permitted) and report what to do if not open

PR #474 (2021-01-25): PSS signals for doors and C station
* FIX #303: watch more PSS signals, including mono shutter permit
* FIX #456: Don't fail modeRadiography when mono shutter cannot be opened
closed 2021-01-25 by jilavsky