APS-USAXS/usaxs-bluesky-ended-2023 #459: Crash step scans in HDF5 file writing

jilavsky, 2020-12-03:
Crashed in end of step scan while writing HDF5 file. Interestingly, worked at least once and then crashed while running same configuration (different sx, sy). 

```
+-----------+------------+------------+------------+------------+
generator uascan ['c3382237'] (scan num: 30)
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/share1/USAXS_data/2020-12/rocks.py in <module>
----> 1 RE(run_command_file("usaxs.mac"))

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
     79             except StopIteration as e:
     80                 # discard the exhausted generator

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in monitor_during_wrapper(plan, signals)
    801     plan1 = plan_mutator(plan, insert_after_open)
    802     plan2 = plan_mutator(plan1, insert_before_close)
--> 803     return (yield from plan2)
    804 
    805 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
[... 301 lines cut ...]
    115                 d = np.array(d)

/home/beams11/USAXS/Documents/projects/apstools/apstools/filewriters.py in h5string(self, text)
   1123         """Format string for h5py interface."""
   1124         if isinstance(text, (tuple, list)):
-> 1125             return [self.h5string(t) for t in text]
   1126         text = text or ""
   1127         return text.encode("utf8")

/home/beams11/USAXS/Documents/projects/apstools/apstools/filewriters.py in <listcomp>(.0)
   1123         """Format string for h5py interface."""
   1124         if isinstance(text, (tuple, list)):
-> 1125             return [self.h5string(t) for t in text]
   1126         text = text or ""
   1127         return text.encode("utf8")

/home/beams11/USAXS/Documents/projects/apstools/apstools/filewriters.py in h5string(self, text)
   1125             return [self.h5string(t) for t in text]
   1126         text = text or ""
-> 1127         return text.encode("utf8")
   1128 
   1129     def writer(self):

AttributeError: 'int' object has no attribute 'encode'
```

prjemian, 2021-01-14:
This is fixed upstream but not released yet.

prjemian, 2021-01-17:
Release of [apstools 1.4.0](https://github.com/BCDA-APS/apstools/releases/tag/1.4.0) installed 2021-01-17 into instrument's conda environment.

closed 2021-01-17 by prjemian