APS-USAXS/usaxs-bluesky-ended-2023 #464: Restared BS and first SAXS collection crashes

jilavsky, 2020-12-12:
This is not related to priming, this was collecting data over 24 hours in spec. 

```
                                                                                                                                                                              
D Sat-08:40:42 - gain=100000000.0  rate: 813640.0  max: 950000  converged=[True, True]
---------------------------------------------------------------------------                                                                                                   
TimeoutError                              Traceback (most recent call last)
<ipython-input-4-04b994fdd9b0> in <module>
----> 1 RE(run_command_file("usaxs.mac"))

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
     79             except StopIteration as e:
     80                 # discard the exhausted generator

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in monitor_during_wrapper(plan, signals)
    801     plan1 = plan_mutator(plan, insert_after_open)
    802     plan2 = plan_mutator(plan1, insert_before_close)
--> 803     return (yield from plan2)
    804 
    805 

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
[... 165 lines cut ...]
    295         self._locked_key_list = False
    296         self._datum_uids.clear()
--> 297         super().stage()
    298         # Subclasses will assemble resource_kwargs and call
    299         # self._generate_resource(resource_kwrags)

/home/beams11/USAXS/Documents/projects/ophyd/ophyd/device.py in stage(self)
    548                                self.name,
    549                                val, original_vals[sig])
--> 550                 set_and_wait(sig, val)
    551                 # It worked -- now add it to this list of sigs to unstage.
    552                 self._original_vals[sig] = original_vals[sig]

/home/beams11/USAXS/Documents/projects/ophyd/ophyd/utils/epics_pvs.py in set_and_wait(signal, val, poll_time, timeout, rtol, atol)
    263         current_value = signal.get()
    264         if expiration_time is not None and ttime.time() > expiration_time:
--> 265             raise TimeoutError("Attempted to set %r to value %r and timed "
    266                                "out after %r seconds. Current value is %r." %
    267                                (signal, val, timeout, current_value))

TimeoutError: Attempted to set EpicsSignalWithRBV(read_pv='9idFLY2:JPEG1:FileTemplate_RBV', name='blackfly_optical_jpeg1_file_template', parent='blackfly_optical_jpeg1', value='%', timestamp=1607784042.9056838, auto_monitor=True, string=True, write_pv='9idFLY2:JPEG1:FileTemplate', limits=False, put_complete=False) to value '%s%s_%4.4d.jpg' and timed out after 10 seconds. Current value is '%'.

In [5]: 

```

jilavsky, 2020-12-12:
This is a puzzle since:

```python
In [8]: blackfly_optical.jpeg1.file_template._write_timeout
Out[8]: 60
```

jilavsky, 2020-12-12:
Seems to be working now:

```python
In [12]: RE(blackfly_optical.take_image())
Out[12]: ()                                                                                                                                                                   
```

Confirmed by seeing new file

```
(/share1/bluesky/conda_envs/bluesky_2020_9) bash-4.2$ ls -lart /share1/USAXS_data/2020-12/12_12_BeaucageNMR/12_12_BeaucageNMR_saxs/ | tail -5
drwxrwxr-x 5 usaxs usaxs    142 Dec 12 09:17 ..
-rw-rw-r-- 1 usaxs usaxs  55041 Dec 12 10:09 AgBehenateLaB6_0001.jpg
-rw-rw-r-- 1 usaxs usaxs  54922 Dec 12 10:09 AgBehenateLaB6_0002.jpg
drwxrwxr-x 2 usaxs usaxs   4096 Dec 12 10:21 .
-rw-rw-r-- 1 usaxs usaxs  31727 Dec 12 10:21 AgBehenateLaB6_0003.jpg
```

jilavsky, 2020-12-12:
Confirm that is the file we just created:

```python
In [14]: blackfly_optical.jpeg1.full_file_name.get()
```

prjemian, 2020-12-12:
Random PV timeouts are a nuisance, this one particularly since it stopped the plan completely for an activity that is a _nice addition_ but not necessary for scientific data collection.

It is a puzzle why the set operation had a 10s timeout when the default timeout is set to 60s and the ophyd signal `blackfly_optical._write_timeout` value is verified to be exactly that: 60.

As a workaround fix to avoid being stopped by another random timeout to take the sample image, wrap the `yield from blackfly_optical.take_image()` call with a `try..except` handler and log any exception that happens, but do not stop the plan.

prjemian, 2020-12-16:
This will be resolved when the beam line changes are pushed to github after the current APS run cycle ends this week.

commit 10983b2f (2020-12-21): fixes #464 temporary timeouts will be logged but not stop the plan
closed 2020-12-21 by prjemian