APS-USAXS/usaxs-bluesky-ended-2023 #466: Remote operations hanged on another PV

jilavsky, 2020-12-15:
User reported instrument hanged after midnight and he cjecked at 4am...

```
I Tue-00:43:52 - Finished SAXS/WAXS data collection in 30.489188194274902 seconds.
I Tue-00:43:53 - I0 value: 23602502.0
I Tue-00:43:53 - memory report: pmem(rss=2177445888, vms=6931382272, shared=52649984, text=2506752, lib=0, data=6304575488, dirty=0)
I Tue-00:43:53 - file line 117:       saxsExp        105.1      85      10      "P14_Kapton_A1"
                                                                                                                                                                            
                                                                                                                                                                            
I Tue-00:44:01 - Area Detector HDF5 file: /share1/USAXS_data/2020-12/12_14_Pozzo/12_14_Pozzo_saxs/P14_Kapton_A1_0261.hdf
I Tue-00:44:01 - Pilatus computer Area Detector HDF5 file: /mnt/usaxscontrol/USAXS_data/2020-12/12_14_Pozzo/12_14_Pozzo_saxs/P14_Kapton_A1_0261.hdf
                                                                                                                                                                            
                                                                                                                                                                            
I Tue-00:44:03 - Autoscaling amplifier for: I0_USAXS
W Tue-00:45:03 - Exception during execution of command usaxs.mac:
Status(obj=EpicsSignal(read_pv='9idcLAX:pd05:seq01:reqrange', name='trd_autorange_controls_reqrange', parent='trd_autorange_controls', value=3, timestamp=1608014643.200544,2020-12-15 03:25:16: auto_collect is waiting for next command from EPICS ...mits=False, put_complete=False), done=True, success=False)
D Tue-03:25:16 - starting user commandscommand
I Tue-03:25:16 - Command file: usaxs.mac
```

seems like this time we hanged on ranging PV.

jilavsky, 2020-12-15:
This is example of PV which would screw up one transmission measurement, but we should have continued.

jilavsky, 2020-12-15:
While running spec we see more timeouts:

```Not connected PV=9idcUSX:pd01:seq02:bkgErr0  ophyd=metadata_0250
Not connected PV=9idcUSX:pd01:seq02:bkgErr1  ophyd=metadata_0253
preliminaryWriteFile(): PV 9idcUSX:pd01:seq02:bkgErr0 is not connected now
preliminaryWriteFile(): PV 9idcUSX:pd01:seq02:bkgErr1 is not connected now
```

prjemian, 2021-01-17:
Will wrap calls to autoscale each scaler's amplifier with `try..except` that will continue after logging _any_ exception.

commit d637b592 (2021-01-17): FIX #466
closed 2021-01-17 by prjemian