APS-USAXS/usaxs-bluesky-ended-2023 #467: Test priming of ourADs

jilavsky, 2020-12-16:
I recall vaguely that our ADs have **still reported** issues when ioc was restarted with priming. When I faced it last, I needed to collect data and so I manually primed ADs from epics. 
Shutdown is great time to test this as we do not need beam for that. I suggest we test this when convenient. I will be trying to crash ioc for Kevin to find reason why they die sometimes which is much higher priority. But after this we can test and evaluate this easily. 
I suggest we restart ioc and test image collection from BS. I suspect that will fail and priming will be issue. 
For our ADs, if not primed, this is what BS could do:
1. Enable proper saving plugin but do not enable saving (we do not need image, just fill this with image)
2. switch off shutter control (no need to open/close shutter)
3. set short time (0.1 sec seems sensible for any detector, common 20-30 seconds we expose is overkill)
4. Collect image
5. Set back to normal conditions - disable savings plugin, enable shutter control, restore back prior exposure time.

prjemian, 2020-12-16:
See #445, not exact duplicate; this is testing that priming is properly detected and sufficient.

prjemian, 2021-01-17:
The new release of [apstools 1.4.0](https://github.com/BCDA-APS/apstools/releases/tag/1.4.0) (on 2021-01-15) includes an improved test of #470 for plugin priming.  Install this update into the instrument's conda environment.

prjemian, 2021-01-17:
update notes:

```
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ conda update apstools -c aps-anl-tag
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /share1/bluesky/conda_envs/bluesky_2021_1

  added / updated specs:
    - apstools

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    anaconda-project-0.9.1     |     pyhd3eb1b0_1         215 KB
    apstools-1.4.0             |             py_0          99 KB  aps-anl-tag
    gevent-20.12.1             |   py38h27cfd23_1         1.7 MB
    importlib_resources-5.0.0  |   py38h06a4308_1          38 KB
    ipywidgets-7.6.3           |     pyhd3eb1b0_1         105 KB
    jupyterlab_widgets-1.0.0   |     pyhd3eb1b0_1         109 KB
    keyring-21.8.0             |   py38h06a4308_0          54 KB
    nbformat-5.1.2             |     pyhd3eb1b0_1          68 KB
    nexusformat-0.6.0          |     pyh9656e83_0          55 KB  conda-forge
    numexpr-2.7.2              |   py38hb2eb853_0         186 KB
    openpyxl-3.0.6             |     pyhd3eb1b0_0         159 KB
    outcome-1.1.0              |     pyhd3eb1b0_0          14 KB
    pillow-8.1.0               |   py38he98fc37_0         626 KB
    ptyprocess-0.7.0           |     pyhd3eb1b0_2          17 KB
    pygments-2.7.4             |     pyhd3eb1b0_0         676 KB
    pytz-2020.5                |     pyhd3eb1b0_0         182 KB
    seaborn-0.11.1             |     pyhd3eb1b0_0         212 KB
    setuptools-51.1.2          |   py38h06a4308_4         742 KB
    sortedcollections-1.2.3    |     pyhd3eb1b0_0          11 KB
    spyder-kernels-1.10.1      |   py38h06a4308_0         101 KB
    terminado-0.9.2            |   py38h06a4308_0          25 KB
    tifffile-2021.1.14         |     pyhd3eb1b0_1         126 KB
    tqdm-4.55.1                |     pyhd3eb1b0_0          77 KB
    trio-0.18.0                |   py38h06a4308_0         450 KB
    typed-ast-1.4.2            |   py38h27cfd23_1         189 KB
    ------------------------------------------------------------
                                           Total:         6.1 MB

The following NEW packages will be INSTALLED:

  jupyterlab_widgets pkgs/main/noarch::jupyterlab_widgets-1.0.0-pyhd3eb1b0_1

The following packages will be UPDATED:

  anaconda-project                               0.8.4-py_0 --> 0.9.1-pyhd3eb1b0_1
  apstools                                       1.3.9-py_0 --> 1.4.0-py_0
  epics-base         conda-forge::epics-base-7.0.4-h68659b~ --> aps-anl-tag::epics-base-7.0.4-pl526_1
  gevent                              20.9.0-py38h7b6447c_0 --> 20.12.1-py38h27cfd23_1
  importlib_resourc~                   3.3.0-py38h06a4308_0 --> 5.0.0-py38h06a4308_1
  ipywidgets                                     7.5.1-py_1 --> 7.6.3-pyhd3eb1b0_1
  keyring                                     21.4.0-py38_1 --> 21.8.0-py38h06a4308_0
  nbformat                                       5.0.8-py_0 --> 5.1.2-pyhd3eb1b0_1
  nexusformat                            0.5.3-pyh95af2a2_0 --> 0.6.0-pyh9656e83_0
  numexpr                              2.7.1-py38h63df603_0 --> 2.7.2-py38hb2eb853_0
  openpyxl                                       3.0.5-py_0 --> 3.0.6-pyhd3eb1b0_0
  pillow                               8.0.1-py38he98fc37_0 --> 8.1.0-py38he98fc37_0
  ptyprocess                             0.6.0-pyhd3eb1b0_2 --> 0.7.0-pyhd3eb1b0_2
  pygments                               2.7.3-pyhd3eb1b0_0 --> 2.7.4-pyhd3eb1b0_0
  pytz                                  2020.4-pyhd3eb1b0_0 --> 2020.5-pyhd3eb1b0_0
  seaborn                                       0.11.0-py_0 --> 0.11.1-pyhd3eb1b0_0
  setuptools                          51.0.0-py38h06a4308_2 --> 51.1.2-py38h06a4308_4
  sortedcollections                              1.2.1-py_0 --> 1.2.3-pyhd3eb1b0_0
  spyder-kernels                      1.10.0-py38h06a4308_0 --> 1.10.1-py38h06a4308_0
  terminado                                    0.9.1-py38_0 --> 0.9.2-py38h06a4308_0
  tifffile                           2020.12.8-pyhd3eb1b0_0 --> 2021.1.14-pyhd3eb1b0_1
  tqdm                                  4.54.1-pyhd3eb1b0_0 --> 4.55.1-pyhd3eb1b0_0
  trio               conda-forge::trio-0.17.0-py38h578d9bd~ --> pkgs/main::trio-0.18.0-py38h06a4308_0
  typed-ast                            1.4.1-py38h7b6447c_0 --> 1.4.2-py38h27cfd23_1

The following packages will be SUPERSEDED by a higher-priority channel:

  outcome            conda-forge::outcome-1.1.0-pyhd8ed1ab~ --> pkgs/main::outcome-1.1.0-pyhd3eb1b0_0
```

prjemian, 2021-01-17:
Here is where `prime` is used in the instrument package:

```
profile_bluesky/startup/instrument/collection.py:from .devices.blackfly import AD_prime_plugin    # TODO: temporary
profile_bluesky/startup/instrument/devices/blackfly.py:from apstools.devices import AD_plugin_primed
profile_bluesky/startup/instrument/devices/blackfly.py:# TODO: from apstools.devices import AD_prime_plugin
profile_bluesky/startup/instrument/devices/blackfly.py:def AD_prime_plugin(detector, detector_plugin):
profile_bluesky/startup/instrument/devices/blackfly.py:    if not AD_plugin_primed(blackfly_optical.jpeg1):
profile_bluesky/startup/instrument/devices/blackfly.py:            "NOTE: blackfly_optical.jpeg1 has not been primed yet."
profile_bluesky/startup/instrument/devices/blackfly.py:            "  AD_prime_plugin(blackfly_optical, blackfly_optical.jpeg1)"
```

prjemian, 2021-01-17:
Aaack!  Cannot test yet:

```
I Sun-12:37:21 - /home/beams11/USAXS/.ipython/profile_bluesky/startup/instrument/devices/blackfly.py
W Sun-12:37:22 - Timeout connecting with PointGrey BlackFly (9idFLY1:)
W Sun-12:37:23 - Timeout connecting with PointGrey BlackFly Optical (9idFLY2:): 9idFLY2:cam1:Acquire_RBV could not connect within 1.0-second timeout.
```

Detectors not turned on?

prjemian, 2021-01-17:
Tests on saxs and waxs detector indicate failure to prime:

```
In [5]: from apstools.devices import  AD_prime_plugin2, AD_plugin_primed

In [6]: AD_plugin_primed(waxs_det.hdf1)
Out[6]: False

In [7]: AD_plugin_primed(saxs_det.hdf1)
Out[7]: False

In [8]: AD_prime_plugin2(saxs_det.hdf1)

In [9]: AD_plugin_primed(saxs_det.hdf1)
Out[9]: False

In [10]: AD_prime_plugin2(waxs_det.hdf1)

In [11]: AD_plugin_primed(waxs_det.hdf1)
Out[11]: False
```

prjemian, 2021-01-17:
Test using canonical warmup from ophyd also indicates plugin is not primed:

```python
In [13]: AD_plugin_primed(waxs_det.hdf1)
Out[13]: False

In [14]: waxs_det.hdf1.warmup()

In [15]: AD_plugin_primed(waxs_det.hdf1)
Out[15]: False
```

prjemian, 2021-01-20:
Perhaps we should switch to using `lazy_open`?  It seems to add some additional time to save the file since it creates the file _after_ the acquisition instead of when `Acquire` is pressed.  But, operationally, operations will be much less fragile, as @markrivers has long suggested (since AD 2.2).

We'll need to choose a version-specific `HDFPlugin`, picking the highest available in ophyd that supports each detector's version.  Here is an example for the Dexela 2315: https://github.com/APS-USAXS/ipython-usaxs/blob/bd3d3d38c829d03893ef524acbaadcd1dc4cfd7f/profile_bluesky/startup/instrument/devices/dexela_module.py#L40  The IOC is AD 3.7, the highest ophyd support is 3.4 so pick that one.

No higher support for the Process plugin so we just cancel the Signal that the 3.7 Process plugin no longer provides: https://github.com/APS-USAXS/ipython-usaxs/blob/bd3d3d38c829d03893ef524acbaadcd1dc4cfd7f/profile_bluesky/startup/instrument/devices/dexela_module.py#L51-L53

prjemian, 2021-01-25:
We'll need to apply a local patch to the `HDF5Plugin.stage()` method to handle the `lazy_open` attribute.  See https://github.com/bluesky/ophyd/issues/962 for details.

prjemian, 2021-01-25:
Using `from apstools.devices import AD_plugin_primed`:

```python
In [27]: AD_plugin_primed(waxs_det.hdf1)
Out[27]: False

In [28]: AD_plugin_primed(saxs_det.hdf1)
Out[28]: False

In [29]: AD_plugin_primed(dexela_det.hdf1)
Out[29]: True
```

Correct for the Dexela, not certain about saxs & waxs det but probably correct.

prjemian, 2021-01-25:
Confirm about SAXS & WAXS:

```python
In [55]: saxs_det.hdf1.array_size.get()
Out[55]: ArraySizeTuple(depth=0, height=0, width=0)

In [56]: waxs_det.hdf1.array_size.get()
Out[56]: ArraySizeTuple(depth=0, height=0, width=0)
```

prjemian, 2021-01-25:
Even after *priming* both SAXS & WAXS detectors from the caQtDM interface, both fail the test for primed because the data types are not the same.  Here is for the SAXS:

```python
In [61]: AD_plugin_primed(saxs_det.hdf1)
Out[61]: False

In [63]: saxs_det.cam.color_mode.get()
Out[63]: 0

In [64]: saxs_det.hdf1.color_mode.get()
Out[64]: 0

In [65]: saxs_det.cam.data_type.get()
Out[65]: 0

In [66]: saxs_det.hdf1.data_type.get()
Out[66]: 'Int32'

In [67]: saxs_det.hdf1.data_type.get(as_string=False)
Out[67]: '4'

In [68]: saxs_det.cam.data_type.get(as_string=True)
Out[68]: 'Int8'

```

Will modify the test for `AD_plugin_primed()` since it seems these detectors cannot pass it when primed.

prjemian, 2021-01-25:
The ophyd test for primed only checks for `array_size` to be zero.  But, plugin needs re-priming when the image dimensions change.  This is the case where `LazyOpen` will help.

prjemian, 2021-01-25:
With the local override, both detectors declare they are primed:

```python
In [76]: %run -m instrument.devices.area_detector_common
/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/runpy.py:126: RuntimeWarning: 'instrument.devices.area_detector_common' found in sys.modules after import of package 'instrument.devices', but prior to execution of 'instrument.devices.area_detector_common'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
I Mon-11:48:22 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/devices/area_detector_common.py

In [77]: Override_AD_plugin_primed(waxs_det.hdf1)
Out[77]: True

In [78]: Override_AD_plugin_primed(saxs_det.hdf1)
Out[78]: True

```

prjemian, 2021-01-25:
Can't change the image size on SAXS or WAXS.  So must accept this override for now.

The only other place the instrument package checks if a plugin is primed is in one of the Blackfly detectors:

```
bash-4.2$ git grep AD_plug
profile_bluesky/startup/instrument/devices/area_detector_common.py:from apstools.devices import AD_plugin_primed
profile_bluesky/startup/instrument/devices/area_detector_common.py:def Override_AD_plugin_primed(plugin):
profile_bluesky/startup/instrument/devices/area_detector_common.py:    if Override_AD_plugin_primed(plugin):
profile_bluesky/startup/instrument/devices/blackfly_module.py:from apstools.devices import AD_plugin_primed
profile_bluesky/startup/instrument/devices/blackfly_module.py:    if not AD_plugin_primed(blackfly_optical.jpeg1):
```

Will choose the override in that module.

Note:  blackfly detector IOC is not running at this time:

```python
I Mon-11:56:49 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/devices/blackfly_module.py
W Mon-11:56:50 - Timeout connecting with PointGrey BlackFly (9idFLY1:)
W Mon-11:56:51 - Timeout connecting with PointGrey BlackFly Optical (9idFLY2:): 9idFLY2:cam1:Acquire_RBV could not connect within 1.0-second timeout.
# ...
In [2]: !caget 9idFLY2:cam1:Acquire
Channel connect timed out: '9idFLY2:cam1:Acquire' not found.

prjemian, 2021-01-25:
Blackfly2 is now running, checking the prime method and discover we need to wait longer for the priming.  Watch the `cam.acquire` attribute.

commit 144fd9cc (2021-01-25): FIX #467
closed 2021-01-25 by prjemian