APS-USAXS/usaxs-bluesky-ended-2023 #468: How did we get this?

jilavsky, 2020-12-16:
This is now on BS terminal afgter not being used for 24 hours:

```
Out[32]: ()                                                                                                                                                                 
                                                                                                                                                                            
In [33]: RE(mode_Radiography())
I Wed-10:08:51 - Preparing for USAXS mode ... please wait ...
I Wed-10:08:51 - Prepared for USAXS mode
mono_shutter [In progress. No progress bar available.]                                                                                                                      
                                                                                                                                                                            
^[Exception in thread Thread-490241:                                                                                                                                        
Traceback (most recent call last):
  File "/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/Documents/projects/apstools/apstools/devices.py", line 462, in move_it
    self.open()
  File "/home/beams11/USAXS/Documents/projects/apstools/apstools/devices.py", line 815, in open
    self.wait_for_state(
  File "/home/beams11/USAXS/Documents/projects/apstools/apstools/devices.py", line 807, in wait_for_state
    raise TimeoutError(msg)
TimeoutError: Timeout (10 s) waiting for shutter state to reach a value in [1, 'O', 'N']

exA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.

```

We will need to evaluate what this means and how did we get here...

jilavsky, 2020-12-16:
And this is what happened when I did ctrl-C to get out of here:

```

exA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.

^CA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.
^Ctrying a second time
Pausing...
---------------------------------------------------------------------------                                                                                                 
RunEngineInterrupted                      Traceback (most recent call last)
<ipython-input-33-890e5f2ff449> in <module>
----> 1 RE(mode_Radiography())

/share1/bluesky/conda_envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    808 
    809         if self._interrupted:
--> 810             raise RunEngineInterrupted(self.pause_msg) from None
    811 
    812         return tuple(self._run_start_uids)

RunEngineInterrupted: 
Your RunEngine is entering a paused state. These are your options for changing
the state of the RunEngine:

RE.resume()    Resume the plan.
RE.abort()     Perform cleanup, then kill plan. Mark exit_stats='aborted'.
RE.stop()      Perform cleanup, then kill plan. Mark exit_status='success'.
RE.halt()      Emergency Stop: Do not perform cleanup --- just stop.

```

prjemian, 2020-12-16:
Looks like duplicate of #456.

prjemian, 2020-12-16:
> [10:29 AM] Ilavsky, Jan
> User reports: I moved to radiography mode when hutch was oepened. 
> Same thing we had, code was suppose to open shutter, could not and decided to suspend.

closed 2020-12-16 by prjemian