APS-USAXS/usaxs-bluesky-ended-2023 #471: AD support for Dexela 2315 detector

prjemian, 2021-01-11:
When @kmpeters has created the soft IOC, we'll need ophyd support.  Need the PV prefix for this (and any special advice specific to the detector).

kmpeters, 2021-01-14:
The IOC has been deployed (see email).  PV prefix = 9idcDEX:

Gitlab repo:
https://git.aps.anl.gov/9ID/9idcDEX

The binning options for this camera look different than other cameras (left side of Dexela.ui).

prjemian, 2021-01-17:
Is this detector turned ON?

```
bash-4.2$ caget -S 9idcDEX:HDF1:FilePath
Channel connect timed out: '9idcDEX:HDF1:FilePath' not found.
```

For comparison:

```
bash-4.2$ caget -S usaxs_pilatus1:HDF1:FilePath
usaxs_pilatus1:HDF1:FilePath /mnt/share1/USAXS_data/2020-12/12_16_Fan_Cast174/12_16_Fan_Cast174_saxs/
```

prjemian, 2021-01-17:
@kmpeters : This is puzzling.

```
bash-4.2$ nslookup ioc9idcDEX
Server:		164.54.109.205
Address:	164.54.109.205#53

** server can't find ioc9idcDEX: NXDOMAIN
```

prjemian, 2021-01-17:
The Dexela IOC is on Windows.  Is the file store drive SMB-mapped to usaxscontrol?  For standard (easy) bluesky integration, the bluesky session needs to be able to see the data files at run time.  Otherwise, integration is possible but will delay use of this detector in bluesky.

prjemian, 2021-01-17:
waiting for more info before proceeding, here is current status:

```python
In [1]: from instrument.devices.area_detector_common import area_detector_EPICS_PV_prefix

In [2]: "Dexela 2315" in area_detector_EPICS_PV_prefix
Out[2]: True

In [3]: from instrument.devices.dexela import dexela_det
I Sun-13:32:26 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/devices/dexela.py
W Sun-13:32:27 - Timeout connecting with Dexela 2315 (9idcDEX:)
```

jilavsky, 2021-01-18:
Data storage is on share1 drive which is mounted as W: Correct path is now: W:\USAXS_data 
Drive should remount on reboot. 
This is linux data drive which is limited to use by only devices and control software, such as spec, BlueSky, or Python support. HDF5/Nexus templates are available on this drive. It is mounted with proper credentials, so you should have read/write privileges.

prjemian, 2021-01-19:
```python
WRITE_HDF5_FILE_PATH_DEXELA = r"W:\USAXS_data\test\dexela\%Y\%m\%d\"
READ_HDF5_FILE_PATH_DEXELA = "/share1/USAXS_data/test/dexela/%Y/%m/%d/"
```

(The Windows file path gets a raw `r` prefix so the backslash is not interpreted by Python as a character escape prefix.)

prjemian, 2021-01-19:
Confirmed IOC is running:

```
usaxs@usaxscontrol /share1/USAXS_data $ caget 9idcDEX:cam1:Acquire
9idcDEX:cam1:Acquire           Done
```

prjemian, 2021-01-19:
This is a good sign:

```python
I Tue-09:21:42 - bluesky startup is complete

In [1]: dexela_det.connected
Out[1]: True
```

prjemian, 2021-01-19:
Before declaring this issue as resolved, test the support by taking an image.  Configure the plugin workflow as shown above.

prjemian, 2021-01-19:
While priming the `Proc1` plugin, came across [this exception](https://github.com/BCDA-APS/apstools/issues/485), that the Dexela does not have a trigger mode called `Internal`.  Instead, the trigger modes are:

```
In [13]: dexela_det.cam.trigger_mode.enum_strs
Out[13]: 
('Int. Free Run',
 'Int. Fixed Rate',
 'Int. Software',
 'Ext. Edge Single',
 'Ext. Edge Multi',
 'Ext. Bulb')
```

prjemian, 2021-01-19:
As found now, the plugin chain (visualized with ophyd command: `dexela_det.visualize_asyn_digraph()`) looks like:

[image]

Need to put PROC1 plugin _between_ DEX1 and (Image1 & FileHDF1).  This involves changing the NDArrayPort (which **is** available in the ophyd support) of the Image1 and FileHDF1 plugins.

prjemian, 2021-01-19:
This is not difficult:

```python
In [32]: proc_port = dexela_det.proc1.port_name.get()
    ...: dexela_det.image.nd_array_port.put(proc_port)
    ...: dexela_det.hdf1.nd_array_port.put(proc_port)
    ...: 

In [33]: dexela_det.visualize_asyn_digraph()
```

[image]

prjemian, 2021-01-19:
But we don't *need* the image plugin to get the processed images, so set it back to the cam's port:

```python
In [37]: dexela_det.image.nd_array_port.put(dexela_det.cam.port_name.get())

In [38]: dexela_det.visualize_asyn_digraph()
```

[image]

prjemian, 2021-01-19:
HDF plugin staging failed:

```
TimeoutError: Attempted to set EpicsPathSignal(read_pv='9idcDEX:HDF1:FilePath_RBV', name='dexela_det_hdf1_file_path', parent='dexela_det_hdf1', value='\\', timestamp=1611075323.4999194, auto_monitor=True, string=True, write_pv='9idcDEX:HDF1:FilePath', limits=False, put_complete=False, path_semantics='posix') to value '/' and timed out after 10 seconds. Current value is '\\'.
```

prjemian, 2021-01-19:
Now, in response to `dexela_det.hdf1.stage()`, this exception:

```python
TimeoutError: Attempted to set EpicsSignalWithRBV(read_pv='9idcDEX:HDF1:Capture_RBV', name='dexela_det_hdf1_capture', parent='dexela_det_hdf1', value=0, timestamp=1611080586.7580442, auto_monitor=True, string=False, write_pv='9idcDEX:HDF1:Capture', limits=False, put_complete=False) to value 1 and timed out after 10 seconds. Current value is 0.
```

prjemian, 2021-01-19:
Capture won't go to `1` unless some unset condition is satisfied.  What's the condition?

@markrivers has long suggested using the *Lazy open* feature instead of relying on priming the plugin.  Try that next.

prjemian, 2021-01-19:
Ok, moving on to next problem when staging the HDF plugin:

```python
/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/areadetector/filestore_mixins.py in _generate_resource(self, resource_kwargs)
    300 
    301     def _generate_resource(self, resource_kwargs):
--> 302         fn = PurePath(self._fn).relative_to(self.reg_root)
    303         resource, self._datum_factory = resource_factory(
    304             spec=self.filestore_spec,

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/pathlib.py in relative_to(self, *other)
    897         if (root or drv) if n == 0 else cf(abs_parts[:n]) != cf(to_abs_parts):
    898             formatted = self._format_parsed_parts(to_drv, to_root, to_parts)
--> 899             raise ValueError("{!r} does not start with {!r}"
    900                              .format(str(self), str(formatted)))
    901         return self._from_parsed_parts('', root if n == 1 else '',

ValueError: 'W:\\\\USAXS_data\\\\test\\\\dexela\\\\%Y\\\\%m\\\\%d\\\\bluesky_test_0004.h5' does not start with '/'
```

prjemian, 2021-01-19:
Tougher problem, asked for help from bluesky working group on slack.com.

prjemian, 2021-01-19:
@tacaswell suggested to simplify the problem and confirm that custom file naming code is not in use.  This will help to divide the problem space into ophyd v. apstools.

prjemian, 2021-01-19:
read this page: https://blueskyproject.io/ophyd/area-detector.html

jilavsky, 2021-01-20:
Ouch, my head hurts from that page...

prjemian, 2021-01-20:
I'm having trouble getting the HDF plugin to `stage()` correctly.  The problem is the Window file path and how AD appends a `\` to the end of the FilePath PV when it is not supplied.  The default ophyd code is removing the `\` that I have supplied.  Just now found where and probably how to resolve that.

prjemian, 2021-01-20:
Success staging and acquiring with a simple example (from a jupyter notebook):

```python
from collections import OrderedDict
import datetime
from ophyd import ADComponent
from ophyd import DexelaDetector
from ophyd import HDF5Plugin
from ophyd import ImagePlugin
from ophyd import ProcessPlugin
from ophyd import SingleTrigger
from ophyd.areadetector.filestore_mixins import FileStoreHDF5IterativeWrite
from ophyd.areadetector.filestore_mixins import new_short_uid
from ophyd.areadetector.plugins import HDF5Plugin_V34
import time

class MyHDF5Plugin(FileStoreHDF5IterativeWrite, HDF5Plugin_V34):

    def make_filename(self):
        """Override from ophyd.areadetector.filestore_mixins.FileStorePluginBase."""
        # filename = new_short_uid()
        filename = self.file_name.get()  # get name from EPICS PV
        now = datetime.datetime.now()
        write_path = now.strftime(self.write_path_template) + "\\"  # the FIX!
        read_path = now.strftime(self.read_path_template)
        return filename, read_path, write_path

class MyProcessPlugin(ProcessPlugin):

    pool_max_buffers = None

    
class MyDexelaDetector(SingleTrigger, DexelaDetector):
    """Dexela detector(s) as used by 9-ID-C USAXS."""

    hdf1 = ADComponent(
        MyHDF5Plugin,
        "HDF1:",
        write_path_template="W:\\USAXS_data\\test\\dexela\\%Y\\%m\\%d\\",
        read_path_template="/share1/USAXS_data/test/dexela/%Y/%m/%d/",
        path_semantics="windows",
    )
    image = ADComponent(ImagePlugin, "image1:")
    proc1 = ADComponent(MyProcessPlugin, "Proc1:")

dexela_det = MyDexelaDetector("9idcDEX:", name="dexela_det")
```

```python
dexela_det.read_attrs.append("hdf1")
dexela_det.hdf1.create_directory.put(-5)  # MUST come before staging writes file_path
dexela_det.hdf1.file_name.put("bluesky")

dexela_det.hdf1.stage_sigs = OrderedDict()
dexela_det.hdf1.stage_sigs["enable"] = 1
dexela_det.hdf1.stage_sigs["blocking_callbacks"] = "Yes"
dexela_det.hdf1.stage_sigs["parent.cam.array_callbacks"] = 1
dexela_det.hdf1.stage_sigs["auto_increment"] = "Yes"
dexela_det.hdf1.stage_sigs["array_counter"] = 0
dexela_det.hdf1.stage_sigs["auto_save"] = "Yes"
dexela_det.hdf1.stage_sigs["num_capture"] = 0
dexela_det.hdf1.stage_sigs["file_template"] = "%s%s_%6.6d.h5"
dexela_det.hdf1.stage_sigs["file_write_mode"] = "Capture"
dexela_det.hdf1.stage_sigs["lazy_open"] = 1
dexela_det.hdf1.stage_sigs["compression"] = "LZ4"
dexela_det.hdf1.stage_sigs["capture"] = 1
```

```python
dexela_det.stage()
```

```python
t0 = time.time()
dexela_det.trigger()
while dexela_det.cam.acquire.get() not in (0, "Stop", "Done"):
    time.sleep(0.02)
print(f"acquire time: {time.time()-t0:.2f}")
```

    acquire time: 2.71

```python
dexela_det.unstage()
```

prjemian, 2021-01-20:
Working:

```python
In [12]: acquire_Dexela_N(2.1)
I Tue-23:39:46 - for acquire time of 2.1 s, will acquire 42 frames
D Tue-23:39:46 - staging ...
D Tue-23:39:46 - triggering ...
I Tue-23:39:46 - sleep: 2.1
I Tue-23:39:49 - acquire time: 2.69
D Tue-23:39:49 - unstaging ...
D Tue-23:39:49 - acquire complete
```

The file:

```
(/share1/bluesky/conda_envs/bluesky_2021_1) usaxs@usaxscontrol .../USAXS_data/test $ ll -h dexela/2021/01/19/
total 23M
-rw-rw-r-- 1 usaxs 12M Jan 19 23:39 bluesky_000000.h5
-rw-rw-r-- 1 usaxs 12M Jan 19 23:39 bluesky_000001.h5
-rw-rw-r-- 1 usaxs 12K Jan 19 22:54 bluesky_test_000000.h5
-rw-rw-r-- 1 usaxs 12K Jan 19 23:04 my_bluesky_test_000000.h5
(/share1/bluesky/conda_envs/bluesky_2021_1) usaxs@usaxscontrol .../USAXS_data/test $ punx tree dexela/2021/01/19/bluesky_000001.h5

!!! WARNING: this program is not ready for distribution.

/share1/USAXS_data/test/dexela/2021/01/19/bluesky_000001.h5 : NeXus data file
  entry:NXentry
    @NX_class = NXentry
    data:NXdata
      @NX_class = NXdata
      data:NX_UINT16[1944,3072] = __array
        __array = [
            [26192, 32741, 32738, '...', 32738]
            [22924, 22926, 26203, '...', 32748]
            [16374, 13100, 3271, '...', 13098]
            ...
            [62252, 22924, 26200, '...', 26201]
          ]
        @NDArrayDimBinning = 1
        @NDArrayDimOffset = 0
        @NDArrayDimReverse = 0
        @NDArrayNumDims = 2
        @signal = 1
    instrument:NXinstrument
      @NX_class = NXinstrument
      NDAttributes:NXcollection
        @NX_class = NXcollection
        @hostname = usaxspc8
        NDArrayEpicsTSSec:NX_UINT32 = 979969150
          @NDAttrDescription = The NDArray EPICS timestamp seconds past epoch
          @NDAttrName = NDArrayEpicsTSSec
          @NDAttrSource = Driver
          @NDAttrSourceType = NDAttrSourceDriver
        NDArrayEpicsTSnSec:NX_UINT32 = 175389685
          @NDAttrDescription = The NDArray EPICS timestamp nanoseconds
          @NDAttrName = NDArrayEpicsTSnSec
          @NDAttrSource = Driver
          @NDAttrSourceType = NDAttrSourceDriver
        NDArrayTimeStamp:NX_FLOAT64 = 979969150.1753886
          @NDAttrDescription = The timestamp of the NDArray as float64
          @NDAttrName = NDArrayTimeStamp
          @NDAttrSource = Driver
          @NDAttrSourceType = NDAttrSourceDriver
        NDArrayUniqueId:NX_INT32 = 6731
          @NDAttrDescription = The unique ID of the NDArray
          @NDAttrName = NDArrayUniqueId
          @NDAttrSource = Driver
          @NDAttrSourceType = NDAttrSourceDriver
      detector:NXdetector
        @NX_class = NXdetector
        data:NX_UINT16[1944,3072] = __array
          __array = [
              [26192, 32741, 32738, '...', 32738]
              [22924, 22926, 26203, '...', 32748]
              [16374, 13100, 3271, '...', 13098]
              ...
              [62252, 22924, 26200, '...', 26201]
            ]
          @NDArrayDimBinning = 1
          @NDArrayDimOffset = 0
          @NDArrayDimReverse = 0
          @NDArrayNumDims = 2
          @signal = 1
        NDAttributes:NXcollection
          @NX_class = NXcollection
      performance
        timestamp:NX_FLOAT64[1,5] = __array
          __array = [
              [0.004611247, 0.010766016, 0.004613549, 8464.133807714943, 19751.605542717767]
            ]
```

commit 7f26981c (2021-01-20): FIX #471
PR #472 (2021-01-20): Bluesky (ophy) setup for Dexela 2315 area detector
fix #471
closed 2021-01-20 by prjemian