APS-USAXS/usaxs-bluesky-ended-2023 #478: 2 FE-EPS PVs not available during 2021-1 startup

prjemian, 2021-01-26:
We used to connect to these two PVs but today (and yesterday) they are not responding:

```
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ caget EPS:09:ID:Minor
Channel connect timed out: 'EPS:09:ID:Minor' not found.
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ caget EPS:09:ID:Major
Channel connect timed out: 'EPS:09:ID:Major' not found.
```

prjemian, 2021-01-26:
@kmpeters : Have these two PVs in the FE-EPS changed?

prjemian, 2021-01-26:
These are strictly for logging now so not essential, just informative:

```
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ git grep major_fault
profile_bluesky/startup/instrument/devices/diagnostics.py:    # major_fault = Component(EpicsSignalRO, "EPS:09:ID:Major", string=True)
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ git grep minor_fault
profile_bluesky/startup/instrument/devices/diagnostics.py:    # minor_fault = Component(EpicsSignalRO, "EPS:09:ID:Minor", string=True)
```

kmpeters, 2021-02-01:
> @kmpeters : Have these two PVs in the FE-EPS changed?

I am not aware of any FE-EPS PV changes, nor do I believe I'm in the loop if changes are made.