APS-USAXS/usaxs-bluesky-ended-2023 #479: Rething folder creation issues

jilavsky, 2021-01-26:
Currently we let BS create folder structure based on 
YYYY-MM/MM_DD_name/MM_DD_name_saxs,waxs,usaxs
this is different than spec which creates only the : 
MM_DD_name/MM_DD_name_saxs,waxs,usaxs

which is better? What should be done?

prjemian, 2021-01-26:
We found the current code causes a problem when the data is being stored into a different path than the bluesky working directory.  On 2021-01-26, we started in 2021-02 with usaxs.mac and other needed files.  With the `newUser()` command, that defaulted to 2021-01 for the data.  The consequence we saw was in the HDF5 file produced by a fly scan, the AR_PulsePositions was not the expected array of 8000 values but a single value (not an array).  When all config files relocated and restarted in the 2021-01 directory, the array size was as expected.

prjemian, 2021-01-26:
We know how to use this now but `newUser()` must be refactored along with our directory handling for saving data.

prjemian, 2022-01-24:
@jilavsky : What should be done here?  The docstring of `newUser()` says: https://github.com/APS-USAXS/ipython-usaxs/blob/670313a88fe0dc5873b14e1dd7f26d176c35deab/profile_bluesky/startup/instrument/utils/setup_new_user.py#L201-L223

jilavsky, 2022-01-25:
I think we are used to spec behavior = create new folder MM_DD_USER in current BS working directory. If needed, let staff shut down BS and manually move BS to new working directory.

Issues was, that newUser() created also YYY_MM directory above (the CWD - line 222) while BS was started in different working directory. That broke some of the functionality. 
Alternatively, we can keep this behavior, but need to make sure it works. This is difficult to test, not many chances to do so.

PR #544 (2022-02-01): refactor newUser(), use CWD
- FIX #479
closed 2022-02-01 by prjemian