APS-USAXS/usaxs-bluesky-ended-2023 #480: "Stop after this scan" stopped working

jilavsky, 2021-01-27:
I check the checkbox, after Flyscan is finished, code helpfully unchecks the checkbox BUT does not stop. This used to work and is now failing. Something to test and fix tomorrow, this is important feature - and bad bug.

jilavsky, 2021-01-27:
Got these e-mails:
subject: RequestAbort during attempt 1 of 5 of command '('USAXSscan', ['-2.945', '62.94', '0.2', 'GlassyCarbonHT'], 413, '      USAXSscan        -2.945      62.94      0.2      "GlassyCarbonHT"')''

date: 2021-01-26 21:20:40.400029
command file: /share1/USAXS_data/2021-01/usaxs.mac
line number: 413
command: ('USAXSscan', ['-2.945', '62.94', '0.2', 'GlassyCarbonHT'], 413, '      USAXSscan        -2.945      62.94      0.2      "GlassyCarbonHT"')
raw command:       USAXSscan        -2.945      62.94      0.2      "GlassyCarbonHT"
attempt: 1 of 5
exception: User requested stop data collection before next scan

prjemian, 2021-01-27:
This has to do with retries if one command in a command file set raises an exception.  We are catching `Exception` when we should not catch the request to stop.  Needs attention for sure.

prjemian, 2021-01-27:
So, after we catch `Exception`, if it was `RequestAbort`, we continue instead.

jilavsky, 2021-01-28:
Well, it did not stop when trying to stop WAXS. We need to verify, that this works form SAXs and WAX salso, was tested only on FLyscan and tunes...

jilavsky, 2021-01-28:
On yes, killing this was nightmare, I ended up doing many ctrl-C - which somehow paused only part of the process, but SAXS counting kept going, RE.abort had no effect and even ctrl \ seems to have no effect. Eventually I killed python and dumped the core.

prjemian, 2021-01-29:
Curious to know what was the console output from these events.

prjemian, 2021-01-29:
The core dump leaves an exact time marker:

```
-rw-------  1 usaxs usaxs 3048574976 Jan 28 09:23 core
```

prjemian, 2021-01-29:
Setting `Stop after this scan?` PV with various scans:

* [x] `preUSAXStune`
* [ ] `SAXS`  - unchecked the box yet continued.
* [ ] `WAXS`

prjemian, 2021-01-29:
Aha!  We catch the `RequestAbort` exception but continue the loop.  We need to break out of the loop, as well.

prjemian, 2021-01-29:
SAXS tested successfuly with the fix.  Assume the WAXS will work.

prjemian, 2021-01-29:
WAXS tested successfully, too.

commit f497e386 (2021-01-27): FIX #480 and tested, too
commit 5a1da49c (2021-01-29): FIX #480 (again)
closed 2021-01-29 by prjemian