APS-USAXS/usaxs-bluesky-ended-2023 #482: Want to run a separate heater process asynchronous from Bluesky session

prjemian, 2021-01-27:
User wants (sometimes) to run a heater schedule that executes independent of any Bluesky session.  The BS session will configure and initiate the heater schedule but the heater schedule should proceed independent of whether BS even continues to run (such as the RE gets interrupted or the entire BS process needs to be terminated, such as #481, or stored beam is lost or ...).

prjemian, 2021-01-27:
A separate ophyd script (does not need Bluesky) can be run in the background to run this process.

prjemian, 2021-01-27:
All this is similar to infrastructure for 8-ID-I XPCS:

* process: https://github.com/aps-8id-dys/ipython-8idiuser/blob/master/profile_bluesky/startup/spec_support/helper.py
* workflow: https://github.com/aps-8id-dys/ipython-8idiuser/blob/master/profile_bluesky/startup/spec_support/spec_DM_support.py
* starter: https://github.com/aps-8id-dys/ipython-8idiuser/blob/master/profile_bluesky/startup/spec_support/daemon.sh

prjemian, 2021-01-27:
If the profile is to import devices from the instrument package, then it needs to deal with possible contention from the `from instrument.session_logs import logger`.  Each device import also includes the session logger so it is not easy to avoid.

One possible solution is to create a `heater_process` subdirectory in the user's pwd for this process to be started.  That will create a separate set of logs just for the heater process.

prjemian, 2021-01-27:
Needs to know linkam rate, temperature, set point, and settled.  (This is because we cannot import from the instrument package due to conflicts with session logging.)  We'll need to define our own ophyd devices.

prjemian, 2021-01-27:
We'll need support for both controllers.

prjemian, 2021-01-27:
The CI94 is the older controller and will be dropped later this year. The other is T96, expect to use for years.
[image]

prjemian, 2021-01-27:
Quick demo (console output) of a do-nothing plan that calls for exit after running the first temperature profile:

```
(/share1/bluesky/conda_envs/bluesky_2020_1) bash-4.2$ /share1/bluesky/conda_envs/bluesky_2020_1/bin/python /home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/heater_profile_process.py
I Wed-14:35:47 - /home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/heater_profile_process.py
I Wed-14:35:47 - 10s Check if another process is running...
I Wed-14:35:57 - Starting the 10 Hz pulse...
I Wed-14:35:57 - Watch for the EPICS trigger to start heater profile.
D Wed-14:36:03 - Calling user heater plan
sleep for 1.54 second(s)
D Wed-14:36:05 - Returned from RE(planHeaterProcess())
I Wed-14:36:05 - Exit signal received from EPICS.
```

prjemian, 2021-01-27:
Here is an example when a plan raises an exception:

```python
sleep for 1.54 second(s)
D Wed-15:43:37 - Returned from RE(planHeaterProcess())
I Wed-15:43:37 - Exit signal received from EPICS.
(/share1/bluesky/conda_envs/bluesky_2020_1) bash-4.2$ /share1/bluesky/conda_envs/bluesky_2020_1/bin/python /home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/heater_profile_process.py
I Wed-15:44:08 - /home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/heater_profile_process.py
I Wed-15:44:08 - 10s Check if another process is running...
I Wed-15:44:18 - Starting the 10 Hz pulse...
I Wed-15:44:18 - Watch for the EPICS trigger to start heater profile.
D Wed-15:44:33 - Calling user heater plan
sleep for 5 second(s)
%s: %.1f  settled: %s ci94 40.0 True
%s: %.1f  settled: %s linkam_tc1 26.139131546020508 True
D Wed-15:44:38 - Returned from RE(planHeaterProcess())
I Wed-15:44:38 - Exit signal received from EPICS.
(/share1/bluesky/conda_envs/bluesky_2020_1) bash-4.2$ /share1/bluesky/conda_envs/bluesky_2020_1/bin/python /home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/heater_profile_process.py
I Wed-15:45:42 - /home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/heater_profile_process.py
I Wed-15:45:42 - 10s Check if another process is running...
I Wed-15:45:52 - Starting the 10 Hz pulse...
I Wed-15:45:52 - Watch for the EPICS trigger to start heater profile.
D Wed-15:46:19 - Calling user heater plan
sleep for 5 second(s)
ci94: 40.0  settled: True
linkam_tc1: 26.1  settled: True
D Wed-15:46:24 - Returned from RE(planHeaterProcess())
I Wed-15:46:24 - Exit signal received from EPICS.
(/share1/bluesky/conda_envs/bluesky_2020_1) bash-4.2$ /share1/bluesky/conda_envs/bluesky_2020_1/bin/python /home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/heater_profile_process.py
I Wed-15:48:09 - /home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/heater_profile_process.py
I Wed-15:48:09 - 10s Check if another process is running...
I Wed-15:48:19 - Starting the 10 Hz pulse...
I Wed-15:48:19 - Watch for the EPICS trigger to start heater profile.
D Wed-15:48:23 - Calling user heater plan
sleep for 5 second(s)
[E 15:48:28.369 run_engine:1261] Run aborted
    Traceback (most recent call last):
      File "/share1/bluesky/conda_envs/bluesky_2020_1/lib/python3.7/site-packages/bluesky/run_engine.py", line 1150, in _run
        msg = self._plan_stack[-1].send(resp)
      File "/home/beams/USAXS/.ipython/user/heater_profile.py", line 85, in planHeaterProcess
        f"{c.name} {c.egu}:"
      File "/share1/bluesky/conda_envs/bluesky_2020_1/lib/python3.7/site-packages/ophyd/device.py", line 1191, in __getattr__
        raise AttributeError(name)
    AttributeError: rate
E Wed-15:48:28 - RE(planHeaterProcess()) raised exception: rate
```

prjemian, 2021-01-28:
To help run the starter shell script, a soft link has been created in `~/bin/heater_profile_manager.sh` so the `heater_profile_manager.sh` command is available all the time.

```
bash-4.2$ which heater_profile_manager.sh
/home/beams/USAXS/bin/heater_profile_manager.sh
bash-4.2$ heater_profile_manager.sh
Usage: heater_profile_manager.sh {start|stop|restart|status|checkup|console|run}
```

The starter script runs the `heater_profile_process.py` program in a detached screen session.

prjemian, 2021-01-28:
Python code to start the heater process (using the `checkup` command which starts if not already running):

```python
import subprocess
STARTER = "/home/beams/USAXS/bin/heater_profile_manager.sh"
response = subprocess.run(f"{STARTER} checkup".split())
print(response)
```

prjemian, 2021-01-28:
This command completes within a second (0.3s or less is typical of observations) since it only issues the commands to start the process in a screen session.

```python
In [8]: import subprocess
   ...: import time
   ...: STARTER = "/home/beams/USAXS/bin/heater_profile_manager.sh"
   ...: t0 = time.time()
   ...: response = subprocess.run(f"{STARTER} checkup".split())
   ...: print(f"{response}: {time.time()-t0:.3f}s")
   ...: 
heater_profile_process is not running
Starting heater_profile_process
CompletedProcess(args=['/home/beams/USAXS/bin/heater_profile_manager.sh', 'checkup'], returncode=0): 0.230s

In [9]: import subprocess
   ...: import time
   ...: STARTER = "/home/beams/USAXS/bin/heater_profile_manager.sh"
   ...: t0 = time.time()
   ...: response = subprocess.run(f"{STARTER} checkup".split())
   ...: print(f"{response}: {time.time()-t0:.3f}s")
   ...: 
CompletedProcess(args=['/home/beams/USAXS/bin/heater_profile_manager.sh', 'checkup'], returncode=0): 0.299s
```

prjemian, 2021-01-28:
Can capture the output from the starter script by adding `capture_output=True`:

```python
import subprocess
import time
STARTER = "/home/beams/USAXS/bin/heater_profile_manager.sh"
t0 = time.time()
response = subprocess.run(f"{STARTER} checkup".split(), capture_output=True)
print(response.stdout.decode())  # , time.time()-t0)
```

observations:

```python
In [44]: response = subprocess.run(f"{STARTER} stop".split(), capture_output=True)

In [45]: response
Out[45]: CompletedProcess(args=['/home/beams/USAXS/bin/heater_profile_manager.sh', 'stop'], returncode=0, stdout=b'Stopping heater_profile_process (pid=23480)\n', stderr=b'')

In [46]: t0 = time.time()
    ...: response = subprocess.run(f"{STARTER} checkup".split(), capture_output=True)
    ...: print(response.stdout.decode())  # , time.time()-t0)
    ...: 
heater_profile_process is not running
Starting heater_profile_process
```

prjemian, 2021-01-29:
Process is established now.  Needs testing in operations now.

jilavsky, 2021-01-29:
BUG: Control did not start Linkam heater heating on when tested. If heating (ci94) is off, system needs to push also: 9idcLAX:ci94:start.PROC 
this can be pushed any time, so just push after each change of PVs.

prjemian, 2021-01-29:
That gets trickier.  We need to change from `bps.mv` which waits, to `bps.abs_set` which sets and does not wait.  Then pruch the PROC, then wait...

jilavsky, 2021-02-04:
The background program does not seem to stop when bit is raised. I tried manually in GUI display and from bps.mv(terms.HeaterProcess.linkam_trigger, 1) and the pulse is still going. I assume the process must stop so next time we start it, with potentially different parameters to run, it will reread its configuration and follow proper temperature trajectory.

commit 28a255ad (2021-02-02): FIX #482
closed 2021-02-04 by prjemian