APS-USAXS/usaxs-bluesky-ended-2023 #484: Linkam bug 2

jilavsky, 2021-01-29:
```
~/.ipython/user/james_linkam.py in jamesLinkamPlan(pos_X, pos_Y, thickness, scan_title, md)
     90     logger.info(f"Ramping temperature to {250} C")
     91     yield from linkam.set_target(250, wait=False)     #sets the temp of next ramp
---> 92     while not linkam.done.get():                      #runs data collection until next temp
     93         yield from bps.sleep(2)                       # sleep until settled
     94     t0 = time.time()

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/device.py in __getattr__(self, name)
   1195         # Components will be instantiated through the descriptor mechanism in
   1196         # the Component class, so anything reaching this point is an error.
-> 1197         raise AttributeError(name)
   1198 
   1199     def _instantiate_component(self, attr):

AttributeError: done

```

prjemian, 2021-01-29:
In the heater_plan.py, the linkams are configured as motor-like PVPositioners (which is the future) while in the linkam_new.py and others there is a different interface as defined in the `instrument` package.
You've mixed up how this device works.

prjemian, 2021-01-29:
I did not have the time to revise the `instrument` package (and all the places that is called) for this beam time.  So the two interfaces are different.  You'll have to sort this out for now.

prjemian, 2021-01-29:
Both this issue #484 and #483 are due to th this mixup.  I'll mark the first one as a duplicate.

prjemian, 2021-02-02:
@jilavsky What should we do with this now? (especially after #485)

jilavsky, 2022-01-12:
Expect this to be solved with #494.  Not closing until that is complete.

prjemian, 2022-01-22:
In the new support, this attribute is `linkam.temperature.done` but it may be easier to check the new property `linkam.temperature.inposition` for `True` or `False`.  This property is used to determine the `done` attribute now.

closed 2022-01-22 by prjemian