APS-USAXS/usaxs-bluesky-ended-2023 #485: Linkam support - check this is added

jilavsky, 2021-01-30:
While I understand Linkam support is now complicated (two different methods) which causes mess, one bug in current )old) support I am using needs to be fixed for final Linkam support.
Any time we command Linkam (e.g., yield from bps.mv(linkam.rate, 150),   yield from linkam.set_target(30, wait=False) ) or whatever the command will be, Linkam also MUST start running. We can have discussion if rate should trigger ON, but temperature surely should. User expects, that when they tell Linkam to go to 1000C, it is going to go and not be waiting for some additional command. 
While programmatically this is clearly different thing, users, like Andrew, Fan and Lyle will scream bloody hell, if they came in the morning to find heater at room temperature ;-)

prjemian, 2021-02-02:
Right.  The older controller is more fussy and the `trigger()` methods must be customized.  Rats.  Not the best time for an overhaul of the code (but the switch to PVPositioner would make the interface the same).  Taking a look how to customize the trigger.

prjemian, 2021-02-02:
It's best if have only one thing that triggers the controller ON, that would be setting the temperature set point.  Setting the ramp rate is a configuration issue, I believe.  Much like setting a motor's speed.  Setting ramp rate, alone, should not set things in motion.

prjemian, 2021-02-02:
Which method to override?

* `trigger()` -- only one trigger signal is supported (the first in `.trigger_signals`)
* `set()` -- calls `move()`
* `move()` -- calls `setpoint.put()`, then `.actuate.put(actuate_value)` if actuate is defined -- *This might be it*
* `put()` -- Put a value to all components of the device

@jilavsky : Looks like we should connect the `start.PROC` to the `actuate` signal with an `actuate_value` of 1 and try it.

prjemian, 2021-02-02:
What about the *stop* button (`stop.PROC`)?

prjemian, 2021-02-02:
```python
In [22]: linkam.stop??
Signature: linkam.stop(*, success=False)
Docstring: Stop the Device and all (instantiated) subdevices
Source:   
    def stop(self, *, success=False):
        if self.stop_signal is not None:
            self.stop_signal.put(self.stop_value, wait=False)
        super().stop(success=success)
File:      /share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/pv_positioner.py
Type:      method
```

Use the `stop_signal`.

prjemian, 2021-02-02:
@jilavsky This should be ready to try.  Be cautious but I expect it to set the temperature and begin.  Similarly, stop right away.

jilavsky, 2022-01-12:
Expect this to be solved with #494.  Not closing until that is complete.

prjemian, 2022-01-21:
TC96 support will now turn on heater power when temperature is set (https://github.com/BCDA-APS/apstools/pull/625 and tested locally).

Not guaranteed for the CI94.

closed 2022-01-21 by prjemian