APS-USAXS/usaxs-bluesky-ended-2023 #486: Instrument hanged in the middle of measurements

jilavsky, 2021-01-31:
Was running complicated Linkam ramp profile. 

at 2am instrument hanged with following errors:

```
I Sun-02:17:05 - Finished SAXS/WAXS data collection in 45.479679107666016 seconds.
I Sun-02:17:05 - I0 value: 6512932.0
D Sun-02:17:05 - waxsx start collection =0.000624999999956799
D Sun-02:17:05 - waxsx after before plan =0.000624999999956799
I Sun-02:17:05 - Found UsaxsSaxsMode = SAXS in beam
I Sun-02:17:05 - Moving to proper WAXS mode
                                                                                                                                                                            
I Sun-02:17:14 - Removed SAXS from beam position
I Sun-02:17:14 - Moving USAXS out of beam
I Sun-02:17:14 - Removed USAXS from beam position
I Sun-02:17:14 - Moving to WAXS mode
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
~/.ipython/user/james_linkam.py in <module>
----> 1 RE(jamesLinkamPlan(0,0,0.074,"V208"))

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1363                     else:
   1364                         try:
-> 1365                             msg = self._plan_stack[-1].send(resp)
   1366                         # We have exhausted the top generator
   1367                         except StopIteration:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    168                     continue
    169                 else:
--> 170                     raise ex
    171         # if inserting / mutating, put new generator on the stack
    172         # and replace the current msg with the first element from the

[... 100 lines cut ...]
    197         yield from move_USAXSOut()
--> 198         yield from move_WAXSIn()
    199 
    200     # move SAXS slits in, used for WAXS mode also

~/.ipython/profile_bluesky/startup/instrument/plans/move_instrument.py in move_WAXSIn()
     81     logger.info("Moving to WAXS mode")
     82 
---> 83     confirmUsaxsSaxsOutOfBeam()
     84     yield from plc_protect.wait_for_interlock()
     85 

~/.ipython/profile_bluesky/startup/instrument/plans/move_instrument.py in confirmUsaxsSaxsOutOfBeam()
     47         msg = "Incorrect UsaxsSaxsMode mode found."
     48         msg += " If SAXS, WAXS, and USAXS are out of beam, terms.SAXS.UsaxsSaxsMode.put(%d)"
---> 49         raise ValueError(msg % UsaxsSaxsModes["out of beam"])
     50 
     51 

ValueError: Incorrect UsaxsSaxsMode mode found. If SAXS, WAXS, and USAXS are out of beam, terms.SAXS.UsaxsSaxsMode.put(1)
```

This is confusing, this is done and works every time we run WAXS. 

NOTE: Instrument thinks it is in OpenBeamTPath mode (PV status indicates so) AND physically the instrument actually is in OpenBeamPath state, all limit sxitches are ingaged...

prjemian, 2021-02-02:
This fails to recognize that it is in the desired position, right?  Not that we need to try moving it again but that we need to read the positions again.  I'll look at the `confirmUsaxsSaxsOutOfBeam()` parts next.

Agree this perplexing.  There is STILL a deeper problem (in ophyd) that must be unresolved. I'll focus on how the USAXS will handle this, possibly retries.

prjemian, 2021-02-02:
Take note of this new proposition for bluesky plan stubs, a move (change a Signal, that is) with retries.  https://github.com/bluesky/bluesky/issues/1435

prjemian, 2021-02-02:
Instead of retries (in this case), force a new non-string read from the IOC and not the cache.

```python
    actual = terms.SAXS.UsaxsSaxsMode.get(
        timeout=10, as_string=False, use_monitor=False
    )
    expected = UsaxsSaxsModes["out of beam"]
    if actual != expected:
```

and improve the error report (to show the actual mode that was found).

prjemian, 2021-02-02:
Allow for 60s timeout.

prjemian, 2021-02-02:
example:

```python
I Mon-21:24:28 - bluesky startup is complete

In [1]: from instrument.plans.move_instrument import confirmUsaxsSaxsOutOfBeam

In [2]: RE(confirmUsaxsSaxsOutOfBeam())
W Mon-21:24:35 - Found UsaxsSaxsMode = 4 (WAXS in beam)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-2-d96f652ed386> in <module>
----> 1 RE(confirmUsaxsSaxsOutOfBeam())

~/.ipython/profile_bluesky/startup/instrument/plans/move_instrument.py in confirmUsaxsSaxsOutOfBeam()
     51         actual_str = xref.get(actual, "undefined")
     52         logger.warning("Found UsaxsSaxsMode = %s (%s)", actual, actual_str)
---> 53         raise ValueError(
     54             f"Incorrect UsaxsSaxsMode mode found ({actual}, {actual_str})."
     55             "  If SAXS, WAXS, and USAXS are out of beam, type: "

ValueError: Incorrect UsaxsSaxsMode mode found (4, WAXS in beam).  If SAXS, WAXS, and USAXS are out of beam, type:  terms.SAXS.UsaxsSaxsMode.put(1)

In [3]: 
```

jilavsky, 2021-02-04:
Hung again after 5 hours of heater profile, this time moving to SAXS

```
                                                                                                                                                                            
I Wed-23:57:21 - Found UsaxsSaxsMode = USAXS in beam
I Wed-23:57:21 - Moving to proper SAXS mode
                                                                                                                                                                            
                                                                                                                                                                            
---------------------------------------------------------------------------                                                                                                 
LimitError                                Traceback (most recent call last)                                                                                                 
~/.ipython/user/linkam_new.py in <module>                                                                                                                                   
----> 1 RE(myLinkamPlan(0,0,0.103,"Fe3Si",15))

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
     79             except StopIteration as e:
     80                 # discard the exhausted generator

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in monitor_during_wrapper(plan, signals)
    801     plan1 = plan_mutator(plan, insert_after_open)
    802     plan2 = plan_mutator(plan1, insert_before_close)
--> 803     return (yield from plan2)
    804 
    805 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
[... 147 lines cut ...]
    189             timeout = self._timeout
    190 
--> 191         self.check_value(position)
    192 
    193         self._run_subs(sub_type=self._SUB_REQ_DONE, success=False)

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/epics_motor.py in check_value(self, pos)
    234     def check_value(self, pos):
    235         '''Check that the position is within the soft limits'''
--> 236         self.user_setpoint.check_value(pos)
    237 
    238     @required_for_connection

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/signal.py in check_value(self, value)
   1563 
   1564         if not (low_limit <= value <= high_limit):
-> 1565             raise LimitError('Value {} outside of range: [{}, {}]'
   1566                              .format(value, low_limit, high_limit))
   1567 

LimitError: Value 20.0013 outside of range: [43.001, 140.0]

In [5]: 

```

jilavsky, 2021-02-04:
Situation after BS hung:
saxs_y is in 13.8, its proper position, SAXS_z is 45, which is in out of beam position. 
It hung here: yield from bps.mv(saxs_stage.z, terms.SAXS.z_in.get())
but epics limits to saxs_z are 18 to 140. We (BS or spec) do not manipulate these limits, so why does BS think the z stage limit is 43 - 140? It looks like it misread the limit and that killed it.

jilavsky, 2021-02-04:
It is getting difficult to finish experiments. I have 3 more days of these experiments in Linkam just this week.

jilavsky, 2021-02-04:
SInce this does not need X-rays, we should plan to run test on this without the beam at convenient time. I mean we should be simply able to move between different modes - it failed once going to WAXZS and once going to SAXS modes. No data collection involved at all.

prjemian, 2021-03-24:
Memory corruption?

prjemian, 2021-03-24:
@klauer, @tacaswell:  We've got a motor on the USAXS instrument that raised a limits error after a long sequence of operations, repeating the same code many times with no such error.  We have not changed the limits from python code or from the EPICS IOC.  Why would we get `LimitError: Value 20.0013 outside of range: [43.001, 140.0]` when the EPICS limits are 18 .. 140?

klauer, 2021-03-24:
What motor record support is this? At PCDS, we've had custom ones that have failed to update the setpoint limit metadata correctly. 

How sure are you that there's no client or record logic changing these values intermittently? The "corrupted" values seem too "nice"/normal that I'd imagine they're legitimately coming from _somewhere_.

If not the above, I have no clue, sorry. Would need more debugging details (perhaps `%debug` when it happens and poke around)

prjemian, 2021-03-24:
Thanks @klauer.  It *is* a subclass of `ophyd.EpicsMotor` that has a `set_lim` method.

https://github.com/APS-USAXS/ipython-usaxs/blob/f497e3867a511ac49af1800412fb8f4f2be1de74/profile_bluesky/startup/instrument/devices/usaxs_motor_devices.py#L23-L57

But we are certain that we do not want to move the limits on this axis once we have completed the initial instrument setup.

prjemian, 2021-03-24:
And looks like this local `set_lim()` method is to provide a local fix for an upstream ophyd problem that was fixed later. (https://github.com/bluesky/ophyd/issues/874)

prjemian, 2021-03-24:
Since that code is in released ophyd (I see it in 1.6.1), we can remove this local override.

klauer, 2021-03-24:
What's the underlying motor support module? Since it's at the APS, standard motor record I suppose?

prjemian, 2021-03-24:
Right, it's the EPICS motor record.  Probably synApps 5.8 but not certain.

prjemian, 2021-03-24:
We're suspicious of various memory leaks during longer operations sessions.  How well does Python defend itself against overwriting memory?

prjemian, 2021-03-24:
synApps 5.7, not sure that is involved but you are right that the wrong values look like sensible numbers, close to the correct ones.  We'll check our code for a very-hard-to-find typo.

klauer, 2021-03-24:
Python objects are located pretty randomly in memory to my knowledge (with a bunch of overhead for `PyObject` wrapping every single thing, ref counts/garbage collection tracking, etc). It's not impossible to corrupt them, but I'd expect you'd have run into something much stranger than a simple value swap here.

Here was the issue for our metadata-not-propagated device, a SLAC-customized motor record that doesn't behave 100% like the upstream one: https://github.com/pcdshub/pcdsdevices/issues/781 Since yours is 'stock', so to speak, it's likely not the issue.

prjemian, 2021-03-24:
We confirmed that our code does not call `.set_lim()` for this axis at any place.  Either the limit was changed from the EPICS side or this is a much more strange problem to resolve.

Anyone could type a bad value into a GUI screen by mistake.  At 2:17 AM during a long sequence?  No one was at the facility.  Remote staff _might_ have done this but that is also far-fetched.

klauer, 2021-03-24:
Any EPICS record logic that looks at those fields whatsoever? 

Is it defined anywhere as part of a pseudopositioner that might have calculated limits?

prjemian, 2021-03-24:
## Summary

This issue reported problems running in remote operations for long times.  First, a problem with a wrong mode value when changing instrument modes.  This was handled by changing the code that reads the instrument state.  Then, a problem with the soft limits on a motor axis where we never change the soft limits.

We suspect the root cause of this is beyond our python code.  Possibly the motor limit was changed by another EPICS client (not so likely), we don't have any code that changes the limit values.

One thought occurs, if the `motor.low_limit_switch` is logged in the descriptor document, we can check its timestamp for difference with other runs.

Our next best root cause hypothesis is memory corruption.  Needs a link to the issues we have on that from December 2019.  Both here and in ophyd repo.  Maybe another repo also.

prjemian, 2021-03-24:
Looking at the data from the last run in the databroker (Feb 5, scan_id=692), the `saxs_stage` is not in the baseline data (or the other streams).  Too bad since the limit values and time stamps would be logged there.  Add that to the baseline for next time.

prjemian, 2021-03-24:
We can't fix this.  Let's close and remember if it happens again.

closed 2021-03-24 by prjemian