APS-USAXS/usaxs-bluesky-ended-2023 #487: SampleTitle Function

jilavsky, 2021-02-03:
Need to check when is it called. Seems to me to be called too early, before potential mode change. At high heating rates mode change take too much time and then values in names do nto match reality. This is just a check and potentially move to place closer to start of exposure.

prjemian, 2022-01-24:
@jilavsky : Where to proceed?

```bash
bash-4.2$ git grep SampleTitle
profile_bluesky/startup/instrument/plans/scans.py:from ..utils.user_sample_title import getSampleTitle
profile_bluesky/startup/instrument/plans/scans.py:    title = getSampleTitle(title)
profile_bluesky/startup/instrument/plans/scans.py:    scan_title = getSampleTitle(scan_title)
profile_bluesky/startup/instrument/plans/scans.py:    scan_title = getSampleTitle(scan_title)
profile_bluesky/startup/instrument/utils/user_sample_title.py:    getSampleTitle
profile_bluesky/startup/instrument/utils/user_sample_title.py:    resetSampleTitleFunction
profile_bluesky/startup/instrument/utils/user_sample_title.py:    setSampleTitleFunction
profile_bluesky/startup/instrument/utils/user_sample_title.py:def plainSampleTitle(sample_title):
profile_bluesky/startup/instrument/utils/user_sample_title.py:    Once you have a function, call ``setSampleTitleFunction()``
profile_bluesky/startup/instrument/utils/user_sample_title.py:    with the function object.  Call ``resetSampleTitleFunction()``
profile_bluesky/startup/instrument/utils/user_sample_title.py:        setSampleTitleFunction(myTitleFunction)
profile_bluesky/startup/instrument/utils/user_sample_title.py:        >>> getSampleTitle("blank")
profile_bluesky/startup/instrument/utils/user_sample_title.py:def setSampleTitleFunction(func_object):
profile_bluesky/startup/instrument/utils/user_sample_title.py:def resetSampleTitleFunction():
profile_bluesky/startup/instrument/utils/user_sample_title.py:    _sample_title_function = plainSampleTitle
profile_bluesky/startup/instrument/utils/user_sample_title.py:def getSampleTitle(sample_title):
profile_bluesky/startup/instrument/utils/user_sample_title.py:resetSampleTitleFunction()
user/localfile.py:from instrument.utils import getSampleTitle
user/localfile.py:from instrument.utils import setSampleTitleFunction
user/localfile.py:setSampleTitleFunction(myNamer)
user/localfile.py:print(getSampleTitle(__file__))
user/localfile.py:print(getSampleTitle("purple"))
user/localfile.py:print(cleanupText(getSampleTitle("Merging can be performed automatically")))
```

jilavsky, 2022-01-25:
My suggestion is to modify code in scans.py. Example:
we call this on line 708 in SAXS and that is before yield from mode_SAXS() on line 719. 
This means, that if we are heating fast, 20+ seconds move to SAXS may be significant change in temperature. And if we trigger tune on line 717, we add even more time and hence temperature difference. 
I think the block 708-711 could move after aat least line 739. Even then we will have additional time on transmission measurement, but I am willing to take that as part of measurement. 
Similar move should be done to USAXS and WAXS, move the name creation closer to real exposure, after mode change and possible tune.

jilavsky, 2022-01-27:
I will create branch for this and make the change. Submit for review to you.

jilavsky, 2022-01-27:
Submitted pull request #540 for this

closed 2022-01-30 by jilavsky

jilavsky, 2022-01-31:
Fixe #540

PR #540 (2022-01-30): move call to getSampleTitle() closer to actual collection events
- FIXES #487

Move getSampleTitile to better locations