APS-USAXS/usaxs-bluesky-ended-2023 #489: AR_PulsePositions not written as array in HDF5 file

prjemian, 2021-02-03:
During USAXS Fly Scans, sometimes the `AR_PulsePositions` value is not written as an array of 8000 values as expected.

HDF5 address: `/entry/flyScan/AR_PulsePositions`

```
usaxs@usaxscontrol /share1/USAXS_data $ cd ./2021-02/02_03_Fancher1/02_03_Fancher1_usaxs/
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ ls -lart
total 5560
-rw-rw-r-- 1 usaxs usaxs   28872 Feb  3 15:54 HeaterBlank_0824.jpg
-rw-rw-r-- 1 usaxs usaxs   28125 Feb  3 16:11 HeaterBlank_0825.jpg
-rw-rw-r-- 1 usaxs usaxs   29992 Feb  3 16:41 Fe3Si_0826.jpg
-rw-rw-r-- 1 usaxs usaxs   29995 Feb  3 17:02 Fe3Si_33C_0min_0001.jpg
-rw-rw-r-- 1 usaxs usaxs 1358476 Feb  3 17:07 HeaterBlank_0824.h5
drwxrwxr-x 5 usaxs usaxs    4096 Feb  3 17:07 ..
-rw-rw-r-- 1 usaxs usaxs   30135 Feb  3 17:08 Fe3Si_33C_0min_0002.jpg
drwxrwxr-x 2 usaxs usaxs    4096 Feb  3 17:08 .
-rw-rw-r-- 1 usaxs usaxs 1144012 Feb  3 17:12 HeaterBlank_0825.h5
-rw-rw-r-- 1 usaxs usaxs  883140 Feb  3 17:30 Fe3Si_0826.h5
-rw-rw-r-- 1 usaxs usaxs  802940 Feb  3 17:30 Fe3Si_33C_0min_0001.h5
-rw-rw-r-- 1 usaxs usaxs  752948 Feb  3 17:30 Fe3Si_33C_0min_0002.h5
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ punx tree Fe3Si_33C_0min_0001.h5 | less
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ punx tree HeaterBlank_0824.h5 | grep AR_PulsePositions
      AR_PulsePositions:NX_FLOAT64[8000] = [8.748185199437586, 8.748157494096276, 8.748129861765703, '...', 7.131788391572308]
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ punx tree HeaterBlank_0825.h5 | grep AR_PulsePositions
      AR_PulsePositions:NX_FLOAT64[8000] = [8.748185199437586, 8.748157494096276, 8.748129861765703, '...', 7.131788391572308]
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $       AR_PulsePositions:NX_FLOAT64[8000] = [8.748185199437586, 8.748157494096276, 8.748129861765703, ^C
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ punx tree HeaterBlank_0825.h5 | grep AR_PulsePositions
      AR_PulsePositions:NX_FLOAT64[8000] = [8.748185199437586, 8.748157494096276, 8.748129861765703, '...', 7.131788391572308]
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ punx tree Fe3Si_0826.h5 | grep AR_PulsePositions
      AR_PulsePositions:NX_FLOAT64[8000] = [8.748185199437586, 8.748157494096276, 8.748129861765703, '...', 7.131788391572308]
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ punx tree Fe3Si_33C_0min_0002.h5 | grep AR_PulsePositions
      AR_PulsePositions:NX_FLOAT64 = 8.748019799437586
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ ls -lart
total 6616
-rw-rw-r-- 1 usaxs usaxs   28872 Feb  3 15:54 HeaterBlank_0824.jpg
-rw-rw-r-- 1 usaxs usaxs   28125 Feb  3 16:11 HeaterBlank_0825.jpg
-rw-rw-r-- 1 usaxs usaxs   29992 Feb  3 16:41 Fe3Si_0826.jpg
-rw-rw-r-- 1 usaxs usaxs   29995 Feb  3 17:02 Fe3Si_33C_0min_0001.jpg
-rw-rw-r-- 1 usaxs usaxs 1358476 Feb  3 17:07 HeaterBlank_0824.h5
drwxrwxr-x 5 usaxs usaxs    4096 Feb  3 17:07 ..
-rw-rw-r-- 1 usaxs usaxs   30135 Feb  3 17:08 Fe3Si_33C_0min_0002.jpg
-rw-rw-r-- 1 usaxs usaxs 1144012 Feb  3 17:12 HeaterBlank_0825.h5
-rw-rw-r-- 1 usaxs usaxs   30024 Feb  3 17:31 Fe3Si_0003.jpg
drwxrwxr-x 2 usaxs usaxs    4096 Feb  3 17:31 .
-rw-rw-r-- 1 usaxs usaxs  932292 Feb  3 17:36 Fe3Si_0826.h5
-rw-rw-r-- 1 usaxs usaxs  852092 Feb  3 17:36 Fe3Si_33C_0min_0001.h5
-rw-rw-r-- 1 usaxs usaxs  802100 Feb  3 17:36 Fe3Si_33C_0min_0002.h5
-rw-rw-r-- 1 usaxs usaxs  805604 Feb  3 17:36 Fe3Si_0003.h5
usaxs@usaxscontrol .../02_03_Fancher1/02_03_Fancher1_usaxs $ punx tree Fe3Si_0003.h5 | grep AR_PulsePositions
      AR_PulsePositions:NX_FLOAT64[8000] = [8.748019799437586, 8.747990182055233, 8.747960641291073, '...', 7.131593967973208]
```

prjemian, 2021-02-03:
Don't understand why this problem occurs.  It's random but has happened before.

How are we getting those numbers?  `some.object.get()`?  Perhaps we should `.get(auto_monitor=False)` or such   ...?

jilavsky, 2021-02-04:
Happened immediately after quit and restart of BlueSky. Happened for two fly scans. Aborted user experiment in progress and for lack of better solution, tried to quit and start BlueSky **again**. After BluSky restart fly scan data are saved correctly.  By now we collected 50+ fly scans and all I tested are correct. 
This suggest random event on start of BlueSky?

prjemian, 2021-03-24:
See also https://github.com/APS-USAXS/ipython-usaxs/issues/479#issuecomment-767868625, does not appear related.

prjemian, 2021-03-24:
This is the code that gets the data before writing:

https://github.com/APS-USAXS/ipython-usaxs/blob/ff0c3dc07ce922e9918011327b680489f4221167/profile_bluesky/startup/usaxs_support/saveFlyData.py#L107

Let's apply the same type of change used previously in https://github.com/APS-USAXS/ipython-usaxs/issues/486#issuecomment-771324856 in the `usaxs_support/saveFlyData.py` code:

```python
    pv_spec.ophyd_signal.get(timeout=10, use_monitor=False)
```

prjemian, 2021-03-24:
close now, open *new* if problem happens again

closed 2021-03-24 by prjemian