APS-USAXS/usaxs-bluesky-ended-2023 #490: external heater_profile program does not stop when requested

prjemian, 2021-02-04:
The background program does not seem to stop when bit is raised. I tried manually in GUI display and from bps.mv(terms.HeaterProcess.linkam_trigger, 1) and the pulse is still going. I assume the process must stop so next time we start it, with potentially different parameters to run, it will reread its configuration and follow proper temperature trajectory.

_Originally posted by @jilavsky in https://github.com/APS-USAXS/ipython-usaxs/issues/482#issuecomment-773264284_

prjemian, 2021-02-04:
Since we use `bps.mv` to set the temperature and `bps.sleep(hours)` to hold, neither of these will watch for the exit signal during their operation.  Switch to use a set and then a wait loop that polls.

prjemian, 2021-02-04:
If user requests exit, then heater should be stopped.

prjemian, 2021-02-04:
T96 needs to get a signal for `.stop`.

jilavsky, 2022-01-12:
Incoming [Linkam code from apstools](https://github.com/BCDA-APS/apstools/blob/main/apstools/devices/linkam_controllers.py) has temperature that includes a [configurable](https://github.com/BCDA-APS/apstools/blob/7645cf133bd6dadbd29a3a8f26f89c1fea4f4456/apstools/devices/positioner_soft_done.py#L197-L202) [`stop`](https://github.com/BCDA-APS/apstools/blob/7645cf133bd6dadbd29a3a8f26f89c1fea4f4456/apstools/devices/linkam_controllers.py#L33) method.

It would be a good idea to add unit tests of that stop method to apstools test suite.  See https://github.com/BCDA-APS/apstools/issues/618

jilavsky, 2022-01-12:
Linkam 1500 is connected., can be run and controlled. Try not to run too long at higher temperatures, at most 300C for 10-30 minutes is fine.

prjemian, 2022-01-12:
Test between 70 & 90, maybe a little higher if cooling is too slow.  Then back to RT when done.

prjemian, 2022-01-12:
WHen user sets a new setpoint, they expect the heating power to be set to ON as a consequence.

prjemian, 2022-01-12:
This is the [`actuate`](https://github.com/BCDA-APS/apstools/blob/7645cf133bd6dadbd29a3a8f26f89c1fea4f4456/apstools/devices/positioner_soft_done.py#L60-L61) feature.

prjemian, 2022-01-22:
In commit a727500 , the program https://github.com/APS-USAXS/ipython-usaxs/blob/a727500496e28db95c782e80c36579af9c7fb591/user/heater_profile.py#L224-L244

### Complete heater plan, uninterrupted

Full execution of the heater plan as shown in the log file:

```
2022-01-22 13:35:18.743: Starting planHeaterProcess() for linkam_tc1
2022-01-22 13:35:18.744: linkam_tc1 T=29.2C setpoint=40.0C ramp:20.0 settled: False done: False
2022-01-22 13:35:18.749: Set linkam_tc1 rate to 20 C/min
2022-01-22 13:35:18.913: Set linkam_tc1 setpoint to 80.00 C
2022-01-22 13:36:19.048: linkam_tc1 T=47.4C setpoint=80.0C ramp:20.0 settled: False done: False
2022-01-22 13:37:19.182: linkam_tc1 T=68.2C setpoint=80.0C ramp:20.0 settled: False done: False
2022-01-22 13:37:52.252: Done, that took 153.50s
2022-01-22 13:37:52.253: linkam_tc1 T=79.4C setpoint=80.0C ramp:20.0 settled: True done: True
2022-01-22 13:37:52.253: linkam_tc1 holding for 31.17s
2022-01-22 13:38:24.337: linkam_tc1 holding period ended
2022-01-22 13:38:24.338: linkam_tc1 T=80.1C setpoint=80.0C ramp:20.0 settled: True done: True
2022-01-22 13:38:24.343: Set linkam_tc1 rate to 20 C/min
2022-01-22 13:38:24.353: Set linkam_tc1 setpoint to 40.00 C
2022-01-22 13:39:24.486: linkam_tc1 T=61.1C setpoint=40.0C ramp:20.0 settled: False done: False
2022-01-22 13:40:24.615: linkam_tc1 T=41.9C setpoint=40.0C ramp:20.0 settled: False done: False
2022-01-22 13:40:28.624: Done, that took 124.28s
2022-01-22 13:40:28.627: linkam_tc1 T=40.9C setpoint=40.0C ramp:20.0 settled: True done: True
2022-01-22 13:40:28.628: Ending planHeaterProcess() for linkam_tc1
```

### Interrupted run

When the stop bit is set early, the profile holds at temperature and exits with heater power on.  Here's the first part of the log file:

```
2022-01-22 13:41:18.470: Starting planHeaterProcess() for linkam_tc1
2022-01-22 13:41:18.472: linkam_tc1 T=32.3C setpoint=40.0C ramp:20.0 settled: False done: False
2022-01-22 13:41:18.476: Set linkam_tc1 rate to 20 C/min
2022-01-22 13:41:18.802: Set linkam_tc1 setpoint to 80.00 C
```

Externally, user requests stop by setting the bit (in a separate console):

```bash
bash-4.2$ caput 9idcLAX:bit14 0
Old : 9idcLAX:bit14                  ON
New : 9idcLAX:bit14                  OFF
```

The log file continues ...

```
2022-01-22 13:42:07.919: User requested exit during set after 0.82m. Stopping the heater and holding at current temperature.
2022-01-22 13:42:07.922: linkam_tc1 T=46.7C setpoint=46.7C ramp:20.0 settled: True done: True
```

closed 2022-01-22 by prjemian