APS-USAXS/usaxs-bluesky-ended-2023 #491: Changing modes while beam is not availabel still hungs instrument

jilavsky, 2021-02-05:
I was in the hutch and issues RE(mode_radiography()) to get to radiography mode while I search the hutch. This caused error and instrument hung. 
No obvious way to et out. I tried enter key, nothing.. I opened the shutter, nothing. There must be simple way to get out of this!
```                                                                                                                                                        
                                                                                                                                                                            
mono_shutter [In progress. No progress bar available.]                                                                                                                      
Exception in thread Thread-111268:                                                                                                                                          
Traceback (most recent call last):                                                                                                                                          
  File "/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run() - Prepared for USAXS mode
  File "/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/apstools/devices.py", line 499, in move_it
    self.open()
  File "/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/apstools/devices.py", line 861, in open
    self.wait_for_state(
  File "/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/apstools/devices.py", line 853, in wait_for_state
    raise TimeoutError(msg)
TimeoutError: Timeout (10 s) waiting for shutter state to reach a value in [1, 'O', 'N']

```

jilavsky, 2021-02-05:
After abort I get errors, that cannot run ... shutter is still operating. Eventually:

```
I Fri-08:55:18 - USAXS is in position
                                                                                                                                                                            
---------------------------------------------------------------------------                                                                                                 
RuntimeError                              Traceback (most recent call last)                                                                                                 
~/.ipython/user/linkam_new.py in <module>                                                                                                                                   
----> 1 RE(mode_Radiography()) configured for Radiography now.

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
     79             except StopIteration as e:
     80                 # discard the exhausted generator

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in monitor_during_wrapper(plan, signals)
    801     plan1 = plan_mutator(plan, insert_after_open)
    802     plan2 = plan_mutator(plan1, insert_before_close)
--> 803     return (yield from plan2)
    804 
    805 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
[... 81 lines cut ...]
   1423                         # exceptions (coming in via throw) can be
   1424                         # raised
-> 1425                         new_response = await coro(msg)
   1426 
   1427                     # special case `CancelledError` and let the outer

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _set(self, msg)
   1919         group = kwargs.pop('group', None)
   1920         self._movable_objs_touched.add(msg.obj)
-> 1921         ret = msg.obj.set(*msg.args, **kwargs)
   1922         p_event = asyncio.Event(loop=self._loop_for_kwargs)
   1923         pardon_failures = self._pardon_failures

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/apstools/devices.py in set(self, value, **kwargs)
    481         """
    482         if self.busy.get():
--> 483             raise RuntimeError("shutter is operating")
    484 
    485         __value__ = self.lowerCaseString(value)

RuntimeError: shutter is operating

In [25]: 

```

prjemian, 2021-02-05:
Looking through the [`ShutterBase` source code](https://github.com/BCDA-APS/apstools/blob/main/apstools/devices.py#L323), the `busy` attribute is only changed back to `True` during the [`move_it()`](https://github.com/BCDA-APS/apstools/blob/main/apstools/devices.py#L495-L503) method:

```python
            def move_it():
                # runs in a thread, no need to "yield from"
                self.busy.put(True)
                if __value__ in self.valid_open_values:
                    self.open()
                elif __value__ in self.valid_close_values:
                    self.close()
                self.busy.put(False)
                status._finished(success=True)
```

So, a way to change the `busy` attribute (until the code in apstools is changed) is this command line:

```python
mono_shutter.busy.put(False)
```

prjemian, 2021-02-05:
Related to #456, tried to open `mono_shutter` when doors were open.  That issue was _resolved_ by adding a watch for `diagnostics.BL_EPS.station_shutter_b_permit.get() not in (1, 'GOOD')`. 

How did the fail reported here escape this test?

prjemian, 2021-02-05:
Here is the context:  https://github.com/APS-USAXS/ipython-usaxs/blob/160dbdf35809b6e4bfe06dcd50bdd159a1c09460/profile_bluesky/startup/instrument/plans/mode_changes.py#L283-L289

It failed with Timeout during execution of line 287: `yield from bps.mv(...` which requires thus shutter to reach its destination.

We *could* replace `bps.mv(` with `bps.abs_set(` but we would have to do the wait ourselves.  Do it?

prjemian, 2021-02-05:
The issue title says _while beam is not available_, was this the case?  Doesn't `diagnostics.BL_EPS.station_shutter_b_permit` already consider this or is it an additional test to be applied?

prjemian, 2021-02-05:
According to the APS status plot, beam was stored.
[image]

prjemian, 2021-02-05:
```python
In [1]: diagnostics.BL_EPS.station_shutter_b_permit.get()
Out[1]: 'GOOD'

In [2]: listdevice(diagnostics.BL_EPS)
================================================= ================== ==========================
name                                              value              timestamp
================================================= ================== ==========================
diagnostics_BL_EPS_red_light                      0                  2021-02-05 14:28:25.206554
diagnostics_BL_EPS_station_shutter_b_permit       GOOD               2021-02-05 14:28:26.144825
diagnostics_BL_EPS_station_shutter_b              OPENED             2021-02-05 14:28:25.206556
diagnostics_BL_EPS_flow_1                         1.38824462890625   2021-02-05 14:43:41.260901
diagnostics_BL_EPS_flow_2                         1.8227159976959229 2021-02-05 14:43:41.260944
diagnostics_BL_EPS_flow_1_setpoint                1.0                2021-02-05 14:28:25.151613
diagnostics_BL_EPS_flow_2_setpoint                1.0                2021-02-05 14:28:25.151616
diagnostics_BL_EPS_temperature_1_chopper          87.07781982421875  2021-02-05 14:43:41.260977
diagnostics_BL_EPS_temperature_2                  85.54721069335938  2021-02-05 14:43:41.260985
diagnostics_BL_EPS_temperature_3                  85.07667541503906  2021-02-05 14:43:41.260992
diagnostics_BL_EPS_temperature_4                  82.8914794921875   2021-02-05 14:43:41.265839
diagnostics_BL_EPS_temperature_5                  244.6383056640625  2021-02-05 14:43:41.265891
diagnostics_BL_EPS_temperature_6                  277.1344299316406  2021-02-05 14:43:41.265897
diagnostics_BL_EPS_temperature_7                  268.125244140625   2021-02-05 14:43:41.265939
diagnostics_BL_EPS_temperature_8                  279.990478515625   2021-02-05 14:43:41.265985
diagnostics_BL_EPS_temperature_1_chopper_setpoint 93.0               2021-02-05 14:28:25.151643
diagnostics_BL_EPS_temperature_2_setpoint         93.0               2021-02-05 14:28:25.151646
diagnostics_BL_EPS_temperature_3_setpoint         93.0               2021-02-05 14:28:25.156260
diagnostics_BL_EPS_temperature_4_setpoint         95.0               2021-02-05 14:28:25.156296
diagnostics_BL_EPS_temperature_5_setpoint         270.0              2021-02-05 14:28:25.156300
diagnostics_BL_EPS_temperature_6_setpoint         300.0              2021-02-05 14:28:25.156303
diagnostics_BL_EPS_temperature_7_setpoint         300.0              2021-02-05 14:28:25.156305
diagnostics_BL_EPS_temperature_8_setpoint         310.0              2021-02-05 14:28:25.156308
diagnostics_BL_EPS_shutter_permit                 Permit OK          2021-02-05 08:56:40.589459
diagnostics_BL_EPS_vacuum_permit                  BL Vac Permit OK   2021-02-05 08:56:40.589464
diagnostics_BL_EPS_vacuum_ok                      BL Vacuum Bad      2021-02-05 08:56:40.589463
================================================= ================== ==========================
```

prjemian, 2021-02-05:
from  :

ophyd symbol | PV | meaning
--- | --- | ---
diagnostics.BL_EPS.station_shutter_b | 9idBLEPS:SBS_PERMIT | B shutter is permitted
diagnostics.BL_EPS.shutter_permit | EPS:09:ID:BLEPS:SPER | BL EPS shutter permit

```
In [7]: !caget 9idBLEPS:SBS_PERMIT
9idBLEPS:SBS_PERMIT            GOOD

In [8]: !caget 9idBLEPS:SBS_PERMIT.DESC
9idBLEPS:SBS_PERMIT.DESC       Station B Shutter Permit

In [9]: !cainfo 9idBLEPS:SBS_PERMIT
9idBLEPS:SBS_PERMIT
    State:            connected
    Host:             usaxsserver.xray.aps.anl.gov:43186
    Access:           read, write
    Native data type: DBF_ENUM
    Request type:     DBR_ENUM
    Element count:    1

In [4]: !caget EPS:09:ID:BLEPS:SPER
EPS:09:ID:BLEPS:SPER           Permit OK

In [5]: !cainfo EPS:09:ID:BLEPS:SPER
EPS:09:ID:BLEPS:SPER
    State:            connected
    Host:             s9pvgate.xray.aps.anl.gov:5064
    Access:           read, no write
    Native data type: DBF_ENUM
    Request type:     DBR_ENUM
    Element count:    1
In [6]: !caget EPS:09:ID:BLEPS:SPER.DESC
EPS:09:ID:BLEPS:SPER.DESC      BL-EPS Shutter Perm
```