APS-USAXS/usaxs-bluesky-ended-2023 #492: FLyscnas forgot to close something

jilavsky, 2021-02-06:
```
D Fri-20:28:18 - 35.03        8.6683773    -0.28769     11.50446     1396         29.85      
D Fri-20:28:23 - 40.04        8.6248533    -0.44458     10.82022     1850         34.87      
D Fri-20:28:28 - 45.05        8.5700076    -0.64207     9.95264      2328         39.87      
D Fri-20:28:33 - 50.05        8.5032643    -0.88347     8.88453      2830         44.87      
D Fri-20:28:38 - 55.07        8.4223988    -1.17466     7.58649      3347         49.88      
D Fri-20:28:43 - 60.07        8.3289151    -1.51615     6.09071      3876         54.88      
D Fri-20:28:48 - 65.08        8.2186101    -1.90948     4.33214      4416         59.88      
D Fri-20:28:53 - 70.08        8.0955738    -2.35257     2.37107      4965         64.90      
D Fri-20:28:58 - 75.09        7.9532405    -2.86293     0.09521      5522         69.92      
D Fri-20:29:03 - 80.10        7.7968991    -3.42646     -2.38472     6083         74.92      
D Fri-20:29:08 - 85.10        7.6191767    -4.06539     -5.21430     6650         79.92      
D Fri-20:29:13 - 90.11        7.4235736    -4.76856     -8.27609     7222         84.93      
D Fri-20:29:18 - 95.12        7.2132803    -5.52439     -11.67791    7796         89.93      
D Fri-20:29:22 - writing fly scan HDF5 file: /share1/USAXS_data/2021-02/02_05_Albern/02_05_Albern_usaxs/X13112_0153.h5
I Fri-20:29:22 - 99.01        7.0996510    -5.89606     -13.32695    7999         93.78      
D Fri-20:29:22 - 99.01030778884888s - progress_reporting is done
I Fri-20:29:22 - finished writing fly scan HDF5 file: /share1/USAXS_data/2021-02/02_05_Albern/02_05_Albern_usaxs/X13112_0153.h5
I Fri-20:29:22 - HDF5 output complete: /share1/USAXS_data/2021-02/02_05_Albern/02_05_Albern_usaxs/X13112_0153.h5
New stream: 'mca'                                                                                                                                                           
D Fri-20:29:23 - after return: 99.31456303596497s
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
I Fri-20:29:25 - memory report: pmem(rss=1797873664, vms=6404853760, shared=65175552, text=2506752, lib=0, data=5775667200, dirty=0)
                                                                                                                                                                            
                                                                                                                                                                            
I Fri-20:29:28 - self._handler: Nonemode
                                                                                                                                                                            
                                                                                                                                                                            
D Fri-20:29:32 - gain=10000000.0  rate: 124560.0  max: 950000  converged=[True, True]
D Fri-20:29:32 - gain=100000.0  rate: 171540.0  max: 950000  converged=[True, True, True, True]
                                                                                                                                                                            
                                                                                                                                                                            
I Fri-20:29:38 - Measured USAXS transmission values:
======== ============= ==========
detector counts        gain      
======== ============= ==========
pinDiode 513584.000000 100000.0  
I0       372848.000000 10000000.0
                                                                                                                                                                            
                                                                                                                                                                            
D Fri-20:29:39 - gain=10000.0  rate: 373020.0  max: 950000  converged=[True, True]
D Fri-20:29:39 - gain=10000000.0  rate: 362440.0  max: 950000  converged=[True, True, True, True]
D Fri-20:29:39 - gain=1000000000.0  rate: 2160.0  max: 950000  converged=[True, True, True, True, True, True]
                                                                                                                                                                            
I Fri-20:29:40 - sample image file: /share1/USAXS_data/2021-02/02_05_Albern/02_05_Albern_usaxs/X13113_0154.jpg                                                              
                                                                                                                                                                            
Transient Scan ID: 681     Time: 2021-02-05 20:29:40
Persistent Unique Scan ID: '5d7c3cf8-c4b2-46e2-b39f-3b9fd9973673'
New stream: 'baseline'
New stream: 'aps_current_monitor'
I Fri-20:29:40 - HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
I Fri-20:29:40 - HDF5 output: /share1/USAXS_data/2021-02/02_05_Albern/02_05_Albern_usaxs/X13113_0154.h5
D Fri-20:29:40 - progress_reporting has arrived
E Fri-20:29:40 - Exception RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['160', '184.18', '4', 'X13113'], 297, '      USAXSscan        160      184.18      4      "X13113"')''
subject: RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['160', '184.18', '4', 'X13113'], 297, '      USAXSscan        160      184.18      4      "X13113"')''

date: 2021-02-05 20:29:40.473941
command file: /share1/USAXS_data/2021-02/usaxs.mac
line number: 297
command: ('USAXSscan', ['160', '184.18', '4', 'X13113'], 297, '      USAXSscan        160      184.18      4      "X13113"')
raw command:       USAXSscan        160      184.18      4      "X13113"
attempt: 1 of 5
exception: Another set() call is still in progress for usaxs_flyscan_flying
W Fri-20:29:40 - Exception while reporting instrument state: Status(obj=EpicsSignal(read_pv='9idcLAX:state', name='user_data_state', parent='user_data', value='Moving USAXS to USAXS mode', timestamp=1612578580.498969, auto_monitor=True, string=True, write_pv='9idcLAX:state', limits=False, put_complete=False), done=True, success=False)
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
[... 373 lines cut ...]
2021-02-05 20:40:25: auto_collect is waiting for next command from EPICS ...
D Fri-20:40:25 - starting user commands
I Fri-20:40:25 - Command file: short.mac
====== ======================= ==========================
line # action                  parameters                
====== ======================= ==========================
1      CURRENT_EXPERIMENT_NAME AXT_SET_MEASUREMENT       
18     USAXSscan               140, 184.18, 4, X13010    
19     USAXSscan               150, 184.18, 4, X13112    
20     USAXSscan               160, 184.18, 4, X13113    
21     USAXSscan               170, 184.18, 4, X13114    
22     USAXSscan               180, 184.18, 4, X13115    
23     USAXSscan               190, 184.18, 4, X13116    
24     USAXSscan               200, 184.18, 4, X13117    
25     USAXSscan               30, 24.181, 4, Water_Blank
====== ======================= ==========================

I Fri-20:40:25 - memory report: pmem(rss=1807007744, vms=6421635072, shared=65191936, text=2506752, lib=0, data=5792448512, dirty=0)
W Fri-20:40:25 - Exception during execution of command short.mac:
A 'close_run' message was not received before the 'open_run' message
I Fri-20:40:25 - waiting for next user command
2021-02-05 20:41:09: auto_collect is waiting for next command from EPICS ...
I Fri-20:41:09 - auto_collect is ending
Out[3]: 
```

jilavsky, 2021-02-06:
Tried to stop remote ops and start again, this is error:

 Fri-20:47:08 - 0.03         8.7489334    0.00385      12.80534     0            0.00       
E Fri-20:47:08 - 0.0317997932434082s - progress_reporting timeout!!
D Fri-20:47:13 - 5.00         8.7490898    0.00441      12.80782     0            0.00

prjemian, 2021-02-06:
Don't even know why this last one.  Ignoring for now.

Sample line 160 triggered illegal message sequence.  Read there an error
before that?  Look at the command file for problem near that line.

jilavsky, 2021-02-06:
Command file was generated using Igor code and I could not find any error there. 
Igor sanitizes up user typos anyway. Code should be really clean there.

jilavsky, 2021-02-06:
First e-mail error message:
subject: RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['160', '184.18', '4', 'X13113'], 297, '      USAXSscan        160      184.18      4      "X13113"')''

date: 2021-02-05 20:29:40.473941
command file: /share1/USAXS_data/2021-02/usaxs.mac
line number: 297
command: ('USAXSscan', ['160', '184.18', '4', 'X13113'], 297, '      USAXSscan        160      184.18      4      "X13113"')
raw command:       USAXSscan        160      184.18      4      "X13113"
attempt: 1 of 5
exception: Another set() call is still in progress for usaxs_flyscan_flying

jilavsky, 2021-02-06:
Note, data files in question:
X13112_0153.h5 is fine 
X13113_0154.h5 is missing mca1 and is unusable.

prjemian, 2021-03-24:
as @tacaswell [noted](https://github.com/bluesky/ophyd/pull/755#issuecomment-521403838): 

> To "fix" it you can do `obj._set_thread = None`.

prjemian, 2021-03-24:
In our case: `usaxs_flyscan_flying._set_thread = None`

prjemian, 2021-03-24:
# Aha!

We stopped the remote plan and tried to restart.  The`usaxs_flyscan.flying` signal was not reset by any cleanup operation (do we have any?) so it assumed the flyscan had not yet completed.

prjemian, 2021-03-24:
Related to #255

prjemian, 2021-03-24:
First exception, our internal signal was in the wrong state during a long sequence of the same operations:

```python
E Fri-20:29:40 - Exception RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['160', '184.18', '4', 'X13113'], 297, '      USAXSscan        160      184.18      4      "X13113"')''
subject: RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['160', '184.18', '4', 'X13113'], 297, '      USAXSscan        160      184.18      4      "X13113"')''

date: 2021-02-05 20:29:40.473941
command file: /share1/USAXS_data/2021-02/usaxs.mac
line number: 297
command: ('USAXSscan', ['160', '184.18', '4', 'X13113'], 297, '      USAXSscan        160      184.18      4      "X13113"')
raw command:       USAXSscan        160      184.18      4      "X13113"
attempt: 1 of 5
exception: Another set() call is still in progress for usaxs_flyscan_flying
W Fri-20:29:40 - Exception while reporting instrument state: Status(obj=EpicsSignal(read_pv='9idcLAX:state', name='user_data_state', parent='user_data', value='Moving USAXS to USAXS mode', timestamp=1612578580.498969, auto_monitor=True, string=True, write_pv='9idcLAX:state', limits=False, put_complete=False), done=True, success=False)
```

How was this situation triggered?  We don't know but it may be related to #486 where we suspect something beyond our python code to be trouble.

Later 4 exception were from the retry algorithm (different exception: IllegalMessageSequence) and due to failed run not properly closed

closed 2021-03-24 by prjemian