APS-USAXS/usaxs-bluesky-ended-2023 #493: Suspendor is still not working right

jilavsky, 2021-02-09:
After Monday shut down, 
Instrument is in OK state and everything is correct - not issues, when using spec little bit later, no issues, no errors. 

Tried to use BS first, this happens:

```
In [5]: Unexpected problem with CA circuit to server "ioc9idclax.xray.aps.anl.gov:5064" was "Connection reset by peer" - disconnecting
CA.Client.Exception...............................................
    Warning: "Virtual circuit disconnect"
    Context: "ioc9idclax.xray.aps.anl.gov:5064"
    Source File: ../cac.cpp line 1236
    Current Time: Mon Feb 08 2021 14:30:56.998294396
..................................................................
Suspender SuspendFloor(EpicsSignalRO(read_pv='S:SRcurrentAI', name='aps_current', parent='aps', value=10.129474027091332, timestamp=1612843008.939857, auto_monitor=True, string=False), sleep=100, pre_plan=None, post_plan=None,tripped_message=) reports a return to nominal conditions. Will sleep for 100 seconds and then release suspension at 2021-02-08 21:58:28.
Suspender SuspendFloor(EpicsSignalRO(read_pv='S:SRcurrentAI', name='aps_current', parent='aps', value=10.017592091091332, timestamp=1612845047.939863, auto_monitor=True, string=False), sleep=100, pre_plan=None, post_plan=None,tripped_message=) reports a return to nominal conditions. Will sleep for 100 seconds and then release suspension at 2021-02-08 22:32:27.
Suspender SuspendFloor(EpicsSignalRO(read_pv='S:SRcurrentAI', name='aps_current', parent='aps', value=10.009337443091331, timestamp=1612846782.939817, auto_monitor=True, string=False), sleep=100, pre_plan=None, post_plan=None,tripped_message=) reports a return to nominal conditions. Will sleep for 100 seconds and then release suspension at 2021-02-08 23:01:22.
Suspender SuspendFloor(EpicsSignalRO(read_pv='S:SRcurrentAI', name='aps_current', parent='aps', value=10.110665895091334, timestamp=1612866623.439893, auto_monitor=True, string=False), sleep=100, pre_plan=None, post_plan=None,tripped_message=) reports a return to nominal conditions. Will sleep for 100 seconds and then release suspension at 2021-02-09 04:32:03.
Suspender SuspendFloor(EpicsSignalRO(read_pv='S:SRcurrentAI', name='aps_current', parent='aps', value=10.105815323091331, timestamp=1612873371.939832, auto_monitor=True, string=False), sleep=100, pre_plan=None, post_plan=None,tripped_message=) reports a return to nominal conditions. Will sleep for 100 seconds and then release suspension at 2021-02-09 06:24:31.
In [5]: 

In [5]: 

In [5]: 

In [5]: RE(mode_SAXS())
At least one suspender has tripped. The plan will begin when all suspenders are ready. Justification:
    1. Signal plc_protect_operations_status, got "1", expected "1"  .
    Significant equipment problem.  Do these steps:
    1. ^C twice      # interrupt the ipython kernel
    2. RE.abort()    # finalize current data streams (if any)
    3. exit          # quit bluesky session
    4. call beamline scientists

    

Suspending... To get to the prompt, hit Ctrl-C twice to pause.
^CA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.
^Ctrying a second time
Pausing...
---------------------------------------------------------------------------
RunEngineInterrupted                      Traceback (most recent call last)
<ipython-input-5-37fdfe140a05> in <module>
----> 1 RE(mode_SAXS())

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    808 
    809         if self._interrupted:
--> 810             raise RunEngineInterrupted(self.pause_msg) from None
    811 
    812         return tuple(self._run_start_uids)

RunEngineInterrupted: 
Your RunEngine is entering a paused state. These are your options for changing
the state of the RunEngine:

RE.resume()    Resume the plan.
RE.abort()     Perform cleanup, then kill plan. Mark exit_stats='aborted'.
RE.stop()      Perform cleanup, then kill plan. Mark exit_status='success'.
RE.halt()      Emergency Stop: Do not perform cleanup --- just stop.

In [6]: RE.abort()
Aborting: running cleanup and marking exit_status as 'abort'...
Out[6]: ()

In [7]: 

```

I suspect Monday shut down caused suspender to engage and it simply stayed engaged? We need to fix this at some point.

prjemian, 2021-02-09:
Right.  The suspender engaged *Current Time: Mon Feb 08 2021 14:30:56.998294396*.  Curious that it did not reset when beam came back at ~8 AM today.

[image]

The last report from the suspender was at 6:22:51 AM today (timestamp=1612873371.939832):

```
Suspender SuspendFloor(EpicsSignalRO(read_pv='S:SRcurrentAI', name='aps_current', parent='aps', value=10.105815323091331, timestamp=1612873371.939832, auto_monitor=True, string=False), sleep=100, pre_plan=None, post_plan=None,tripped_message=) reports a return to nominal conditions. Will sleep for 100 seconds and then release suspension at 2021-02-09 06:24:31.
```

Note, timestamp converted into human terms:

```python
In [15]: import datetime

In [16]: datetime.datetime.fromtimestamp(1612873371.939832)
Out[16]: datetime.datetime(2021, 2, 9, 6, 22, 51, 939832)

In [17]: datetime.datetime.fromtimestamp(1612873371.939832).isoformat(sep=" ")
Out[17]: '2021-02-09 06:22:51.939832'
```

prjemian, 2021-02-09:
According to the log file:

```
# Fri, 05 Feb 2021 08:57:32
RE(auto_collect.remote_ops())
#[Out]# ('2f83c476-375c-4cb3-a797-f230837e4a68',
#[Out]#  '6cd8e3eb-617c-4e54-aac3-dc2d45f7a712',
#[Out]#  '77f6a60b-f221-428e-8323-2c6742cad945',
#[Out]#  'd47ea29a-4604-4d07-8e3a-e76ba51e02f4',
#[Out]#  'ea2bf44b-4dad-4f5f-84d2-67ac2c7644cd',
#[Out]#  '183b69cd-3b00-45c7-a941-300e44e4a520',
#[Out]#  '781a83a6-1379-4a2c-a2a8-5a906be2ec02',
#[Out]#  'b787931c-fa36-4c8e-8da5-a0f134ff7ed8',
#[Out]#  'fb13ee3d-d796-4f16-8e46-84c8336c2a4c',
#[Out]#  '4ed6189b-3b50-46d2-8782-65ba35758e36',
#[Out]#  '28f1ba6f-21ac-4e16-af21-5b0185288ccb',
#[Out]#  'abf82a3e-da27-4958-923a-3c3fa477f47c',
/09 Feb
...skipping
#[Out]#  '82b68817-f9eb-41a1-87d0-4f0395840dcd',
#[Out]#  '2ab19ea8-e223-4050-939e-70f896cb958f')
# Tue, 09 Feb 2021 09:03:22
RE(mode_SAXS())
# Tue, 09 Feb 2021 09:03:32
RE.abort()
#[Out]# ()
```

You started a new acquisition at 9:03 this morning, when the `aps.current` was ~100 mA, long after the value had exceed the 10 mA   settings for `resume_thresh=10, sleep=100`.

https://github.com/APS-USAXS/ipython-usaxs/blob/03fe162a3cf6099e880ae51a3a47c95da2bf24e0/profile_bluesky/startup/instrument/devices/suspenders.py#L26-L28

jilavsky, 2021-03-24:
User error, works as expected. Needs restart after mode change. Next time, make sure were record when BS was restarted so we can evaluate.

closed 2021-03-24 by jilavsky