APS-USAXS/usaxs-bluesky-ended-2023 #494: rebuild linkam support using best practices

prjemian, 2021-03-24:
Related problems appeared in these issues, (#484, #485, #487, #490).  Time to rebuild this support and make it rely on common tools that *any* APS beam line can use.

Make sure that the same support can be used at other beam lines.

Look at PTC10, Linkam, Lakeshore, and others.  (4ID-Polar and 8IDI)

jilavsky, 2021-05-23:
I have been looking at linkam.py and various user plans we have and I am more and more confused. My recollection is, that we had two support versions. I cannot figure this all out as I think I see only one support version, linkam.py. Does this one needs to be modified still? 
I need to make sure only one version is available (new one) and user plans are all converted or deleted.

jilavsky, 2022-01-12:
This will happen with installation of new release (1.6.0rc5+) of apstools.

prjemian, 2022-01-21:
New support in apstools 1.6.0 (https://github.com/BCDA-APS/apstools/pull/560 and https://github.com/BCDA-APS/apstools/pull/625)

prjemian, 2022-01-21:
Note:

```python
# make a common term for the ramp rate (devices use different names)
linkam_ci94.ramp = linkam_ci94.rate
linkam_tc1.ramp = linkam_tc1.ramprate
```

prjemian, 2022-01-21:
Note

### change temperature and wait to reach `inposition`, use `linkam.temperature`

* plan:  `yield from bps.mv(linkam.temperature, value)`
  * TC96: also turns on the heater power
  * CI94: ?
* interactive: `linkam.temperature.move(value)
  * does not turn on heater power

### change temperature and do not wait to reach `inposition`, use `linkam.temperature.setpoint`

* plan:  `yield from bps.mv(linkam.temperature.setpoint, value)`
* interactive: `linkam.temperature.setpoint.put(value)

prjemian, 2022-01-21:
I hope this plan will be much easier to edit: https://github.com/APS-USAXS/ipython-usaxs/blob/87ec787bb2cad97eb2a00ebb7abf99e7f7f5ba18/user/heater_profile.py#L202-L218

prjemian, 2022-01-21:
See also #482

prjemian, 2022-01-22:
Instances of "linkam" (case-independent) in the repo now:

<details>

```bash
examples/README.md:* [Use a supported Linkam heater](use_linkam.md)
examples/custom_loop.md:    yield from bps.mv(linkam.rate, 100)			# degrees C/minute
examples/custom_loop.md:    yield from linkam.set_target(temperature, wait=True)	# degrees C
examples/use_linkam.md:# Use a supported Linkam heater
examples/use_linkam.md:There are two Linkam sample heaters already configured for use 
examples/use_linkam.md:`linkam_ci94` | Linkam_CI94 (the old one) | `9idcLAX:ci94:`
examples/use_linkam.md:`linkam_tc1`  | Linkam_T96 | `9idcLINKAM:tc1:`
examples/use_linkam.md:In the examples below, replace `linkam_tc1` with the name of your temperature controller if it is different.
examples/use_linkam.md:### `.target.value` : What temperature is `linkam_tc1` set to? 
examples/use_linkam.md:    linkam_tc1.target.value
examples/use_linkam.md:### `.set_target()` : Set `linkam_tc1` to a specific temperature
examples/use_linkam.md:    linkam_tc1.set_target(new_temperature)
examples/use_linkam.md:By default, this will wait until `linkam_tc1` is *settled* before it returns.
examples/use_linkam.md:### `.value` : What is the `linkam_tc1` actual temperature now?
examples/use_linkam.md:    print(linkam_tc1.value)
examples/use_linkam.md:    print(linkam_tc1.signal.value)
examples/use_linkam.md:### `.settled` : Has `linkam_tc1` reached the desired temperature?
examples/use_linkam.md:    print(linkam_tc1.settled)
examples/use_linkam.md:How close must the actual temperature (signal) be to the target temperature for `linkam_tc1` to indicate it is *settled*?
examples/use_linkam.md:    print(linkam_tc1.settled)
examples/use_linkam.md:From the command line: `linkam_tc1.tolerance.put(2)` (sets it to 2 degrees)
examples/use_linkam.md:From a plan: `yield from bps.mv(linkam_tc1.tolerance, 2)`
examples/use_linkam.md:    yield from bps.mv(linkam_tc1.rate, 100)            # degrees C/minute
examples/use_linkam.md:        yield from linkam_tc1.set_target(temperature, wait=True)    # degrees C
examples/use_linkam.md:        md["temperature_actual"] = linkam_tc1.value
examples/use_linkam.md:        md["temperature_actual"] = linkam_tc1.value
examples/use_linkam.md:        md["temperature_actual"] = linkam_tc1.value
profile_bluesky/startup/instrument/devices/__init__.py:from .linkam import *
profile_bluesky/startup/instrument/devices/general_terms.py:    linkam_exit = Component(EpicsSignal, "9idcLAX:bit14")
profile_bluesky/startup/instrument/devices/general_terms.py:    linkam_pulse = Component(EpicsSignal, "9idcLAX:long16")
profile_bluesky/startup/instrument/devices/general_terms.py:    linkam_ready = Component(EpicsSignal, "9idcLAX:bit15")
profile_bluesky/startup/instrument/devices/general_terms.py:    linkam_trigger = Component(EpicsSignal, "9idcLAX:bit16")
profile_bluesky/startup/instrument/devices/linkam.py:Linkam temperature controllers
profile_bluesky/startup/instrument/devices/linkam.py:    'linkam_ci94',
profile_bluesky/startup/instrument/devices/linkam.py:    'linkam_tc1',
profile_bluesky/startup/instrument/devices/linkam.py:class Linkam_CI94(UsaxsProcessController):
profile_bluesky/startup/instrument/devices/linkam.py:    Linkam model CI94 temperature controller
profile_bluesky/startup/instrument/devices/linkam.py:        In [1]: linkam_ci94 = Linkam_CI94("9idcLAX:ci94:", name="ci94")
profile_bluesky/startup/instrument/devices/linkam.py:        In [2]: linkam_ci94.settled
profile_bluesky/startup/instrument/devices/linkam.py:        In [3]: linkam_ci94.settled
profile_bluesky/startup/instrument/devices/linkam.py:        linkam_ci94.record_signal()
profile_bluesky/startup/instrument/devices/linkam.py:        yield from (linkam_ci94.set_target(50))
profile_bluesky/startup/instrument/devices/linkam.py:    controller_name = "Linkam CI94"
profile_bluesky/startup/instrument/devices/linkam.py:class Linkam_T96(UsaxsProcessController):
profile_bluesky/startup/instrument/devices/linkam.py:    Linkam model T96 temperature controller
profile_bluesky/startup/instrument/devices/linkam.py:        linkam_tc1 = Linkam_T96("9idcLINKAM:tc1:", name="linkam_tc1")
profile_bluesky/startup/instrument/devices/linkam.py:    controller_name = "Linkam T96"
profile_bluesky/startup/instrument/devices/linkam.py:linkam_ci94 = Linkam_CI94("9idcLAX:ci94:", name="ci94")
profile_bluesky/startup/instrument/devices/linkam.py:linkam_tc1 = Linkam_T96("9idcLINKAM:tc1:", name="linkam_tc1")
profile_bluesky/startup/usaxs_support/heater_profile_manager.sh:# a Linkam heater sequence: ./heater_profile_manager.py.
profile_bluesky/startup/usaxs_support/heater_profile_process.py:    linkam_exit = Component(EpicsSignal, "9idcLAX:bit14")
profile_bluesky/startup/usaxs_support/heater_profile_process.py:    linkam_pulse = Component(EpicsSignal, "9idcLAX:long16")
profile_bluesky/startup/usaxs_support/heater_profile_process.py:    linkam_ready = Component(EpicsSignal, "9idcLAX:bit15")
profile_bluesky/startup/usaxs_support/heater_profile_process.py:    linkam_trigger = Component(EpicsSignal, "9idcLAX:bit16")
profile_bluesky/startup/usaxs_support/heater_profile_process.py:    pulse0 = process_control.linkam_pulse.get()
profile_bluesky/startup/usaxs_support/heater_profile_process.py:    pulses = (process_control.linkam_pulse.get() - pulse0) % PULSE_MAX
profile_bluesky/startup/usaxs_support/heater_profile_process.py:            process_control.linkam_pulse.put(
[... cut ...]
user/old/linkam_new.py:    logger.info("Triggering (starting) the External Linkam heating plan ...")
user/old/linkam_new.py:    yield from bps.mv(terms.HeaterProcess.linkam_trigger, 1)
user/old/linkam_new.py:    # tell the Linkam control python program to exit...
user/old/linkam_new.py:    logger.info("Stopping the External Linkam heating plan ...")
user/old/linkam_new.py:    yield from bps.mv(terms.HeaterProcess.linkam_exit, 1)  # orderly
```

</details>

Updating the issue description above with the list of files.

closed 2022-01-22 by prjemian