APS-USAXS/usaxs-bluesky-ended-2023 #497: run into suspender

jilavsky, 2021-04-09:
Run into suspender. 
Searched hutch, did not open B shutter, but waited long enough that B shutter could be opened by code (spec opened). Created newUser and tried to go to Radiography. 
plc protection was not engaged, could move instrument around in epics and spec. 
spec had not problems going to radiography  

Tried RE.abort() but see below, still did not recover, could not do anything. Eventually quit BS and restarted. 
Calling myself was not helpful, still confused ;-)
And used spec while recording this. 

Now, there was moment 16+ hours ago when PLC protection was engaged and reset by me. That was last afternoon and we run samples over night in spec. 

Is PLC signal plc_protect_operations_status sticky this way??? 

```
In [16]: RE(mode_Radiography())
At least one suspender has tripped. The plan will begin when all suspenders are ready. Justification:
    1. Signal plc_protect_operations_status, got "1", expected "1"  .
    Significant equipment problem.  Do these steps:
    1. ^C twice      # interrupt the ipython kernel
    2. RE.abort()    # finalize current data streams (if any)
    3. exit          # quit bluesky session
    4. call beamline scientists

    

Suspending... To get to the prompt, hit Ctrl-C twice to pause.
^CA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.
^Ctrying a second time
Pausing...
---------------------------------------------------------------------------
RunEngineInterrupted                      Traceback (most recent call last)
<ipython-input-16-890e5f2ff449> in <module>
----> 1 RE(mode_Radiography())

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    808 
    809         if self._interrupted:
--> 810             raise RunEngineInterrupted(self.pause_msg) from None
    811 
    812         return tuple(self._run_start_uids)

RunEngineInterrupted: 
Your RunEngine is entering a paused state. These are your options for changing
the state of the RunEngine:

RE.resume()    Resume the plan.
RE.abort()     Perform cleanup, then kill plan. Mark exit_stats='aborted'.
RE.stop()      Perform cleanup, then kill plan. Mark exit_status='success'.
RE.halt()      Emergency Stop: Do not perform cleanup --- just stop.

In [17]: RE.resume()
^CA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.
^Ctrying a second time
Pausing...
---------------------------------------------------------------------------
RunEngineInterrupted                      Traceback (most recent call last)
<ipython-input-17-53ef60f855c5> in <module>
----> 1 RE.resume()

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in resume(self)
    845         self._resume_task()
    846         if self._interrupted:
--> 847             raise RunEngineInterrupted(self.pause_msg) from None
    848         return tuple(self._run_start_uids)
    849 

RunEngineInterrupted: 
Your RunEngine is entering a paused state. These are your options for changing
the state of the RunEngine:

RE.resume()    Resume the plan.
RE.abort()     Perform cleanup, then kill plan. Mark exit_stats='aborted'.
RE.stop()      Perform cleanup, then kill plan. Mark exit_status='success'.
RE.halt()      Emergency Stop: Do not perform cleanup --- just stop.

In [18]: RE.abort
Out[18]: <bound method RunEngine.abort of <bluesky.run_engine.RunEngine object at 0x7f7786076df0>>

In [19]: RE(mode_Radiography())
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-19-890e5f2ff449> in <module>
----> 1 RE(mode_Radiography())

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    752         # If we are in the wrong state, raise.
    753         if not self._state.is_idle:
--> 754             raise RuntimeError("The RunEngine is in a %s state" % self._state)
    755 
    756         futs = []

RuntimeError: The RunEngine is in a paused state

In [20]: RE.abort()
Aborting: running cleanup and marking exit_status as 'abort'...
Out[20]: ()

In [21]: 

In [21]: 

In [21]: RE(mode_Radiography())
At least one suspender has tripped. The plan will begin when all suspenders are ready. Justification:
    1. Signal plc_protect_operations_status, got "1", expected "1"  .
    Significant equipment problem.  Do these steps:
    1. ^C twice      # interrupt the ipython kernel
    2. RE.abort()    # finalize current data streams (if any)
    3. exit          # quit bluesky session
    4. call beamline scientists

    

Suspending... To get to the prompt, hit Ctrl-C twice to pause.
^CA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.
^Ctrying a second time
Pausing...
---------------------------------------------------------------------------
RunEngineInterrupted                      Traceback (most recent call last)
<ipython-input-21-890e5f2ff449> in <module>
----> 1 RE(mode_Radiography())

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    808 
    809         if self._interrupted:
--> 810             raise RunEngineInterrupted(self.pause_msg) from None
    811 
    812         return tuple(self._run_start_uids)

RunEngineInterrupted: 
Your RunEngine is entering a paused state. These are your options for changing
the state of the RunEngine:

RE.resume()    Resume the plan.
RE.abort()     Perform cleanup, then kill plan. Mark exit_stats='aborted'.
RE.stop()      Perform cleanup, then kill plan. Mark exit_status='success'.
RE.halt()      Emergency Stop: Do not perform cleanup --- just stop.

In [22]: exit
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ blueskyUSAXS 
Python 3.8.2 (default, Mar 26 2020, 15:53:00) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.19.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: bluesky
Activating auto-logging. Current session state plus future input saved.
Filename       : /share1/USAXS_data/2021-04/.logs/ipython_console.log
Mode           : rotate

```

jilavsky, 2021-04-10:
Still same issue after restart, cannot run at all:
```
I Fri-08:23:06 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py
I Fri-08:23:06 - bluesky startup is complete

In [1]: 

In [1]: 

In [1]: 

In [1]: newUser("Beaucage2")
I Sat-08:10:51 - Creating user directory: /share1/USAXS_data/2021-04/04_10_Beaucage2
I Sat-08:10:51 - Change working directory is still /share1/USAXS_data/2021-04
I Sat-08:10:51 - SPEC file name : /share1/USAXS_data/2021-04/04_10_Beaucage2/04_10_Beaucage2.dat
I Sat-08:10:51 - File will be created at end of next bluesky scan.
W Sat-08:10:52 - No unexpired ESAFs found that match user Beaucage2
W Sat-08:10:53 - No unexpired proposals found that match user Beaucage2
W Sat-08:10:53 - APSBSS not updated.
W Sat-08:10:53 - You should check that PVs in APSBSS contain correct information.
Out[1]: '/share1/USAXS_data/2021-04/04_10_Beaucage2'

In [2]: RE(mode_Radiography())
At least one suspender has tripped. The plan will begin when all suspenders are ready. Justification:
    1. Signal plc_protect_operations_status, got "1", expected "1"  .
    Significant equipment problem.  Do these steps:
    1. ^C twice      # interrupt the ipython kernel
    2. RE.abort()    # finalize current data streams (if any)
    3. exit          # quit bluesky session
    4. call beamline scientists

    

Suspending... To get to the prompt, hit Ctrl-C twice to pause.

```

prjemian, 2021-04-12:
It's a shutdown day.  Testing the signal:

```python
In [1]: plc_protect.operations_status.get()
Out[1]: 1

In [2]: type(_)
Out[2]: int

In [3]: plc_protect.operations_status.enum_strs
Out[3]: ('Low', 'High')
```

Not understanding why this report is a suspender condition:

```python
    1. Signal plc_protect_operations_status, got "1", expected "1"  .
```

----

Looking at the active suspenders now:

```python
In [4]: RE.suspenders
Out[4]: (PlcProtectSuspendWhenChanged(EpicsSignal(read_pv='9idcLAX:plc:Y0', name='plc_protect_operations_status', parent='plc_protect', value=1, timestamp=1617975543.908513, auto_monitor=True, string=False, write_pv='9idcLAX:plc:Y0', limits=False, put_complete=False), sleep=0, pre_plan=None, post_plan=None,tripped_message=),)
```

There's only one active suspender (this one) and it expects to receive an integer `1`:

```python
In [22]: RE.suspenders[0].expected_value
Out[22]: 1

In [23]: type(RE.suspenders[0].expected_value)
Out[23]: int
```

prjemian, 2021-04-12:
Still not understanding why this fails when `1 == 1`.  Possibly ... it registered a failure due to a previous CA monitor when PLC's `Y0` output relay went to zero.  Since this suspender is designed to compel a session restart by design, even when we check it later, after the `Y0` comes back to 1, it still remembers that intermediate state.  Just a theory.  BUT, if that's the case, there are two possibilities:

1. restart the bluesky session
2. redesign the suspender

Since you've demonstrated that *1* does not clear the problem, we need to advance to *2*.  Agree?

jilavsky, 2021-05-22:
Agree here. Even though as noted, it is highly suspicious that 1 does not solve the issue. Where do we have "memory" of this state in between the sessions? Suspenders are complicated, they need to be. But this is downright weird.

prjemian, 2021-06-15:
I believe there might be a problem with "late reporting" by bluesky.  (The thread reports long after it is necessary but only at next update of signal.)