APS-USAXS/usaxs-bluesky-ended-2023 #498: BS does not start

jilavsky, 2021-04-10:
Following prior suspender issue, I hit ctrl-C until BS died and I was in command line.

Cannot start... 

```
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ blueskyUSAXS 
Python 3.8.2 (default, Mar 26 2020, 15:53:00) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.19.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: bluesky
Activating auto-logging. Current session state plus future input saved.
Filename       : /share1/USAXS_data/2021-04/.logs/ipython_console.log
Mode           : rotate
Output logging : True
Raw input log  : False
Timestamping   : True
State          : active
I Sat-08:12:39 - ############################################################ startup
I Sat-08:12:39 - logging started
I Sat-08:12:39 - logging level = 10
I Sat-08:12:39 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/collection.py
I Sat-08:12:39 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/mpl/console.py
I Sat-08:12:41 - bluesky framework
I Sat-08:12:41 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/framework/check_python.py
I Sat-08:12:47 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/framework/initialize.py
[TerminalIPythonApp] WARNING | Unknown error in handling startup files:
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/IPython/core/shellapp.py in _exec_file(self, fname, shell_futures)
    375                     else:
    376                         # default to python, even without extension
--> 377                         self.shell.safe_execfile(full_filename,
    378                                                  self.shell.user_ns,
    379                                                  shell_futures=shell_futures,

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/IPython/core/interactiveshell.py in safe_execfile(self, fname, exit_ignore, raise_exceptions, shell_futures, *where)
   2738             try:
   2739                 glob, loc = (where + (None, ))[:2]
-> 2740                 py3compat.execfile(
   2741                     fname, glob, loc,
   2742                     self.compile if shell_futures else None)

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/IPython/utils/py3compat.py in execfile(fname, glob, loc, compiler)
    166     with open(fname, 'rb') as f:
    167         compiler = compiler or compile
--> 168         exec(compiler(f.read(), fname, 'exec'), glob, loc)
    169 
    170 # Refactor print statements in doctests.

~/.ipython/profile_bluesky/startup/00-instrument.py in <module>
      4 """
      5 
----> 6 from instrument.collection import *
      7 
      8 # show_ophyd_symbols()

~/.ipython/profile_bluesky/startup/instrument/collection.py in <module>
     11 logger.info("bluesky framework")
     12 
---> 13 from .framework import *
     14 from .devices import *
     15 from .callbacks import *

~/.ipython/profile_bluesky/startup/instrument/framework/__init__.py in <module>
      2 from .check_bluesky import *
      3 
----> 4 from .initialize import *
      5 from .user_dir import *
      6 from .metadata import *

~/.ipython/profile_bluesky/startup/instrument/framework/initialize.py in <module>
     75 # Set up a RunEngine and use metadata backed PersistentDict
     76 RE = RunEngine({})
---> 77 RE.md = PersistentDict(md_path)
     78 if old_md is not None:
     79     logger.info("migrating RE.md storage to PersistentDict")

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/utils.py in __init__(self, directory)
    763         self._cache = {}
    764         super().__init__(self._dump, self._load, self._file)
--> 765         self.reload()
    766 
    767         # Similar to flush() or _do_update(), but without reference to self

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/utils.py in reload(self)
    818     def reload(self):
    819         """Force a reload from disk, overwriting current cache"""
--> 820         self._cache = dict(super().items())
    821 
    822 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/zict/func.py in <genexpr>(.0)
     54 
     55     def items(self):
---> 56         return ((k, self.load(v)) for k, v in self.d.items())
     57 
     58     def _do_update(self, items):

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/utils.py in _load(file)
    806     @staticmethod
    807     def _load(file):
--> 808         return msgpack.unpackb(
    809             file,
    810             object_hook=msgpack_numpy.decode,

msgpack/_unpacker.pyx in msgpack._cmsgpack.unpackb()

ValueError: Unpack failed: incomplete input

In [1]: 
```

jilavsky, 2021-04-10:
quit BS (whatever was running, try to start again. No luck, same messages

```
In [1]: exit
(/share1/bluesky/conda_envs/bluesky_2021_1) bash-4.2$ blueskyUSAXS 
Python 3.8.2 (default, Mar 26 2020, 15:53:00) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.19.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: bluesky
Activating auto-logging. Current session state plus future input saved.
Filename       : /share1/USAXS_data/2021-04/.logs/ipython_console.log
Mode           : rotate
Output logging : True
Raw input log  : False
Timestamping   : True
State          : active
I Sat-08:14:49 - ############################################################ startup
I Sat-08:14:49 - logging started
I Sat-08:14:49 - logging level = 10
I Sat-08:14:49 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/collection.py
I Sat-08:14:49 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/mpl/console.py
I Sat-08:14:49 - bluesky framework
I Sat-08:14:49 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/framework/check_python.py
I Sat-08:14:49 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/framework/initialize.py
[TerminalIPythonApp] WARNING | Unknown error in handling startup files:
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/IPython/core/shellapp.py in _exec_file(self, fname, shell_futures)
    375                     else:
    376                         # default to python, even without extension
--> 377                         self.shell.safe_execfile(full_filename,
    378                                                  self.shell.user_ns,
    379                                                  shell_futures=shell_futures,

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/IPython/core/interactiveshell.py in safe_execfile(self, fname, exit_ignore, raise_exceptions, shell_futures, *where)
   2738             try:
   2739                 glob, loc = (where + (None, ))[:2]
-> 2740                 py3compat.execfile(
   2741                     fname, glob, loc,
   2742                     self.compile if shell_futures else None)

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/IPython/utils/py3compat.py in execfile(fname, glob, loc, compiler)
    166     with open(fname, 'rb') as f:
    167         compiler = compiler or compile
--> 168         exec(compiler(f.read(), fname, 'exec'), glob, loc)
    169 
    170 # Refactor print statements in doctests.

~/.ipython/profile_bluesky/startup/00-instrument.py in <module>
      4 """
      5 
----> 6 from instrument.collection import *
      7 
      8 # show_ophyd_symbols()

~/.ipython/profile_bluesky/startup/instrument/collection.py in <module>
     11 logger.info("bluesky framework")
     12 
---> 13 from .framework import *
     14 from .devices import *
     15 from .callbacks import *

~/.ipython/profile_bluesky/startup/instrument/framework/__init__.py in <module>
      2 from .check_bluesky import *
      3 
----> 4 from .initialize import *
      5 from .user_dir import *
      6 from .metadata import *

~/.ipython/profile_bluesky/startup/instrument/framework/initialize.py in <module>
     75 # Set up a RunEngine and use metadata backed PersistentDict
     76 RE = RunEngine({})
---> 77 RE.md = PersistentDict(md_path)
     78 if old_md is not None:
     79     logger.info("migrating RE.md storage to PersistentDict")

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/utils.py in __init__(self, directory)
    763         self._cache = {}
    764         super().__init__(self._dump, self._load, self._file)
--> 765         self.reload()
    766 
    767         # Similar to flush() or _do_update(), but without reference to self

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/utils.py in reload(self)
    818     def reload(self):
    819         """Force a reload from disk, overwriting current cache"""
--> 820         self._cache = dict(super().items())
    821 
    822 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/zict/func.py in <genexpr>(.0)
     54 
     55     def items(self):
---> 56         return ((k, self.load(v)) for k, v in self.d.items())
     57 
     58     def _do_update(self, items):

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/utils.py in _load(file)
    806     @staticmethod
    807     def _load(file):
--> 808         return msgpack.unpackb(
    809             file,
    810             object_hook=msgpack_numpy.decode,

msgpack/_unpacker.pyx in msgpack._cmsgpack.unpackb()

ValueError: Unpack failed: incomplete input

In [1]: 

```

jilavsky, 2021-04-10:
back to spec...

prjemian, 2021-04-12:
Thanks for the error dump, it is helpful. Looks like the problem happens while initializing the Bluesky Framework.  Either some package version conflict I/we installed or some change I/we made to the initialization steps.  I'll walk through that line by line.  This diagnostic should not affect you.

------------

Got my suspicions here

```python
msgpack/_unpacker.pyx in msgpack._cmsgpack.unpackb()
ValueError: Unpack failed: incomplete input
In [1]: !conda list msgpack
# packages in environment at /share1/bluesky/conda_envs/bluesky_2021_1:
#
# Name                    Version                   Build  Channel
msgpack-numpy             0.4.7.1            pyhd3eb1b0_0  
msgpack-python            1.0.1            py38hff7bd54_0  
suitcase-msgpack          0.2.4                      py_0    conda-forge
```

Looks like a package version problem ... digging deeper.  Alternative is to install a new conda environment and expect this problem to vanish.  That's my second choice.  Time to update the environment requirements for next run anyway.

prjemian, 2021-04-12:
bluesky package version (1.6.7) is the latest, so are msgpack-numpy and suitcase-msgpack.  msgpack-python latest is 1.0.2 but installed is 1.0.1.  Will check zict package next (it's involved in the error output).

zict 2.0.0 is latest, installed

prjemian, 2021-04-12:
Not really obvious where this problem originates.  Easiest step now is to install new environment and try it, since that does not modify any existing code of yours.  And, it is an alternative (trying it has no adverse consequences).

prjemian, 2021-04-12:
Noting that it fails when restoring data saved from previous session by [`PersistentDict()`](https://blueskyproject.io/bluesky/metadata.html?highlight=persistentdict#persistence-between-sessions), will next examine the saved copy of session data.  That might be the real problem here.

```python
In [19]: md_path
Out[19]: '/home/beams/USAXS/.config/Bluesky_RunEngine_md'
```

prjemian, 2021-04-12:
That's a directory.  Renamed it so that next start will create a new one with current defaults [(and reset the `scan_id` back to 1)](https://blueskyproject.io/bluesky/metadata.html?highlight=persistentdict#special-fields).

Now, `blueskyUSAXS` starts.

closed 2021-04-12 by prjemian