APS-USAXS/usaxs-bluesky-ended-2023 #499: Crash on 04-14-21

jilavsky, 2021-04-14:
We were running larger sequence of saxs, waxs, and USAXS scans. Command file generated from Igor macro, running unattended while staff and all personnel at home. Instrument at some point started to send (350) e-mails with scans failing. Following are records from command line:

```
D Tue-23:27:57 - writing fly scan HDF5 file: /share1/USAXS_data/2021-04/04_13_Hu4/04_13_Hu4_usaxs/ND_FB_U__0258.h5
I Tue-23:27:57 - 99.14        7.1009570    -5.89689     -13.26881    7999         93.85      
D Tue-23:27:57 - 99.13744187355042s - progress_reporting is done
I Tue-23:27:57 - finished writing fly scan HDF5 file: /share1/USAXS_data/2021-04/04_13_Hu4/04_13_Hu4_usaxs/ND_FB_U__0258.h5
I Tue-23:27:57 - HDF5 output complete: /share1/USAXS_data/2021-04/04_13_Hu4/04_13_Hu4_usaxs/ND_FB_U__0258.h5
New stream: 'mca'                                                                                                                                                           
D Tue-23:27:57 - after return: 99.46606206893921s
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
I Tue-23:28:00 - memory report: pmem(rss=1874190336, vms=6621495296, shared=65900544, text=2506752, lib=0, data=5992308736, dirty=0)
                                                                                                                                                                            
                                                                                                                                                                            
I Tue-23:28:03 - self._handler: Nonemode
                                                                                                                                                                            
                                                                                                                                                                            
D Tue-23:28:07 - gain=10000000.0  rate: 110280.0  max: 950000  converged=[True, True]
D Tue-23:28:07 - gain=100000.0  rate: 138440.0  max: 950000  converged=[True, True, True, True]
                                                                                                                                                                            
                                                                                                                                                                            
I Tue-23:28:14 - Measured USAXS transmission values:
======== ============= ==========
detector counts        gain      
======== ============= ==========
pinDiode 414823.000000 100000.0  
I0       330424.000000 10000000.0
                                                                                                                                                                            
                                                                                                                                                                            
D Tue-23:28:14 - gain=10000.0  rate: 253200.0  max: 950000  converged=[True, True]
D Tue-23:28:14 - gain=10000000.0  rate: 320880.0  max: 950000  converged=[True, True, True, True]
D Tue-23:28:14 - gain=1000000000.0  rate: 2140.0  max: 950000  converged=[True, True, True, True, True, True]
                                                                                                                                                                            
I Tue-23:28:15 - sample image file: /share1/USAXS_data/2021-04/04_13_Hu4/04_13_Hu4_usaxs/ND_FB_M__0259.jpg                                                                  
                                                                                                                                                                            
Transient Scan ID: 234     Time: 2021-04-13 23:28:15
Persistent Unique Scan ID: '9ebba3f7-39af-46f4-a0dd-b0d1e57add7e'
New stream: 'baseline'
New stream: 'aps_current_monitor'
I Tue-23:28:16 - HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
I Tue-23:28:16 - HDF5 output: /share1/USAXS_data/2021-04/04_13_Hu4/04_13_Hu4_usaxs/ND_FB_M__0259.h5
D Tue-23:28:16 - progress_reporting has arrived
E Tue-23:28:16 - Exception RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['60', '160', '0.75', 'ND_FB_M_'], 214, '      USAXSscan        60      160      0.75      "ND_FB_M_"')''
subject: RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['60', '160', '0.75', 'ND_FB_M_'], 214, '      USAXSscan        60      160      0.75      "ND_FB_M_"')''

date: 2021-04-13 23:28:16.144191
command file: /share1/USAXS_data/2021-04/usaxs.mac
line number: 214
command: ('USAXSscan', ['60', '160', '0.75', 'ND_FB_M_'], 214, '      USAXSscan        60      160      0.75      "ND_FB_M_"')
raw command:       USAXSscan        60      160      0.75      "ND_FB_M_"
attempt: 1 of 5
exception: Another set() call is still in progress for usaxs_flyscan_flying
W Tue-23:28:16 - Exception while reporting instrument state: Status(obj=EpicsSignal(read_pv='9idcLAX:state', name='user_data_state', parent='user_data', value='Moving USAXS to USAXS mode', timestamp=1618374496.183794, auto_monitor=True, string=True, write_pv='9idcLAX:state', limits=False, put_complete=False), done=True, success=False)
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
                                                                                                                                                                            
D Tue-23:28:22 - gain=10000000.0  rate: 110900.0  max: 950000  converged=[True, True]
                                                                                                                                                                            
D Tue-23:28:22 - gain=10000000.0  rate: 110880.0  max: 950000  converged=[True, True]
                                                                                                                                                                            
D Tue-23:28:22 - gain=10000000.0  rate: 110080.0  max: 950000  converged=[True, True]
D Tue-23:28:22 - gain=10000000.0  rate: 172860.0  max: 950000  converged=[True, True, True, True]
                                                                                                                                                                            
                                                                                                                                                                            
[... 13 lines cut ...]
                                                                                                                                                                            
D Tue-23:30:00 - gain=10000000000.0  rate: 20.0  max: 950000  converged=[False, True]
D Tue-23:30:00 - gain=10000000.0  rate: 316620.0  max: 950000  converged=[False, True, True, True]
[... cut ...]
command: ('USAXSscan', ['60', '160', '0.75', 'ND_FB_M_'], 214, '      USAXSscan        60      160      0.75      "ND_FB_M_"')
raw command:       USAXSscan        60      160      0.75      "ND_FB_M_"
attempt: 2 of 5
exception: A 'close_run' message was not received before the 'open_run' message
I Tue-23:30:03 - Preparing for USAXS mode ... please wait ...
I Tue-23:30:04 - Prepared for USAXS mode
                                                                                                                                                                            
```

and records

jilavsky, 2021-04-14:
[MacFileCrashedBS.txt](https://github.com/APS-USAXS/ipython-usaxs/files/6311579/MacFileCrashedBS.txt)
Attached is command file which was running while this happened. Last successful run was: /share1/USAXS_data/2021-04/04_13_Hu4/04_13_Hu4_usaxs/ND_FB_U__0258.h5

prjemian, 2021-04-14:
## Forensics

The `RuntimeError` was reported

```python
E Tue-23:28:16 - Exception RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['60', '160', '0.75', 'ND_FB_M_'], 214, '      USAXSscan        60      160      0.75      "ND_FB_M_"')''
subject: RuntimeError during attempt 1 of 5 of command '('USAXSscan', ['60', '160', '0.75', 'ND_FB_M_'], 214, '      USAXSscan        60      160      0.75      "ND_FB_M_"')''
```

during this run
```python
Transient Scan ID: 234     Time: 2021-04-13 23:28:15
Persistent Unique Scan ID: '9ebba3f7-39af-46f4-a0dd-b0d1e57add7e'
```

Here's what the databroker shows for the runs with `scan_id` 233 & 234:

```python
In [12]: db = databroker.catalog["mongodb_config"].v1

In [13]: db.v2[234].metadata["start"]["plan_name"]
Out[13]: 'Flyscan'

In [14]: db.v2[233].metadata["start"]["plan_name"]
Out[14]: 'Flyscan'

In [15]: db.v2[233]
Out[15]: 
BlueskyRun
  uid='34636145-ab0a-4d18-9bb5-83e3b55b8759'
  exit_status='success'
  2021-04-13 23:26:18.124 -- 2021-04-13 23:27:57.886
  Streams:
    * aps_current_monitor
    * mca
    * baseline

In [16]: db.v2[234]
Out[16]: 
BlueskyRun
  uid='9ebba3f7-39af-46f4-a0dd-b0d1e57add7e'
  exit_status='success'
  2021-04-13 23:28:15.927 -- 2021-04-13 23:46:49.924
  Streams:
    * aps_current_monitor
    * baseline
```

Run 233 was the last successful run.  It has metadata matching the [above mention](https://github.com/APS-USAXS/ipython-usaxs/issues/499#issuecomment-819575142): ` 'sample_image_name': '/share1/USAXS_data/2021-04/04_13_Hu4/04_13_Hu4_usaxs/ND_FB_U__0258.jpg'`

So it is also curious why the `exit_status='success'` of run 234.  It is missing the `mca` data stream.  This was marked as `success` since the `RuntimeException` did not render the run as `failed`.  This should be investigated after the original problem is solved.

prjemian, 2021-04-14:
The `RuntimeError` comes from *Another ` .set()` in progress* on `usaxs_flyscan.flying`: https://github.com/APS-USAXS/ipython-usaxs/blob/09ef7a1d578e3bbf638b456ec6278cb14f8b27cf/profile_bluesky/startup/instrument/devices/usaxs_fly_scan.py#L47

This is an `ophyd.Signal()` which means that it is not EPICS.  The instrument package controls this in `devices/usaxs_fly_scan.py`: 

1. cleared when setting up for the flyscan: https://github.com/APS-USAXS/ipython-usaxs/blob/09ef7a1d578e3bbf638b456ec6278cb14f8b27cf/profile_bluesky/startup/instrument/devices/usaxs_fly_scan.py#L187 
2. set when flyscan is started and cleared immediately after: https://github.com/APS-USAXS/ipython-usaxs/blob/09ef7a1d578e3bbf638b456ec6278cb14f8b27cf/profile_bluesky/startup/instrument/devices/usaxs_fly_scan.py#L204-L207

We can't tell from the logs _**which**_ of these calls to set `self.flying` have caused the `RuntimeError`.

prjemian, 2021-04-14:
One possible way this happens:

1. Line 187 executes but since signal is already `False`, the status object for the `abs_set()` never completes.
2. Line 204 is called but the `.set()` from line 187 is still in progress.  Bang, Error!

Another way (far less likely I believe):

1. Line 206 is called, then flyscan ends and Line 207 is called.  How can the status object from this fail?
2. *If* it fails, then next flyscan starts and call at Line 187 fails.

Not likely third way (disproven by lack of `mca` stream in run 234):

1. Line 204 called but status object does not complete
2. fly scan starts
3. Line 207 fails

But why is there no MCA data?

Most likely is the first case above.

prjemian, 2021-04-14:
Why would this fail in this way, after so many successful runs?  Got to be a different problem that triggers this.

prjemian, 2021-04-14:
The `flying` Signal is only used to control the progress reporting.  Run 233 showed the progress reporting output while run 234 only showed that *progress_reporting has arrived* and no output after that.  This confirms the first hypothesis was correct: The status object from line 187 fails to complete.

Refactor to use a local attribute and not a status object, this is a boolean that only controls reporting.

Still, there must be some other problem that triggers this failure.

prjemian, 2021-04-14:
Also, create a new issue label for suspicious problem to track issues that we suspect have some other origin.

prjemian, 2021-04-14:
Note from [`bluesky.utils`](https://github.com/bluesky/bluesky/blob/8ab75871c05236dea9d3141baa9cfe1cc6d99278/bluesky/utils/__init__.py#L159-L171):

```python
class SignalHandler:
    """Context manager for signal handing
    If multiple signals come in quickly, they may not all be seen, quoting
    the libc manual:
      Remember that if there is a particular signal pending for your
      process, additional signals of that same type that arrive in the
      meantime might be discarded. For example, if a SIGINT signal is
      pending when another SIGINT signal arrives, your program will
      probably only see one of them when you unblock this signal.
    https://www.gnu.org/software/libc/manual/html_node/Checking-for-Pending-Signals.html
```

prjemian, 2021-04-14:
Suggestion, insert before line 204:

```python
if self.flying._status is not None and not self.flying._status.done:
    # per https://github.com/APS-USAXS/ipython-usaxs/issues/499
    self.flying._status.set_finished()
```

prjemian, 2021-04-14:
**Or**, refactor from a boolean `Signal` to a `Status` object for this directly.  An example of such use is in a demo notebook developed for 4-ID Polar: https://nbviewer.jupyter.org/github/BCDA-APS/use_bluesky/blob/main/lessons/linux_command_as_Device/demo_doodle.ipynb

basic workflow:

1. create the Status object
2. start the process_reporting thread (reports when busy=True, exits when status=done)
3. start the busy record (set busy=True)
4. wait for busy to go False
5. call status.set_finished()

This status object could generate timeout, if desired.

jilavsky, 2021-05-22:
Where are we on this issue? This is critical for June operations to be solved. It is also way above my BS skills.

prjemian, 2021-06-04:
With #501 complete, we can resume testing on this for the original problem.

prjemian, 2021-06-04:
We believe this issue is resolved now.

closed 2021-06-04 by prjemian