APS-USAXS/usaxs-bluesky-ended-2023 #501: Unknown issue collecting data

jilavsky, 2021-06-03:
Start of 2021-02, running RE(run_command_file("usaxs.mac")) with standard spec compatible static USAXS/SAXS/WAXS list.

Failure occurred on USAXS flyscan.

'''
I Wed-21:47:13 - sample image file: /share1/USAXS_data/2021-06/06_02_BStest/06_02_BStest_usaxs/AirBlank_0071.jpg                                                           
                                                                                                                                                                           
Transient Scan ID: 13     Time: 2021-06-02 21:47:13
Persistent Unique Scan ID: '8a374d72-e46c-43d1-a4e0-29534618490b'
New stream: 'baseline'
New stream: 'aps_current_monitor'
W Wed-21:47:13 - Was not flying but should be.
I Wed-21:47:13 - HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
I Wed-21:47:13 - HDF5 output: /share1/USAXS_data/2021-06/06_02_BStest/06_02_BStest_usaxs/AirBlank_0071.h5
D Wed-21:47:13 - progress_reporting has arrived
E Wed-21:47:13 - Exception AttributeError during attempt 1 of 5 of command '('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')''
subject: AttributeError during attempt 1 of 5 of command '('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')''

date: 2021-06-02 21:47:13.342227
command file: /share1/USAXS_data/2021-06/usaxs.mac
line number: 29
command: ('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')
raw command:       USAXSscan        20      20      1      "AirBlank"
attempt: 1 of 5
exception: 'Signal' object has no attribute '_status'
W Wed-21:47:13 - Exception while reporting instrument state: Status(obj=EpicsSignal(read_pv='9idcLAX:state', name='user_data_state', parent='user_data', value='Moving USAXS to USAXS mode', timestamp=1622688433.3738918, auto_monitor=True, string=True, write_pv='9idcLAX:state', limits=False, put_complete=False), done=True, success=False)
                                                                                                                                                                           
'''

jilavsky, 2021-06-03:
Following errors are slightly different:
'''
I Wed-21:47:26 - Measured USAXS transmission values:
======== ============= ==========
detector counts        gain      
======== ============= ==========
pinDiode 616152.000000 100000.0  
I0       214598.000000 10000000.0
                                                                                                                                                                           
                                                                                                                                                                           
D Wed-21:47:26 - gain=10000.0  rate: 439500.0  max: 950000  converged=[True, True]
D Wed-21:47:26 - gain=10000000.0  rate: 206100.0  max: 950000  converged=[True, True, True, True]
D Wed-21:47:26 - gain=1000000000.0  rate: 2320.0  max: 950000  converged=[True, True, True, True, True, True]
                                                                                                                                                                           
I Wed-21:47:30 - sample image file: /share1/USAXS_data/2021-06/06_02_BStest/06_02_BStest_usaxs/AirBlank_0071.jpg                                                           
E Wed-21:47:30 - **Exception IllegalMessageSequence during attempt 2 of 5 of command '('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')''
subject: IllegalMessageSequence during attempt 2 of 5 of command '('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')''**

date: 2021-06-02 21:47:30.109030
command file: /share1/USAXS_data/2021-06/usaxs.mac
line number: 29
command: ('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')
raw command:       USAXSscan        20      20      1      "AirBlank"
attempt: 2 of 5
**exception: A 'close_run' message was not received before the 'open_run' message**
I Wed-21:47:30 - Preparing for USAXS mode ... please wait ...
I Wed-21:47:30 - Prepared for USAXS mode
                                                                                                                                                                           
                                                                                                                                                                           
                                                                                                                                                                           
                                                                                                                                                                           
D Wed-21:47:35 - gain=10000000.0  rate: 71540.0  max: 950000  converged=[True, True]
D Wed-21:47:35 - gain=100000.0  rate: 205380.0  max: 950000  converged=[True, True, True, True]
                                                                                                                                                                           
                                                                                                                                                                           
I Wed-21:47:41 - Measured USAXS transmission values:
======== ============= ==========
detector counts        gain      
======== ============= ==========
pinDiode 615456.000000 100000.0  
I0       214360.000000 10000000.0
                                                                                                                                                                           
                                                                                                                                                                           
D Wed-21:47:42 - gain=10000.0  rate: 440580.0  max: 950000  converged=[True, True]
D Wed-21:47:42 - gain=10000000.0  rate: 207040.0  max: 950000  converged=[True, True, True, True]
D Wed-21:47:42 - gain=1000000000.0  rate: 2300.0  max: 950000  converged=[True, True, True, True, True, True]
                                                                                                                                                                           
I Wed-21:47:45 - sample image file: /share1/USAXS_data/2021-06/06_02_BStest/06_02_BStest_usaxs/AirBlank_0071.jpg                                                           
E Wed-21:47:45 - **Exception IllegalMessageSequence during attempt 3 of 5 of command '('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')''
subject: IllegalMessageSequence during attempt 3 of 5 of command '('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')''**

date: 2021-06-02 21:47:45.110076
command file: /share1/USAXS_data/2021-06/usaxs.mac
line number: 29
command: ('USAXSscan', ['20', '20', '1', 'AirBlank'], 29, '      USAXSscan        20      20      1      "AirBlank"')
raw command:       USAXSscan        20      20      1      "AirBlank"
attempt: 3 of 5
exception: A 'close_run' message was not received before the 'open_run' message
I Wed-21:47:45 - Preparing for USAXS mode ... please wait ...
I Wed-21:47:45 - Prepared for USAXS mode
'''

jpg images were collected.
Hardware is up and running, spec had no issues for whole night.

jilavsky, 2021-06-03:
Observation:
When started **spec** Flyscan after this BS issue, it took VERY long time to start first flyscan. It was likely recalculating trajectories and uploading to drivers, but that usually takes ~50 sec, this time it was more like 2 minutes. No idea why. But could we have had some issues with Flyscan trajectories? 
I assume flyscan is epics controlled and I was fly scanning in spec before trying to run BS. I do not recall that change between spec and BS would require upload of trajectories. But if flyscan was not running for some **epics** reason for longer time, would that provoke timeout or something in BS?

prjemian, 2021-06-03:
Part of the problem is that the `UsaxsFlyScanDevice().flying` instance of `ophyd.Signal` does not have a `_status` attribute so the logic behind this code is broken:  https://github.com/APS-USAXS/ipython-usaxs/blob/0f7f38fa97aca37da8f5ebc94f51e91cea29e612/profile_bluesky/startup/instrument/devices/usaxs_fly_scan.py#L208-L211

jilavsky, 2021-06-03:
Looking at `.flying.set()`, it is supposed to have the `._status` attribute so what is the problem?

<details>

```python
Signature: usaxs_flyscan.flying.set(value, *, timeout=None, settle_time=None)
Source:   
    def set(self, value, *, timeout=None, settle_time=None):
        '''Set is like `put`, but is here for bluesky compatibility

        Returns
        -------
        st : Status
            This status object will be finished upon return in the
            case of basic soft Signals
        '''
        self.log.info(
            'set(value=%s, timeout=%s, settle_time=%s)',
            value, timeout, settle_time
        )

        def set_thread():
            try:
                self._set_and_wait(value, timeout)
            except TimeoutError:
                success = False
                self.log.warning(
                    '%s: set_and_wait(value=%s, timeout=%s, atol=%s, rtol=%s)',
                    self.name, value, timeout, self.tolerance, self.rtolerance
                )
            except Exception:
                success = False
                self.log.exception(
                    '%s: set_and_wait(value=%s, timeout=%s, atol=%s, rtol=%s)',
                    self.name, value, timeout, self.tolerance, self.rtolerance
                )
            else:
                success = True
                self.log.info(
                    '%s: set_and_wait(value=%s, timeout=%s, atol=%s, rtol=%s) succeeded => %s',
                    self.name, value, timeout, self.tolerance, self.rtolerance, self._readback)

                if settle_time is not None:
                    self.log.info('settling for %d seconds', settle_time)
                    time.sleep(settle_time)
            finally:
                # keep a local reference to avoid any GC shenanigans
                th = self._set_thread
                # these two must be in this order to avoid a race condition
                self._set_thread = None
                st._finished(success=success)
                del th

        if self._set_thread is not None:
            raise RuntimeError('Another set() call is still in progress '
                               f'for {self.name}')

        st = Status(self)
        self._status = st
        self._set_thread = self.cl.thread_class(target=set_thread)
        self._set_thread.daemon = True
        self._set_thread.start()
        return self._status
File:      /share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/signal.py
Type:      method
```

</details>

prjemian, 2021-06-04:
`flying is defined here: https://github.com/APS-USAXS/ipython-usaxs/blob/0f7f38fa97aca37da8f5ebc94f51e91cea29e612/profile_bluesky/startup/instrument/devices/usaxs_fly_scan.py#L47

The `_status` attribute is created in the `ophyd.Signal` class in the [`set()` method](https://github.com/bluesky/ophyd/blob/master/ophyd/signal.py#L322-L323):

```python
        st = Status(self)
        self._status = st
```

It is entirely possible that a `Signal` instance does have a `_status` attribute if the `Signal.set()` method has not yet been called.  Can't rely on it existing.

prjemian, 2021-06-04:
proof of that observation:

```python
In [3]: from ophyd import Signal

In [4]: sig = Signal(name="sig", value=None)

In [5]: print(f"{sig = }")
sig = Signal(name='sig', value=None, timestamp=1622771782.057922)

In [6]: hasattr(sig, "_status")
Out[6]: False

In [7]: sig.set(1.0)
Out[7]: Status(obj=Signal(name='sig', value=None, timestamp=1622771782.057922), done=False, success=False)

In [8]: print(f"{sig = }")
sig = Signal(name='sig', value=1.0, timestamp=1622771822.4205623)

In [9]: hasattr(sig, "_status")
Out[9]: True
```

prjemian, 2021-06-04:
For now, make sure `self.flying._status` exists from the start.

commit adf01c2f (2021-06-04): FIX #501
closed 2021-06-04 by prjemian